fwupd/fwupd #487: Lots of warnings during documentation building

ArchangeGabriel, 2018-05-01:
Thought you might be interested in those, so just in case:
```
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:128: warning: Parameter description for fu_common_get_files_recursive::directory is not used from source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:128: warning: Parameter description for fu_common_get_files_recursive::path is missing in source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:605: warning: Parameter description for fu_common_write_uint16::error is not used from source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:605: warning: Parameter description for fu_common_write_uint16::endian is missing in source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:630: warning: Parameter description for fu_common_write_uint32::error is not used from source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:630: warning: Parameter description for fu_common_write_uint32::endian is missing in source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:656: warning: Parameter description for fu_common_read_uint16::error is not used from source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:656: warning: Parameter description for fu_common_read_uint16::endian is missing in source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:683: warning: Parameter description for fu_common_read_uint32::error is not used from source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-common.c:683: warning: Parameter description for fu_common_read_uint32::endian is missing in source code comment block.
unknown:0: warning: Field descriptions for struct FuDeviceClass are missing in source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-device-list.c:378: warning: Parameter description for fu_device_list_add::error is not used from source code comment block.
unknown:0: warning: Value descriptions for FuHistoryFlags are missing in source code comment block.
unknown:0: warning: Field descriptions for struct FuKeyringClass are missing in source code comment block.
unknown:0: warning: Field descriptions for struct FuPluginClass are missing in source code comment block.
unknown:0: warning: Field descriptions for struct FuUsbDeviceClass are missing in source code comment block.
/build/fwupd/src/fwupd-1.0.7/src/fu-usb-device.c:469: warning: Parameter description for fu_usb_device_new::usb_device is missing in source code comment block.
unknown:0: warning: Field descriptions for struct FwupdClientClass are missing in source code comment block.
unknown:0: warning: Value descriptions for FwupdRemoteKind are missing in source code comment block.
./libfwupd-unused.txt:1: warning: 6 unused declarations. They should be added to libfwupd-sections.txt in the appropriate place.

html/api-index-full.html:93: warning: no link for: "fu-common-store-from-cab-bytes" -> (fu_common_store_from_cab_bytes).
html/api-index-full.html:93: warning: no link for: "libfwupd-fu-common-cab" -> (fu-common-cab).
html/api-index-full.html:105: warning: no link for: "libfwupd-fu-config" -> (fu-config).
html/api-index-full.html:109: warning: no link for: "fu-config-get-archive-size-max" -> (fu_config_get_archive_size_max).
html/api-index-full.html:113: warning: no link for: "fu-config-get-blacklist-devices" -> (fu_config_get_blacklist_devices).
html/api-index-full.html:117: warning: no link for: "fu-config-get-blacklist-plugins" -> (fu_config_get_blacklist_plugins).
html/api-index-full.html:121: warning: no link for: "fu-config-get-remotes" -> (fu_config_get_remotes).
html/api-index-full.html:125: warning: no link for: "fu-config-get-remote-by-id" -> (fu_config_get_remote_by_id).
html/api-index-full.html:129: warning: no link for: "fu-config-load" -> (fu_config_load).
html/api-index-full.html:133: warning: no link for: "fu-config-new" -> (fu_config_new).
html/api-index-full.html:138: warning: no link for: "fu-debug-get-option-group" -> (fu_debug_get_option_group).
html/api-index-full.html:138: warning: no link for: "libfwupd-fu-debug" -> (fu-debug).
html/api-index-full.html:150: warning: no link for: "libfwupd-fu-device-list" -> (fu-device-list).
html/api-index-full.html:310: warning: no link for: "fu-device-list-add" -> (fu_device_list_add).
html/api-index-full.html:314: warning: no link for: "fu-device-list-get-active" -> (fu_device_list_get_active).
html/api-index-full.html:318: warning: no link for: "fu-device-list-get-all" -> (fu_device_list_get_all).
html/api-index-full.html:322: warning: no link for: "fu-device-list-get-by-guid" -> (fu_device_list_get_by_guid).
html/api-index-full.html:326: warning: no link for: "fu-device-list-get-by-id" -> (fu_device_list_get_by_id).
html/api-index-full.html:330: warning: no link for: "fu-device-list-get-old" -> (fu_device_list_get_old).
html/api-index-full.html:334: warning: no link for: "fu-device-list-new" -> (fu_device_list_new).
html/api-index-full.html:338: warning: no link for: "fu-device-list-remove" -> (fu_device_list_remove).
html/api-index-full.html:499: warning: no link for: "libfwupd-fu-engine" -> (fu-engine).
html/api-index-full.html:503: warning: no link for: "fu-engine-add-device" -> (fu_engine_add_device).
html/api-index-full.html:507: warning: no link for: "fu-engine-add-plugin" -> (fu_engine_add_plugin).
html/api-index-full.html:511: warning: no link for: "fu-engine-add-runtime-version" -> (fu_engine_add_runtime_version).
html/api-index-full.html:515: warning: no link for: "fu-engine-check-plugins-pending" -> (fu_engine_check_plugins_pending).
html/api-index-full.html:519: warning: no link for: "fu-engine-check-requirements" -> (fu_engine_check_requirements).
html/api-index-full.html:523: warning: no link for: "fu-engine-clear-results" -> (fu_engine_clear_results).
html/api-index-full.html:527: warning: no link for: "fu-engine-get-action-id-for-device" -> (fu_engine_get_action_id_for_device).
html/api-index-full.html:531: warning: no link for: "fu-engine-get-archive-size-max" -> (fu_engine_get_archive_size_max).
html/api-index-full.html:535: warning: no link for: "fu-engine-get-details" -> (fu_engine_get_details).
html/api-index-full.html:539: warning: no link for: "fu-engine-get-devices" -> (fu_engine_get_devices).
html/api-index-full.html:543: warning: no link for: "fu-engine-get-downgrades" -> (fu_engine_get_downgrades).
html/api-index-full.html:547: warning: no link for: "fu-engine-get-history" -> (fu_engine_get_history).
html/api-index-full.html:551: warning: no link for: "fu-engine-get-releases" -> (fu_engine_get_releases).
html/api-index-full.html:555: warning: no link for: "fu-engine-get-remotes" -> (fu_engine_get_remotes).
html/api-index-full.html:559: warning: no link for: "fu-engine-get-results" -> (fu_engine_get_results).
html/api-index-full.html:563: warning: no link for: "fu-engine-get-status" -> (fu_engine_get_status).
html/api-index-full.html:567: warning: no link for: "fu-engine-get-store-from-blob" -> (fu_engine_get_store_from_blob).
html/api-index-full.html:571: warning: no link for: "fu-engine-get-upgrades" -> (fu_engine_get_upgrades).
html/api-index-full.html:575: warning: no link for: "fu-engine-install" -> (fu_engine_install).
html/api-index-full.html:579: warning: no link for: "fu-engine-load" -> (fu_engine_load).
html/api-index-full.html:583: warning: no link for: "fu-engine-modify-device" -> (fu_engine_modify_device).
html/api-index-full.html:587: warning: no link for: "fu-engine-modify-remote" -> (fu_engine_modify_remote).
html/api-index-full.html:591: warning: no link for: "fu-engine-new" -> (fu_engine_new).
html/api-index-full.html:595: warning: no link for: "fu-engine-profile-dump" -> (fu_engine_profile_dump).
html/api-index-full.html:599: warning: no link for: "fu-engine-unlock" -> (fu_engine_unlock).
[... 197 lines cut ...]
html/libfwupd-FuDevice.html:813: warning: no link for: "api-index-0.9.8" -> (0.9.8).
html/libfwupd-FuDevice.html:859: warning: no link for: "api-index-0.7.2" -> (0.7.2).
html/libfwupd-FuDevice.html:955: warning: no link for: "api-index-0.1.0" -> (0.1.0).
html/libfwupd-FuDevice.html:990: warning: no link for: "api-index-0.9.7" -> (0.9.7).
html/libfwupd-FuDevice.html:1177: warning: no link for: "api-index-0.7.1" -> (0.7.1).
html/libfwupd-FuDevice.html:1205: warning: no link for: "api-index-1.0.2" -> (1.0.2).
html/libfwupd-FuDevice.html:1265: warning: no link for: "api-index-1.0.3" -> (1.0.3).
[... cut ...]
html/libfwupd-FwupdClient.html:442: warning: no link for: "api-index-0.9.3" -> (0.9.3).
html/libfwupd-FwupdClient.html:1142: warning: no link for: "api-index-0.7.3" -> (0.7.3).
html/libfwupd-FwupdClient.html:1196: warning: no link for: "api-index-0.9.6" -> (0.9.6).
html/libfwupd-FwupdDevice.html:1420: warning: no link for: "api-index-0.9.4" -> (0.9.4).
html/libfwupd-FwupdRemote.html:439: warning: no link for: "api-index-0.9.5" -> (0.9.5).
html/libfwupd-fwupd-enums.html:243: warning: no link for: "api-index-0.1.1" -> (0.1.1).
```

closed 2018-05-02 by hughsie

superm1, 2018-05-02:
As part of fixing this is there a way to make these gtk-doc warnings fail the build for CI?  I didn't see anything obvious. I suppose one option is to log the build on one of the distros and scan for the word warning or error in the gtk-doc portion of it.

hughsie, 2018-05-02:
A lot of the warnings are invalid, I'd be hesitant to look for keywords. It's kinda a hack we're using a library-documentation tool to build the plugin docs.