fwupd/fwupd #489: Strange output during build

ArchangeGabriel, 2018-05-02:
While building latest release, I noticed this:
```
[71/220] /usr/bin/g-ir-scanner -pthread -I/usr/include/gobject-introspection-1.0 -I/usr/lib/libffi-3.2.1/include -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include --no-libtool --namespace=Fwupd --nsversion=2.0 --warn-all --output libfwupd/Fwupd-2.0.gir --c-include=fwupd.h -I/build/fwupd/src/fwupd-1.0.7/libfwupd -I/build/fwupd/src/build/libfwupd -I./libfwupd/.. -I../fwupd-1.0.7/libfwupd/.. --filelist=/build/fwupd/src/build/libfwupd/libfwupd@@fwupd@sha/Fwupd_2.0_gir_filelist --include=Gio-2.0 --include=GObject-2.0 --include=Soup-2.4 --symbol-prefix=fwupd --identifier-prefix=Fwupd --pkg-export=fwupd --cflags-begin -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-error=cpp -Wno-unknown-pragmas -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_GNU_SOURCE -I./libfwupd/.. -I../fwupd-1.0.7/libfwupd/.. -pthread -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/glib-2.0/include -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/json-glib-1.0 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -D_FORTIFY_SOURCE=2 --cflags-end -L/build/fwupd/src/build/libfwupd --extra-library=gio-2.0 --extra-library=gobject-2.0 --extra-library=glib-2.0 --extra-library=soup-2.4 --extra-library=json-glib-1.0 --library fwupd
/usr/include/stdlib.h:140: syntax error, unexpected identifier, expecting ',' or ';' in '# 140 "/usr/include/stdlib.h" 3 4' at 'strtof32'
/usr/include/stdlib.h:141: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '     char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:141: syntax error, unexpected ')', expecting ',' or ';' in '     char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:147: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '     char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:147: syntax error, unexpected ')', expecting ',' or ';' in '     char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:158: syntax error, unexpected identifier, expecting ',' or ';' in 'extern _Float32x strtof32x (const char *__restrict __nptr,' at 'strtof32x'
/usr/include/stdlib.h:159: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '       char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:159: syntax error, unexpected ')', expecting ',' or ';' in '       char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:165: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '       char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:165: syntax error, unexpected ')', expecting ',' or ';' in '       char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:233: syntax error, unexpected identifier in '         _Float32 __f)' at '_Float32'
/usr/include/stdlib.h:239: syntax error, unexpected identifier in '         _Float64 __f)' at '_Float64'
/usr/include/stdlib.h:251: syntax error, unexpected identifier in '   _Float32x __f)' at '_Float32x'
/usr/include/stdlib.h:257: syntax error, unexpected identifier in '   _Float64x __f)' at '_Float64x'
/usr/include/stdlib.h:316: syntax error, unexpected identifier, expecting ',' or ';' in '# 316 "/usr/include/stdlib.h" 3 4' at 'strtof32_l'
/usr/include/stdlib.h:317: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '       char **__restrict __endptr,' at 'char'
/usr/include/stdlib.h:318: syntax error, unexpected typedef-name, expecting identifier or '(' or '*' in '       locale_t __loc)' at 'locale_t'
/usr/include/stdlib.h:324: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '       char **__restrict __endptr,' at 'char'
/usr/include/stdlib.h:325: syntax error, unexpected typedef-name, expecting identifier or '(' or '*' in '       locale_t __loc)' at 'locale_t'
/usr/include/stdlib.h:337: syntax error, unexpected identifier, expecting ',' or ';' in 'extern _Float32x strtof32x_l (const char *__restrict __nptr,' at 'strtof32x_l'
/usr/include/stdlib.h:338: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '         char **__restrict __endptr,' at 'char'
/usr/include/stdlib.h:339: syntax error, unexpected typedef-name, expecting identifier or '(' or '*' in '         locale_t __loc)' at 'locale_t'
/usr/include/stdlib.h:345: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '         char **__restrict __endptr,' at 'char'
/usr/include/stdlib.h:346: syntax error, unexpected typedef-name, expecting identifier or '(' or '*' in '         locale_t __loc)' at 'locale_t'
g-ir-scanner: link: cc -o /build/fwupd/src/build/tmp-introspectzvelt9p8/Fwupd-2.0 -D_FORTIFY_SOURCE=2 -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt /build/fwupd/src/build/tmp-introspectzvelt9p8/Fwupd-2.0.o -L. -Wl,-rpath,. -Wl,--no-as-needed -lfwupd -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lsoup-2.4 -ljson-glib-1.0 -L/build/fwupd/src/build/libfwupd -Wl,-rpath,/build/fwupd/src/build/libfwupd -lgio-2.0 -lgobject-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0 -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now
```

The build went fine and tests succeeded, but I’m intrigued.

ArchangeGabriel, 2018-05-02:
F28 has glibc 2.27 I guess?

ArchangeGabriel, 2018-05-02:
This seems to be the culprit: https://sourceware.org/git/?p=glibc.git;a=commit;h=9725517070c1dda41aa3676cf1cf61daa527e14e

hughsie, 2018-05-03:
Indeed: glibc-2.27-8.fc28.x86_64

hughsie, 2018-08-10:
Fixed with https://github.com/hughsie/fwupd/commit/68f12ddb442b1411f593c6bbfa076359934559ae

closed 2018-08-10 by hughsie

ArchangeGabriel, 2018-08-10:
Nice, thanks.

ArchangeGabriel, 2018-08-10:
BTW @hughsie this is a bit off-topic, but if you didn’t read the ColorHug ML lastly, it appears you still need to update https://hughski.com/colorhugplus.html to avoid people sending emails to the ML asking for pre-ordering infos.