fwupd/fwupd #491: The Nintendo Switch Pro controller doesn't work in USB mode when fwupd is running

switchwitch, 2018-05-03:
If I attach a Nintendo Switch Pro controller via USB when fwupd is running, its devices are immediately removed and it can't be used. After killing the fwupd process and reattaching the controller, it works as expected. (Bluetooth mode works regardless of fwupd running or not.)

I use the udev rules posted [here](https://github.com/ValveSoftware/steam-for-linux/issues/4920#issuecomment-386188223)

dmesg output:
```[ 2566.411923] usb 3-8: new full-speed USB device number 21 using xhci_hcd
[ 2566.561412] usb 3-8: New USB device found, idVendor=057e, idProduct=2009
[ 2566.561415] usb 3-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2566.561418] usb 3-8: Product: Pro Controller
[ 2566.561420] usb 3-8: Manufacturer: Nintendo Co., Ltd.
[ 2566.561422] usb 3-8: SerialNumber: 000000000001
[ 2566.563322] input: Nintendo Co., Ltd. Pro Controller as /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37
[ 2566.563553] hid-generic 0003:057E:2009.001E: input,hidraw6: USB HID v1.11 Joystick [Nintendo Co., Ltd. Pro Controller] on usb-0000:00:14.0-8/input0
```

Accoding to udevadm monitor, the kernel and udev adds and then immediately removes the device
```KERNEL[2566.249701] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8 (usb)
KERNEL[2566.250160] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
KERNEL[2566.251207] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
KERNEL[2566.251472] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37 (input)
KERNEL[2566.251577] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/event20 (input)
KERNEL[2566.251616] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/js1 (input)
KERNEL[2566.251676] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/hidraw/hidraw6 (hidraw)
KERNEL[2566.251716] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
KERNEL[2566.251753] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
KERNEL[2566.251793] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8 (usb)
UDEV  [2566.767413] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8 (usb)
KERNEL[2566.770034] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/event20 (input)
UDEV  [2566.770225] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
UDEV  [2566.772481] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
UDEV  [2566.775916] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/hidraw/hidraw6 (hidraw)
KERNEL[2566.788121] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/js1 (input)
KERNEL[2566.824088] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37 (input)
KERNEL[2566.824116] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/hidraw/hidraw6 (hidraw)
KERNEL[2566.824132] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
KERNEL[2566.824148] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
KERNEL[2566.824160] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
KERNEL[2566.824184] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
KERNEL[2566.824417] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
UDEV  [2566.824676] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37 (input)
UDEV  [2566.825674] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/js1 (input)
UDEV  [2566.825886] add      /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/event20 (input)
UDEV  [2566.826445] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
UDEV  [2566.827006] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
UDEV  [2566.827565] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8 (usb)
UDEV  [2566.828229] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/js1 (input)
UDEV  [2566.828263] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37/event20 (input)
UDEV  [2566.828441] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/hidraw/hidraw6 (hidraw)
UDEV  [2566.828711] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E/input/input37 (input)
UDEV  [2566.829075] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
UDEV  [2566.829438] remove   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0/0003:057E:2009.001E (hid)
UDEV  [2566.829811] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
UDEV  [2566.830485] bind     /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
UDEV  [2566.831136] unbind   /devices/pci0000:00/0000:00:14.0/usb3/3-8/3-8:1.0 (usb)
```

This error shows up in syslog:
```fwupd[1786]: failed to add USB device: failed to send to device on ep 0x01: USB error on device 057e:2009 : Input/output error [-1]```

I think what may be happening is that fwupd thinks the controller is a SN30/SF30 pro, but this is an official Nintendo Switch Pro controller.

-
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34```

Please answer the following questions:

- Operating system and version:
Ubuntu Budgie 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
It was installed by default
- Have you tried rebooting?
Yes
- Are you using an NVMe disk?
No

superm1, 2018-05-04:
Well that's unfortunate.  I tested with my Nintendo switch pro controller before this went in and the controller didn't work with USB even before it for me.  I guess we should block that PID as well (similar fix to https://github.com/hughsie/fwupd/commit/ead5bf39295ac10589f1e26133f4bc9e955a956c).

Would you be able to test https://github.com/hughsie/fwupd/pull/493 to see if it fixes it for you?

Directions to build packages for Ubuntu are available here: https://github.com/hughsie/fwupd/tree/master/contrib

Just switch to my branch (wip/superm1/nintendo-switch-pro-controller) before running the build and you should get the integrated fix.

switchwitch, 2018-05-05:
Thank you. I'm away for a few days, but I'll try it when I get back.

juszczakn, 2018-10-13:
For anyone else on Ubuntu 18.04 who found this issue, looks like it's still not made it's way into the release yet. I'm still seeing the same fwupd version and error in syslog :(

superm1, 2018-10-15:
@juszczakn for now you can install the snap instead to get a newer version.
https://snapcraft.io/fwupd/

Plagman, 2018-10-15:
Is there something we could do from the Steam side to inhibit/disable fwupd for these devices on buggy revisions? We could add udev rules or drop another root-owned file somewhere through an update of our .deb if needed.

superm1, 2018-10-15:
FYI here's the SRU bug for updating to 1.0.9 (which will fix the problem).
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1791999

@Plagman 
Regarding if Steam can do anything to inhibit fwupd detecting this device, can you do something in steam deb `postinst`?  If so can you modify `/etc/fwupd/daemon.conf` to add `USB\VID_057E&PID_2009` to `BlacklistDevices` when you detect a package fwupd version less than `1.0.8` installed?

superm1, 2018-10-23:
@Plagman (or anyone else affected)

The 1.0.9 SRU has been accepted into bionic-proposed.  If you can please test it and confirm on the issue that it fixes the issue?  You can add a comment to the launchpad issue.
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1791999

superm1, 2018-12-04:
@Plagman FYI the SRU for 1.0.9 was promoted to -updates yesterday in ubuntu-land.  Anyone encountering it should be able to just do apt updates and it will go away now.