fwupd/fwupd #492: Test fail when /etc/os-release isn't available

Moredread, 2018-05-04:
I'm working on updating fwupd on NixOS to the latest version. When building in a sandbox the fu-self-test fails, as ` /etc/os-release`  isn't available.

````
running tests
ninja: Entering directory `/build/fwupd-1.0.7/build'
ninja: no work to do.
 1/10 fwupd-exported-api                      OK       0.01 s
 2/10 fwupd-self-test                         OK       0.01 s
 3/10 fu-self-test                            FAIL     0.02 s
 4/10 dfu-self-test                           OK       0.01 s
 5/10 nitrokey-self-test                      OK       0.01 s
 6/10 udev-self-test                          OK       0.01 s
 7/10 unifying-self-test                      OK       0.01 s
 8/10 thunderbolt-self-test                   OK       5.57 s
 9/10 dell-self-test                          OK       0.01 s
10/10 synapticsmst-self-test                  OK       0.01 s

OK:         9
FAIL:       1
SKIP:       0
TIMEOUT:    0

The output from the failed tests:

 3/10 fu-self-test                            FAIL     0.02 s

--- command ---
/build/fwupd-1.0.7/build/src/fu-self-test
--- stdout ---
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.787: removing /tmp/fwupd-self-test
/fwupd/device-locker{success}: OK
/fwupd/device-locker{fail}: OK
/fwupd/device{metadata}: OK
/fwupd/device-list: (/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a for device1
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using a050b517-6677-5119-9a77-2d26bbf30507 for foobar
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: ::added 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using 1a8d0d9a96ad3e67ba76cf3033623625dc6d6882 for device2
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using 579a3b1c-d1db-5bdc-b6b9-e2c1b28d5b8a for baz
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: ::added 1a8d0d9a96ad3e67ba76cf3033623625dc6d6882
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: ::removed 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a
OK
/fwupd/device-list{delay}: (/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a for device1
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using a050b517-6677-5119-9a77-2d26bbf30507 for foobar
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: ::added 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: device 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a already exists, ignoring
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: using 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a for device1
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.789: device 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a already exists, ignoring
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: waiting 100ms for device removal
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: found existing device 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a, reusing item
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: ::changed 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a
OK
/fwupd/device-list{compatible}: (/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: using 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a for device1
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: using a050b517-6677-5119-9a77-2d26bbf30507 for foobar
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: using 015370aa-26f2-5daa-9661-a75bf4c1a913 for bootloader
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: ::added 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: using 1a8d0d9a96ad3e67ba76cf3033623625dc6d6882 for device2
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: using 015370aa-26f2-5daa-9661-a75bf4c1a913 for bootloader
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: waiting 100ms for device removal
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: found compatible device 99249eb1bd9ef0b6e192b271a8cb6a3090cfec7a recently removed, reusing item from plugin plugin-for-runtime for plugin plugin-for-bootloader
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: adding GUID a050b517-6677-5119-9a77-2d26bbf30507 to device
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: copying old vendor ID USB:0x20A0 to new device
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: copying old version 1.2.3 to new device
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: ::changed 1a8d0d9a96ad3e67ba76cf3033623625dc6d6882
OK
/fwupd/engine{history-success}: (/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.799: performing init() on /build/fwupd-1.0.7/build/plugins/test/libfu_plugin_test.so
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): FuPluginTest-DEBUG: 11:46:02.800: init
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): As-DEBUG: 11:46:02.800: run FuEngine:load
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fu-DEBUG: 11:46:02.800: loading config values from /build/fwupd-1.0.7/data/tests/daemon.conf
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fwupd-DEBUG: 11:46:02.800: looking for os-release at /etc/os-release
(/build/fwupd-1.0.7/build/src/fu-self-test:17440): Fwupd-DEBUG: 11:46:02.800: looking for os-release at /usr/lib/os-release
--- stderr ---
**
Fu:ERROR:../src/fu-self-test.c:455:fu_engine_history_func: assertion failed (error == NULL): Failed to load config: No os-release found (FwupdError, 5)
-------
````

Would it be possible skipping the test or substituting a default value on your side, when the file isn't available? I can imagine that it might also be an issue when building in certain containerized environments.

hughsie, 2018-05-04:
I think `/etc/os-release` should be provided in the sandbox; it is for flatpak at least.

Moredread, 2018-05-04:
The problem (similarly to https://github.com/hughsie/fwupd/issues/403) is, that we can't write files in the sandbox environment. Do you know where /etc/os-release is standardized? I could make a case that it should be generally provided in the nix build sandbox.

hughsie, 2018-05-04:
For flatpak it's just bind-mounted from the host. See https://www.freedesktop.org/software/systemd/man/os-release.html for the spec.

Moredread, 2018-05-04:
The sandbox is intended to increase isolation from the system the package is build on afaik, i.e. so that packages built on different systems can be as identical as possible. So I expect that bind mounting will not work. But I'll ask if we can provide a dummy file.

superm1, 2018-05-04:
Something else worth mentioning is that systemd expects `/etc/os-release` everywhere since 2012.
http://0pointer.de/blog/projects/os-release.html

closed 2018-05-15 by superm1