fwupd/fwupd #499: Invalid checksum on XPS 13 9360 System Firmware v0.2.6.2

cajun-rat, 2018-05-07:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	0.9.7
daemon version:	0.9.7
compile-time dependency versions
	appstream-glib:	0.7.1
	gusb:	0.2.11

```
Please answer the following questions:

- Operating system and version: Ubuntu 17.10
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get
- Have you tried rebooting? Nope
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Unknown

I think the metadata for the latest XPS 13 9360 System Firmware might be corrupt. It fails metadata validation:

```
$ sudo fwupdmgr refresh
Fetching metadata https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz
Downloading…           [****************************************]
Fetching signature https://s3.amazonaws.com/lvfsbucket/downloads/firmware.xml.gz.asc
Downloading…           [****************************************]
$ sudo fwupdmgr update
Downloading 0.2.6.2 for XPS 13 9360 System Firmware...
Fetching firmware https://fwupd.org/downloads/cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab
Downloading…           [****************************************]
Checksum invalid, expected *1b4d6294c03f6e14d875f4ce42311168aa2f5b8c* got 959f78e66aead47f12954c3fba28344d7d4f7f2b
```

Here's my debugging so far: The relevant chunk of firmware.xml.gz:

```
      <release timestamp="1519627340" urgency="high" version="132610">
        <location>https://fwupd.org/downloads/cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab</location>
        <checksum type="sha1" filename="cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab" target="container">959f78e66aead47f12954c3fba28344d7d4f7f2b</checksum>
        <checksum type="sha1" filename="firmware.bin" target="content">20ba7a32ed0401fc7d2aedb0d8f9fec32f8c33b5</checksum>
        <description><p>Fixes:^M  - Fixed incorrect Windows operating system logon message.^M  ^M Enhancements:^M  - Enhanced SMM Security Mitigation feature.^M  - Updated Embedded Controller Engine Firmware.^M  - Updated Realtek USB LAN firmware&lt;/p&gt;</p></description>
        <size type="installed">11248730</size>
        <size type="download">10965873</size>
      </release>
      <release timestamp="1516880489" version="132353">
        <location>https://fwupd.org/downloads/205ca967cdf0928d2b7bbd3277ea3da890c782ce-Signed_1152921504627452016.cab</location>
        <checksum type="sha1" filename="205ca967cdf0928d2b7bbd3277ea3da890c782ce-Signed_1152921504627452016.cab" target="container">1b4d6294c03f6e14d875f4ce42311168aa2f5b8c</checksum>
        <checksum type="sha1" filename="firmware.bin" target="content">8fa598b30b00f09246a0b0c7bb7e2559e74ee773</checksum>
        <size type="installed">11226623</size>
        <size type="download">10944025</size>
      </release>
````

That file downloads and checksums correctly:

```shell
curl https://fwupd.org/downloads/cddd4fbfb52ded7f300780973d751b2c12c0fe65-Signed_1152921504627521618.cab | sha1sum 
=> *959f78e66aead47f12954c3fba28344d7d4f7f2b*  -
```

But fwupd is somehow expecting the checksum to be `1b4d6294c...`, which is the checksum of the other release.

cajun-rat, 2018-05-11:
OK, will just need to figure out how to build it as a Debian package

superm1, 2018-05-15:
@cajun-rat 

Once you clone the tree and switch to 0_9_X branch you should be able to build packages using this information: https://github.com/hughsie/fwupd/tree/master/contrib

cajun-rat, 2018-06-18:
It turns out that I didn't need upgrade `fwupd` in the end.  I refreshed the metadata from the server and the updates installed correctly.

I can't reproduce the error any more and it works for me, so I'll close this issue.

closed 2018-06-18 by cajun-rat