fwupd/fwupd #500: Please add documentation on how to

MaxTranced, 2018-05-08:
I'm hitting the issue described here:
https://askubuntu.com/questions/1031871/cant-upgrade-logitech-unify-firmware-using-fwupd

I don't know how to manually put the Device Unifying [runtime] in update mode. It would be quite handy if the message included details on how to do that.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	0.8.3
daemon version:	0.8.3
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Latitude E5570
  Guid:                 6180aaaa-5529-4bbf-b4fd-65b4f788de5b
  DeviceID:             UEFI-6180aaaa-5529-4bbf-b4fd-65b4f788de5b-dev0
  Plugin:               uefi
  Flags:                internal|allow-offline|require-ac|supported
  Version:              0.1.18.6
  VersionLowest:        0.1.18.6
  Created:              2018-05-08

Razer DeathStalker Essential 2014
  Guid:                 92838407-788f-5ef9-a28a-f31bd39bed39
  Guid:                 23dab496-fb19-5727-bbf4-90b7ae681ead
  DeviceID:             usb:00:05:03
  Plugin:               usb
  Flags:                none
  Version:              2.0
  Created:              2018-05-08

Mars [Radeon HD 8670A/8670M/8750M]
  Guid:                 a3cc284a-dd03-5cb7-9fdf-cd09cc0905e4
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  Plugin:               udev
  Flags:                internal|locked
  DeviceVendor:         Advanced Micro Devices, Inc. [AMD/ATI]
  Created:              2018-05-08

Unifying [runtime]
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  UniqueID:             com.logitech.Unifying.RQR12.firmware
  DeviceID:             usb:00:01
  Description:          <p>A Unifying receiver allows you to connect multiple compatible keyboards and mice to a laptop or desktop computer with a single USB receiver. Updating the firmware on your Unifying receiver improves performance, adds new features and fixes security issues.</p>
  Plugin:               unifying
  Flags:                allow-online|supported|needs-bootloader
  DeviceVendor:         Logitech
  Version:              012.003.00025
  VersionBootloader:    BL.002.015
  Created:              2018-05-08
  AppstreamId:          com.logitech.Unifying.RQR12.firmware
  Summary:              Firmware for the Logitech Unifying receiver
  UpdateDescription:    <p>This release addresses an unencrypted keystroke injection issue known as Bastille security issue #11. The vulnerability is complex to replicate and would require a hacker to be physically close to a target.</p>
  UpdateVersion:        RQR12.07_B0029
  UpdateHash:           d0d33e760ab6eeed6f11b9f9bd7e83820b29e970
  UpdateChecksumKind:   sha1
  License:              Proprietary
  UpdateUri:            https://fwupd.org/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
  UrlHomepage:          http://support.logitech.com/en-us/software/unifying
  Vendor:               Logitech
  Trusted:              none
```
Please answer the following questions:

- Operating system and version:

LSB Version:	core-9.20160110ubuntu0.2-amd64:core-9.20160110ubuntu0.2-noarch:security-9.20160110ubuntu0.2-amd64:security-9.20160110ubuntu0.2-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.4 LTS
Release:	16.04
Codename:	xenial

MaxTranced, 2018-05-08:
I've tried updating to 1.0.x via the instructions here https://medium.com/@george.shuklin/how-to-install-packages-from-a-newer-distribution-without-installing-unwanted-6584fa93208f but it's not trivial.

superm1, 2018-05-08:
I would recommend to try booting from Ubuntu 18.04 live media and trying there. It has 1.0.x and you can try without disrupting your existing installation.

MaxTranced, 2018-05-08:
> I would recommend to try booting from Ubuntu 18.04 live media and trying there. It has 1.0.x and you can try without disrupting your existing installation.

I've done just that. It worked! Thank you @superm1 !

FWIW, using version 1.0.x there was no mention of manual update mode during the process, so I think that this issue is good to close (I don't know if version 0.9.x required this manual update mode or not).

superm1, 2018-05-08:
Great! Glad to hear it.

closed 2018-05-08 by superm1

MaxTranced, 2018-05-08:
That approach to updating is described here https://askubuntu.com/a/958771 as well.