fwupd/fwupd #512: BIOS updates no longer detected for Dell Precision 5510

blunden, 2018-05-15:
I've been using fwupdmgr without issues for quite a while at this point but recently I noticed that it doesn't detect the new 0.1.7.0 update, despite it being included in the XML file it downloads. 

Below are the details from the manually downloaded cab in question: 

```shell
$ fwupdmgr get-details 493cb1217b0a4654bb63b8fdf423a8740d81029c-firmware_XPS9550_Precision5510_1.7.0.wu.cab 
Decompressing…         [***************************************]
Precision 5510 System Firmware
  DeviceId:             2da31270d317b076424992de14a0f08ae373c137
  Guid:                 124c207d-5db8-4d95-bd31-34fd971b34f9
  Description:          <p>Updating the system firmware improves performance.</p>
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  
  [Release]
  AppstreamId:          com.dell.uefi124c207d.firmware
  RemoteId:             lvfs
  Summary:              Firmware for the Dell XPS 15 9550/Precision 5510
  Version:              0.1.7.0
  Filename:             firmware.bin
  License:              proprietary
  Size:                 10,2 MB
  Homepage:             http://support.dell.com/
  Vendor:               Dell Inc.
  TrustFlags:           payload
```

Below is the current device information for that same device:

```shell
$ fwupdmgr get-devices
Precision 5510 System Firmware
  DeviceId:             2da31270d317b076424992de14a0f08ae373c137
  Guid:                 124c207d-5db8-4d95-bd31-34fd971b34f9
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.6.1
  VersionLowest:        0.1.6.1
  Icon:                 computer
  Created:              2018-05-15
```

The DeviceId and Guid values both match but it still isn't detected, which to me suggests that the issue might be with the XML file.

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

- Operating system and version: Ubuntu 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): It was preinstalled if I recall correctly. Regardless, I'm using the version from Ubuntu's repository.
- Have you tried rebooting? Yes 
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No
- Have you used fwupdmgr successfully in the past? Yes, the install process has always worked fine.

superm1, 2018-05-15:
This was also reported on the mailing list.  What's going on is this particular BIOS supports two platforms (Precision 5510 and XPS 9550).  On LVFS it was declared to have Computer Hardware ID (CHIDs) for both platforms as well.  Older fwupd versions (or those compiled against older appstream-glib) ignored this information.  Newer fwupd versions recognize it but there was a bug found in the handling of this with multiple CHIDs.  Fwupd was handling them as a AND relationship rather than OR relationship.

So changes have been made on both the LVFS side as well as the fwupd side to resolve it.  See these for more information.
https://github.com/hughsie/fwupd/commit/3d71c16ed391edd49a72fb1e59e6b4d995635ca3#diff-25d902c24283ab8cfbac54dfa101ad31
https://github.com/hughsie/lvfs-website/issues/110

This will be fixed in the fwupd 1.0.8 release.  As you saw, for now you can manually download the file and install it using fwupdmgr.

closed 2018-05-15 by superm1

blunden, 2018-05-15:
Oh, sorry. I didn't know there was a mailing list I could check too.

Thanks for the information.

superm1, 2018-05-15:
No worries.  It's not at all obvious what's going on here.

blunden, 2018-09-17:
I once again had an issue where fwupdmgr fialed to detect that a new 0.1.8.0 system firmware update was available, forcing me to manually download and install the cab file using `fwupdmgr install`. It seems Ubuntu hasn't updated to the fwupdate 1.0.8 release or later yet though so that's almot certainly why. The upcoming Ubuntu 18.10 release seems to have a much newer version though.

Curiously, after running fwupdmgr again after flashing the update, I got a message asking me if I wanted to report a successful update from version 0.1.6.0 -> 0.1.7.0 (I had just updated from 0.1.7.0 -> 0.1.8.0). I don't know why it finally decided to detect that. :)

superm1, 2018-09-17:
I've filed an SRU bug in Ubuntu 18.04 to pull in the new release here:
https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1791999

In the interim however; we now have a snap available if you would like to use that.

https://github.com/hughsie/fwupd/wiki/fwupd-snap

Ericson2314, 2021-07-02:
Hmm I still get this with version 1.5.7 on NixOS 2021. It claims there is no bios update for me, but then I do
```
 $ fwupdmgr get-details ~/Downloads/b656e631b3724e2dd129333f27e60472e4ef2ff9918669e5e52f9cc8f5323d59-XPS_9550_1.14.0.cab
Decompressing…           [***************************************]
XPS 15 9550
│
└─System Firmware:
  │   Device ID:          2da31270d317b076424992de14a0f08ae373c137
  │   Description:
  │   Updating the system firmware improves performance.
  │   Current version:    1.1.19
  │   GUID:               124c207d-5db8-4d95-bd31-34fd971b34f9
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │
  └─XPS 15 9550/Precision 5510:
        New version:      1.14.0
        Remote ID:        lvfs
        Summary:          Firmware for the Dell XPS 15 9550/Precision 5510
        License:          Proprietary
        Size:             10.6 MB
        Urgency:          Critical
        Vendor:           Dell Inc.
        Flags:            trusted-payload|trusted-metadata
        Description:
        This stable release fixes the following issues:

        • Firmware updates to address the Intel Security Advisories.
        • Fixed the issue where the SMM Security Mitigation gets disabled while loading BIOS defaults.
```
and it change its mind.