fwupd/fwupd #523: Required ESP size for UEFI updates?

ArchangeGabriel, 2018-05-22:
Hi there,

I’m currently doing some research on minimal ESP size. As far as file system goes, unless they are some very strange UEFI implentation, there is not really a limit.

However, since fwupdate needs the capsule to be available on the ESP, this greatly increases the requirements. So, I was wondering what where the typical sizes of capsule updates, and thus the required available space on the ESP.

Thanks!

superm1, 2018-05-22:
So there's a few ways to go about coming up with this number.
* Following the size of the BIOS SPI part isn't a good method because some implementations (Such as Dell's) don't flash the capsule directly to the BIOS SPI part but rather use it to flash many devices.
* Following the average capsule size on LVFS isn't a good method because it's possible to also put multiple SPI payloads in a capsule (such as for different CPU SKUs) and let the flashing process choose the right one, although this isn't common.

So those comments in mind I think the smallest size I've seen is around 11MB and biggest I've seen was maybe 32MB?  I don't think you should use either of those numbers though.

Here's other things on the ESP and relative sizes:
* Dell uses the ESP for BIOS recovery purposes, this will take the size of BIOS for that platform *2 to store (current and previous).
* Linux EFI executables < 1M

Maybe @hughsie can confirm the biggest UEFI capsule present on LVFS, but given everything I've mentioned above I think a safe round number should be 100MB (2x32 for recovery + 32 capsule + EFI executables)

hughsie, 2018-05-22:
14Mb is the biggest firmware on the LVFS. 32Mb isn't crazy, and so double that is 64Mb. I think 100Mb is a very sane number.

closed 2018-05-22 by superm1