fwupd/fwupd #526: Inconsistent licensing

superm1, 2018-05-23:
I was just glancing through headers and noticed that some files are marked GPL, some are LGPL.  The COPYING in the project is GPL though.  Is this all intentional?  Or should they all be LGPL?

hughsie, 2018-05-24:
I wasn't sure what to do here. LGPLv2+ is the correct license for libfwupd (and perhaps libdfu, in case we ever split it out again) but all the other files should be GPLv2+ I suppose. A few of the vendors I've written plugins for have specified LGPLv2+ for the plugin (on the assumption they're re-using the code somewhere else).

I guess we add a COPYING.lgpl? Is there a convention here with other projects?

superm1, 2018-05-24:
In that case I suspect it would make most sense to set up "all" plugins as LGPL. Reasoning being that "simple" plugins will probably be based on existing in 🌲 plugins and it prevents having to mix license of code in tree.

And yes I think an extra copying file should be distributed with the project for the LGPL stuff.

hughsie, 2018-05-24:
The alternative is just to relicense everything to LGPLv2+ -- it's easier to mix and match that way too. I don't know if a daemon that's LGPLv2+ makes sense.

superm1, 2018-05-24:
> The alternative is just to relicense everything to LGPLv2

That does seem easier to me.  I think you just need to get an OK from major contributors (AFAIK you and me) and that sounds good to me.

> I don't know if a daemon that's LGPLv2+ makes sense.

Well there's at least one other big project I know of that does this; systemd.

hughsie, 2018-05-25:
Maybe it makes sense to remove the boilerplate at the same time? e.g.

    /* -*- Mode: C; tab-width: 8; indent-tabs-mode: t; c-basic-offset: 8 -*-
     *
     * Copyright (C) 2016-2017 Richard Hughes <richard@hughsie.com>
     *
     * SPDX-License-Identifier: LGPL-2.1+
     */

superm1, 2018-05-25:
Yes I was going to propose that too in the pr for this :)

hughsie, 2018-05-28:
Okay, that's everyone involved okay with the relicencing. We're good to go.

> Yes I was going to propose that too in the pr for this :)

Do you have a PR ready? If so, that'd be awesome.

commit 51308e64 (2018-05-29): Adjust all licensing to LGPL 2.1+ (Closes: #526)
closed 2018-05-29 by hughsie