fwupd/fwupd #543: Add fwupdtool to build.snapcraft.io

superm1, 2018-05-31:
I don't have admin permissions so i'm not able to do this.

"Don’t have admin permission? Ask a repo admin to add it instead, and it will show up in your repo list too."
@hughsie can you please add it there?  That should let us connect the dots to master here.  (I was doing my initial tests with a fork)
https://build.snapcraft.io/select-repositories

hughsie, 2018-05-31:
A registered name of "fwupd" i assume?

hughsie, 2018-05-31:
Don't we want the generic "fwupd" so we can include them all together in one snap?

superm1, 2018-05-31:
I was thinking if that happens we should stop publishing fwupdtool and publish fwupd.  I just don't want to give the presumption this is a full "fwupd" when someone looks at it in a store.

hughsie, 2018-05-31:
For `fwupdtool` -- "Sorry, that name is already taken. Try a different name."

superm1, 2018-05-31:
... That's probably from my test run, hmm...

hughsie, 2018-05-31:
Can you unregister it and I'll try with that? I'm perfectly happy trying to get the fwupd name if that helps.

superm1, 2018-05-31:
I don't see an obvious way to get it unregistered.  Can you just try fwupdtool2 so we can see if it shows up in my repo list and I can link them?  If not, I guess I'll need to poke at the admins for this build.snapcraft.io

hughsie, 2018-05-31:
I've added hughsie-fwupd as it seems like the most correct thing to do. It's now wanting me to push the snapcraft.yaml file; perhaps any commit to the repo will kick things off?

superm1, 2018-05-31:
Yeah i'd guess so?

superm1, 2018-05-31:
I've posted in the snapcraft forum to see if they can figure out what we're supposed to do.
https://forum.snapcraft.io/t/cant-register-snap/5707

superm1, 2018-05-31:
@hughsie They responded apparently there is a transfer process, but they'll have to do it.
```
1- fwupdtool and fwupd would be “taken away” from Mario and moved to @hugsie’s account.
2- If @hugsie can indicate on his github issue https://github.com/hughsie/fwupd/issues/543 that he wants to accept the transfer, I can take that as his acceptance of this.
```

hughsie, 2018-05-31:
> that he wants to accept the transfer, I can take that as his acceptance of this

This is indeed my intention. I'm happy for @superm1 to have full commit and admin access on the snapcraft bundle too.

superm1, 2018-05-31:
OK you should have fwupdtool registered now.  I think you should be able to finish the linking of them

superm1, 2018-06-01:
I checked and I can access the fwupdtool store object at dashboard.snapcraft.io now that we can both see and change.  I still don't see the builds queuing though or publishing from build.snapcraft.io.  Do you have that part configured now to "fwupdtool"?

superm1, 2018-06-04:
If you'd prefer to keep it all fwupd, then just swap out the title in the snapcraft.yaml to 'fwupd' from 'fwupdtool', that's OK.  If/when we get to the place that we have daemon and fwupdmgr in package we'll figure out messaging then.

superm1, 2018-06-04:
The one thing that gets confusing though is with the way it's put right now (fwupdtool) it adds fwupdtool to $PATH (as it matches package name).  If you swap to fwupd it will put fwupd.fwupdtool to $PATH.

hughsie, 2018-06-04:
> If you'd prefer to keep it all fwupd

I really would. `fwupd` is our "trademark" and it's both confusing and slightly worrying to not be in control of the supposedly-upstream thing that users accommodate with the project.

> If you swap to fwupd it will put fwupd.fwupdtool to $PATH.

Surely `fwupdmgr` would become `fwupd.fwupdmgr` too? I think that's fine. It also means that `fwupdtool.dfu-tool` would be namespaced correctly.

superm1, 2018-06-04:
Yep.  OK, made relevant changes in https://github.com/hughsie/fwupd/pull/549.  I think that this should hopefully start working after that's in.

superm1, 2018-06-04:
Since the snap is renamed I think you need to add me in https://dashboard.snapcraft.io/snaps/fwupd/collaboration/

hughsie, 2018-06-04:
Done.

superm1, 2018-06-04:
OK so hopefully that PR cleared up the error here https://build.snapcraft.io/select-repositories for you and now it will upload a build to the store automatically.  If not, you might need to poke it once or something.

superm1, 2018-06-07:
OK it looks like it's publishing to edge channel properly.  On tagged releases I think we should promote up to beta channel.
I've made some documentation about it here:
https://github.com/hughsie/fwupd/wiki/fwupd-snap

Note: this is currently using '--devmode'.  It was denied as a 'classic' snap which means if we want users to be able to use it without `--devmode` we'll have to sort out confinement rules.  I'm going to open a separate task for that.

closed 2018-06-07 by superm1