fwupd/fwupd #548: Should the flatpak and snap builds use git master from everywhere?

hughsie, 2018-06-03:
An open question really. I'm flip-flopping on either using tarballs for things like fwupdate (on the logic, it will likely match the shipped, tested, QAd release) or maybe we should use fwupdate and libusb git master to find regressions in libraries we're using early in the dev cycle.

superm1, 2018-06-04:
I've been going back and forth on this too.  I'm tending to lean on keep released tarballs for everything that we can.  If something has an important pending fix try to push the upstream to get a new tarball out.

For the purpose of running bleeding edge maybe we can keep org.freedesktop.fwupd.conf and snapcraft.yaml files in tree that we can swap out and build a master build on a whim then.

hughsie, 2018-06-04:
Right, agreed. Thanks for the sanity check.

closed 2018-06-04 by hughsie