fwupd/fwupd #550: snap fwupdtool segfaults during cleanup

superm1, 2018-06-05:
I'm not sure what to make of this one, but I have found that specifically USB plugins cause a segfault when cleaning up if invoked through fwupdtool in a snap.  If compiled outside of a snap and ran in a normal environment it doesn't happen.  I'm guessing this means some bug in a library that's linked in.  I can also extract the snap and run it contained `SNAP=$(pwd) $SNAP_USER_DATA=$(pwd)` and I hit it too.

```
#sudo gdb --args fwupd.fwupdtool --plugin-whitelist=unifying get-devices 2>&1
GNU gdb (Ubuntu 8.1-0ubuntu3) 8.1.0.20180409-git
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from fwupd.fwupdtool...(no debugging symbols found)...done.
(gdb) run
Starting program: /snap/bin/fwupd.fwupdtool --plugin-whitelist=unifying get-devices
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff77c4700 (LWP 32475)]
[New Thread 0x7ffff6fc3700 (LWP 32476)]
[New Thread 0x7ffff67c2700 (LWP 32477)]
[New Thread 0x7ffff5fc1700 (LWP 32478)]
[New Thread 0x7ffff57c0700 (LWP 32479)]
[Thread 0x7ffff57c0700 (LWP 32479) exited]
[Thread 0x7ffff5fc1700 (LWP 32478) exited]
[Thread 0x7ffff67c2700 (LWP 32477) exited]
[Thread 0x7ffff6fc3700 (LWP 32476) exited]
[Thread 0x7ffff77c4700 (LWP 32475) exited]
process 32471 is executing new program: /usr/lib/snapd/snap-confine
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
process 32471 is executing new program: /usr/lib/snapd/snap-exec
[New LWP 32480]
[New LWP 32481]
[New LWP 32482]
[New LWP 32483]
[New LWP 32484]
[LWP 32484 exited]
[LWP 32483 exited]
[LWP 32482 exited]
[LWP 32481 exited]
[LWP 32480 exited]
process 32471 is executing new program: /bin/dash
process 32471 is executing new program: /bin/dash
process 32471 is executing new program: /snap/fwupd/x2/usr/lib/fwupd/fwupdtool
warning: File "/snap/core/4650/lib/x86_64-linux-gnu/libthread_db-1.0.so" auto-loading has been declined by your `auto-load safe-path' set to "$debugdir:$datadir/auto-load".
To enable execution of this file add
	add-auto-load-safe-path /snap/core/4650/lib/x86_64-linux-gnu/libthread_db-1.0.so
line to your configuration file "/home/supermario/.gdbinit".
To completely disable this security protection add
	set auto-load safe-path /
line to your configuration file "/home/supermario/.gdbinit".
For more information about this security protection see the
"Auto-loading safe path" section in the GDB manual.  E.g., run from the shell:
	info "(gdb)Auto-loading safe path"
warning: Unable to find libthread_db matching inferior's thread library, thread debugging will not be available.
[New LWP 32486]
[New LWP 32487]
[New LWP 32488]
Fu-Message: using plugins: unifying
Unifying Receiver
  DeviceId:             61fec1c5753c2c16a9e0455632a4f8ede9817292
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.01_B0019
  VersionBootloader:    BOT01.02_B0014
  Icon:                 preferences-desktop-keyboard
  Created:              2018-06-05

Thread 1 "fwupdtool" received signal SIGSEGV, Segmentation fault.
0x00007fc584cbb89d in ?? ()
(gdb) bt full
#0  0x00007fc584cbb89d in ?? ()
No symbol table info available.
#1  0x00007fc58dcefd9a in g_object_unref () from /snap/fwupd/x2/usr/lib/fwupd/../x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#2  0x00000000004231f3 in fu_device_list_item_free (item=0x828bc0) at ../src/fu-device-list.c:533
No locals.
#3  0x00007fc58d9df5ab in g_ptr_array_foreach () from /snap/fwupd/x2/usr/lib/fwupd/../../../lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#4  0x00007fc58d9df640 in ?? () from /snap/fwupd/x2/usr/lib/fwupd/../../../lib/x86_64-linux-gnu/libglib-2.0.so.0
No symbol table info available.
#5  0x0000000000423363 in fu_device_list_finalize (obj=0x7f8480) at ../src/fu-device-list.c:571
        self = 0x7f8480
#6  0x00007fc58dcefd9a in g_object_unref () from /snap/fwupd/x2/usr/lib/fwupd/../x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#7  0x000000000041e08a in fu_engine_finalize (obj=0x7fe820) at ../src/fu-engine.c:3441
        self = 0x7fe820
#8  0x00007fc58dcefd9a in g_object_unref () from /snap/fwupd/x2/usr/lib/fwupd/../x86_64-linux-gnu/libgobject-2.0.so.0
No symbol table info available.
#9  0x000000000040e4a7 in fu_util_private_free (priv=0x7f6a50) at ../src/fu-tool.c:206
No locals.
#10 0x000000000040e54f in glib_autoptr_cleanup_FuUtilPrivate (_ptr=0x7ffc54829040) at ../src/fu-tool.c:220
No locals.
#11 0x00000000004101cf in main (argc=2, argv=0x7ffc548292b8) at ../src/fu-tool.c:690
        allow_older = 0
        allow_reinstall = 0
        force = 0
        ret = 1
        verbose = 0
        plugin_glob = 0x7fc6c0
        priv = 0x7f6a50
        error = 0x0
        cmd_descriptions = 0x7fbcb0 "h\377\246\214\305\177"
        options = {{long_name = 0x42f0d8 "verbose", short_name = 118 'v', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffc5482902c, description = 0x42f0e0 "Show extra debugging information", 
            arg_description = 0x0}, {long_name = 0x42f101 "allow-reinstall", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffc54829024, 
            description = 0x42f118 "Allow re-installing existing firmware versions", arg_description = 0x0}, {long_name = 0x42f147 "allow-older", short_name = 0 '\000', flags = 0, 
            arg = G_OPTION_ARG_NONE, arg_data = 0x7ffc54829020, description = 0x42f158 "Allow downgrading firmware versions", arg_description = 0x0}, {long_name = 0x42f17c "force", 
            short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7ffc54829028, description = 0x42f182 "Override plugin warning", arg_description = 0x0}, {
            long_name = 0x42f19a "show-all-devices", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7f6a84, description = 0x42f1b0 "Show devices that are not updatable", 
            arg_description = 0x0}, {long_name = 0x42f1d4 "plugin-whitelist", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_STRING_ARRAY, arg_data = 0x7ffc54829038, 
            description = 0x42f1e8 "Manually whitelist specific plugins", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, 
            description = 0x0, arg_description = 0x0}}
(gdb) quit
A debugging session is active.

	Inferior 1 [process 32471] will be killed.

Quit anyway? (y or n) y

```

hughsie, 2018-06-05:
Anything from valgrind?

superm1, 2018-06-05:
Took some time to instrument it (debug symbols for libc6 are hard to put in place), but here's what I can get from valgrind.

```
(fwupdtool:757): Fu-DEBUG: performing destroy() on unifying
==757== Jump to the invalid address stated on the next line
==757==    at 0xE17889D: ???
==757==    by 0x52E4D99: g_object_unref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x4231F2: fu_device_list_item_free (fu-device-list.c:533)
==757==    by 0x55455AA: g_ptr_array_foreach (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x554563F: ??? (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x423362: fu_device_list_finalize (fu-device-list.c:571)
==757==    by 0x52E4D99: g_object_unref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x41E089: fu_engine_finalize (fu-engine.c:3441)
==757==    by 0x52E4D99: g_object_unref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x40E4A6: fu_util_private_free (fu-tool.c:206)
==757==    by 0x40E54E: glib_autoptr_cleanup_FuUtilPrivate (fu-tool.c:220)
==757==    by 0x4101CE: main (fu-tool.c:690)
==757==  Address 0xe17889d is not stack'd, malloc'd or (recently) free'd
==757== 
==757== 
==757== Process terminating with default action of signal 11 (SIGSEGV): dumping core
==757==  Access not within mapped region at address 0xE17889D
==757==    at 0xE17889D: ???
==757==    by 0x52E4D99: g_object_unref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x4231F2: fu_device_list_item_free (fu-device-list.c:533)
==757==    by 0x55455AA: g_ptr_array_foreach (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x554563F: ??? (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x423362: fu_device_list_finalize (fu-device-list.c:571)
==757==    by 0x52E4D99: g_object_unref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x41E089: fu_engine_finalize (fu-engine.c:3441)
==757==    by 0x52E4D99: g_object_unref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x40E4A6: fu_util_private_free (fu-tool.c:206)
==757==    by 0x40E54E: glib_autoptr_cleanup_FuUtilPrivate (fu-tool.c:220)
==757==    by 0x4101CE: main (fu-tool.c:690)
==757==  If you believe this happened as a result of a stack
==757==  overflow in your program's main thread (unlikely but
==757==  possible), you can try to increase the size of the
==757==  main thread stack using the --main-stacksize= flag.
==757==  The main thread stack size used in this run was 8388608.
==757== 
==757== HEAP SUMMARY:
==757==     in use at exit: 231,915 bytes in 1,472 blocks
==757==   total heap usage: 12,588 allocs, 11,116 frees, 6,571,043 bytes allocated
==757== 
==757== 8 bytes in 1 blocks are possibly lost in loss record 136 of 1,296
==757==    at 0x4C2DB8F: malloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==757==    by 0x4C2FDEF: realloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==757==    by 0x55767D7: g_realloc (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x52FC9D8: ??? (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x53022CC: g_type_register_static (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x52E143D: g_enum_register_static (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x58D8376: g_file_monitor_event_get_type (in /root/build/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0.4800.2)
==757==    by 0x5899297: ??? (in /root/build/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0.4800.2)
==757==    by 0x530022C: g_type_class_ref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x52FFE2C: g_type_class_ref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x52FFE2C: g_type_class_ref (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x58A2DB4: ??? (in /root/build/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0.4800.2)
==757== 
==757== 16 bytes in 1 blocks are possibly lost in loss record 446 of 1,296
==757==    at 0x4C2FB55: calloc (in /usr/lib/valgrind/vgpreload_memcheck-amd64-linux.so)
==757==    by 0x5576770: g_malloc0 (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x52FD522: ??? (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x5301F90: g_type_register_fundamental (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x52E130B: ??? (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x52DB177: ??? (in /root/build/usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0.4800.2)
==757==    by 0x40106B9: call_init.part.0 (dl-init.c:72)
==757==    by 0x40107CA: call_init (dl-init.c:30)
==757==    by 0x40107CA: _dl_init (dl-init.c:120)
==757==    by 0x4000C69: ??? (in /snap/core/current/lib64/ld-linux-x86-64.so.2)
==757==    by 0x3: ???
[... 365 lines cut ...]
==757==    by 0x5597E8E: ??? (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x5597F47: g_thread_new (in /root/build/lib/x86_64-linux-gnu/libglib-2.0.so.0.4800.2)
==757==    by 0x5DE6A7E: g_usb_context_initable_init (gusb-context.c:622)
[... cut ...]
==757==                         length64           : 592 bytes in 13 blocks
==757==                         newarray           : 1,632 bytes in 22 blocks
==757==         suppressed: 0 bytes in 0 blocks
==757== Reachable blocks (those to which a pointer was found) are not shown.
==757== To see them, rerun with: --leak-check=full --show-leak-kinds=all
==757== 
==757== For counts of detected and suppressed errors, rerun with: -v
==757== ERROR SUMMARY: 28 errors from 28 contexts (suppressed: 0 from 0)
Killed
```

superm1, 2018-06-06:
I've determined this isn't caused by compiling as a snap.  I reproduced it in regular Ubuntu 16.04 with compiling appstream-glib 0.7.9 and fwupd master.

fwupd compile flags
```
# meson build -Ddaemon=false -Dpkcs7=false -Dgpg=false -Dintrospection=false -Dplugin_uefi=false -Dplugin_uefi_labels=false -Dplugin_dell=false -Dsystemd=false -Dtests=false
```

I've got a better(ish) backtrace now with details in the frame, but not much closer to figuring out yet what's happening.

```
(gdb) bt full
#0  0x00007ffff015a87d in  ()
#1  0x00007ffff770cd9a in g_object_unref (_object=0x6b32e0) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./gobject/gobject.c:3183
        weak_locations = <optimized out>
        old_ref = 1
        object = 0x6b32e0
#2  0x00000000004231d3 in fu_device_list_item_free (item=0x6b5ff0) at ../src/fu-device-list.c:533
#3  0x00007ffff74055ab in g_ptr_array_foreach (array=0x65d420, func=0x423182 <fu_device_list_item_free>, user_data=0x0) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./glib/garray.c:1502
        i = 0
#4  0x00007ffff7405640 in ptr_array_free (array=0x65d420, flags=FREE_SEGMENT) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./glib/garray.c:1088
        rarray = 0x65d420
        segment = <optimized out>
#5  0x0000000000423343 in fu_device_list_finalize (obj=0x65d440 [FuDeviceList]) at ../src/fu-device-list.c:571
        self = 0x65d440 [FuDeviceList]
#6  0x00007ffff770cd9a in g_object_unref (_object=0x65d440) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./gobject/gobject.c:3183
        weak_locations = <optimized out>
        old_ref = 1
        object = 0x65d440 [FuDeviceList]
#7  0x000000000041e06a in fu_engine_finalize (obj=0x662820 [FuEngine]) at ../src/fu-engine.c:3441
        self = 0x662820 [FuEngine]
#8  0x00007ffff770cd9a in g_object_unref (_object=0x662820) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./gobject/gobject.c:3183
        weak_locations = <optimized out>
        old_ref = 1
        object = 0x662820 [FuEngine]
#9  0x000000000040e487 in fu_util_private_free (priv=0x65ca50) at ../src/fu-tool.c:206
#10 0x000000000040e52f in glib_autoptr_cleanup_FuUtilPrivate (_ptr=0x7fffffffe480) at ../src/fu-tool.c:220
#11 0x00000000004101af in main (argc=2, argv=0x7fffffffe6f8) at ../src/fu-tool.c:690
        allow_older = 0
        allow_reinstall = 0
        force = 0
        ret = 1
        verbose = 0
        plugin_glob = 0x660930
        priv = 0x65ca50
        error = 0x0
        cmd_descriptions = 0x65fee0 "hOK\366\377\177"
        options = 
            {{long_name = 0x42f0b8 "verbose", short_name = 118 'v', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7fffffffe46c, description = 0x42f0c0 "Show extra debugging information", arg_description = 0x0}, {long_name = 0x42f0e1 "allow-reinstall", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7fffffffe464, description = 0x42f0f8 "Allow re-installing existing firmware versions", arg_description = 0x0}, {long_name = 0x42f127 "allow-older", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7fffffffe460, description = 0x42f138 "Allow downgrading firmware versions", arg_description = 0x0}, {long_name = 0x42f15c "force", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x7fffffffe468, description = 0x42f162 "Override plugin warning", arg_description = 0x0}, {long_name = 0x42f17a "show-all-devices", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x65ca84, description = 0x42f190 "Show devices that are not updatable", arg_description = 0x0}, {long_name = 0x42f1b4 "plugin-whitelist", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_STRING_ARRAY, arg_data = 0x7fffffffe478, description = 0x42f1c8 "Manually whitelist specific plugins", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
```

superm1, 2018-06-06:
Here is specifically where it's failing.  Presumably that jumps to an invalid address for the finalize method.
https://github.com/GNOME/glib/blob/glib-2-48/gobject/gobject.c#L3183

hughsie, 2018-06-06:
I think it's failing as something is unref-ing the FuDevice so the refcount drops to zero before the FuDeviceList gets finalized. These bugs are tricky to track down. I'll add some code to detect this and it should start exploding when the refcount drops to zero, rather than when we tear down the device list.

hughsie, 2018-06-06:
Can you try with wip/hughsie/FuDeviceList-weak-tests please -- it should issue a critical warning -- I need the backtrace when it "crashes". Thanks.

superm1, 2018-06-06:
Unfortunately it doesn't seem to issue a critical warning for that.  Same as previous.

```
root@44c25e484222:~/fwupd# gdb --args /usr/local/libexec/fwupd/fwupdtool -v get-devices --plugin-whitelist=unifying
GNU gdb (Ubuntu 7.11.1-0ubuntu1~16.5) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/local/libexec/fwupd/fwupdtool...done.
(gdb) run 
Starting program: /usr/local/libexec/fwupd/fwupdtool -v get-devices --plugin-whitelist=unifying
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7ffff0d72700 (LWP 7114)]
(fwupdtool:7110): As-DEBUG: run FuEngine:load
(fwupdtool:7110): Fu-DEBUG: loading config values from /usr/local/etc/fwupd/daemon.conf
(fwupdtool:7110): Fwupd-DEBUG: looking for os-release at /etc/os-release
(fwupdtool:7110): Fu-DEBUG: using config path of /usr/local/etc/fwupd
(fwupdtool:7110): Fu-DEBUG: loading config from /usr/local/etc/fwupd/remotes.d/fwupd.conf
(fwupdtool:7110): Fu-DEBUG: loading config from /usr/local/etc/fwupd/remotes.d/vendor.conf
(fwupdtool:7110): As-DEBUG: run FuEngine:load-smbios
(fwupdtool:7110): As-DEBUG: run FuEngine:load-hwids
(fwupdtool:7110): Fu-DEBUG: smbios property Manufacturer=Dell Inc.
(fwupdtool:7110): Fu-DEBUG: smbios property EnclosureKind=a
(fwupdtool:7110): Fu-DEBUG: smbios property Family=XPS
(fwupdtool:7110): Fu-DEBUG: smbios property ProductName=XPS 13 9370
(fwupdtool:7110): Fu-DEBUG: smbios property ProductSku=07E6
(fwupdtool:7110): Fu-DEBUG: smbios property BiosVendor=Dell Inc.
(fwupdtool:7110): Fu-DEBUG: smbios property BiosVersion=1.3.3
(fwupdtool:7110): Fu-DEBUG: smbios property BiosMajorRelease=01
(fwupdtool:7110): Fu-DEBUG: smbios property BiosMinorRelease=03
(fwupdtool:7110): Fu-DEBUG: smbios property BaseboardManufacturer=Dell Inc.
(fwupdtool:7110): Fu-DEBUG: smbios property BaseboardProduct=0173S1
(fwupdtool:7110): Fu-DEBUG: {b21b8330-56dc-5a64-a05f-29dacfee56c5}   <- Dell Inc.&XPS&XPS 13 9370&7E6&Dell Inc.&1.3.3&01&03
(fwupdtool:7110): Fu-DEBUG: {22c2d18e-fd20-5723-b1ef-92d78fc2de16}   <- Dell Inc.&XPS&XPS 13 9370&Dell Inc.&1.3.3&01&03
(fwupdtool:7110): Fu-DEBUG: {77edbe33-36e6-50d9-8e2f-8d9b8ad135e4}   <- Dell Inc.&XPS 13 9370&Dell Inc.&1.3.3&01&03
(fwupdtool:7110): Fu-DEBUG: {6cb1f0d4-054e-5d7c-a8c8-5358010d9f62}   <- Dell Inc.&XPS&XPS 13 9370&7E6&Dell Inc.&173S1
(fwupdtool:7110): Fu-DEBUG: {d72b8452-80b6-5b66-b95c-8dd75519a30c}   <- Dell Inc.&XPS&XPS 13 9370&7E6
(fwupdtool:7110): Fu-DEBUG: {184da599-9986-5212-9dba-226d2d233133}   <- Dell Inc.&XPS&XPS 13 9370
(fwupdtool:7110): Fu-DEBUG: {3701616e-7261-579b-8b9c-ec94e598b8dd}   <- Dell Inc.&7E6&Dell Inc.&173S1
(fwupdtool:7110): Fu-DEBUG: {74bb5080-7e51-5373-aea1-bfc218a2caaf}   <- Dell Inc.&7E6
(fwupdtool:7110): Fu-DEBUG: {b84621e1-d31e-5350-af5e-f6be8e73ce0f}   <- Dell Inc.&XPS 13 9370&Dell Inc.&173S1
(fwupdtool:7110): Fu-DEBUG: {f33883b9-e996-539e-b35a-73568b903d13}   <- Dell Inc.&XPS 13 9370
(fwupdtool:7110): Fu-DEBUG: {ab9ada1b-9939-56c8-b8a0-47edb7ee0e63}   <- Dell Inc.&XPS&Dell Inc.&173S1
(fwupdtool:7110): Fu-DEBUG: {61178075-a8fd-563c-9045-44227d8c121f}   <- Dell Inc.&XPS
(fwupdtool:7110): Fu-DEBUG: {5c163113-9296-5c8d-a92d-ae04eb59a0f7}   <- Dell Inc.&a
(fwupdtool:7110): Fu-DEBUG: {dd46addd-ee15-5e81-b63f-08f09898c622}   <- Dell Inc.&Dell Inc.&173S1
(fwupdtool:7110): Fu-DEBUG: {85d38fda-fc0e-5c6f-808f-076984ae7978}   <- Dell Inc.
(fwupdtool:7110): As-DEBUG: run FuEngine:load-quirks
(fwupdtool:7110): Fu-DEBUG: loading quirks from /usr/local/share/fwupd/quirks.d/colorhug.quirk
(fwupdtool:7110): Fu-DEBUG: now 17 quirk entries
(fwupdtool:7110): Fu-DEBUG: loading quirks from /usr/local/share/fwupd/quirks.d/csr-aiaiai-h05.quirk
(fwupdtool:7110): Fu-DEBUG: now 23 quirk entries
(fwupdtool:7110): Fu-DEBUG: loading quirks from /usr/local/share/fwupd/quirks.d/csr-aiaiai-h60.quirk
(fwupdtool:7110): Fu-DEBUG: now 28 quirk entries
(fwupdtool:7110): Fu-DEBUG: loading quirks from /usr/local/share/fwupd/quirks.d/dfu.quirk
(fwupdtool:7110): Fu-DEBUG: now 116 quirk entries
(fwupdtool:7110): Fu-DEBUG: loading quirks from /usr/local/share/fwupd/quirks.d/ebitdo.quirk
(fwupdtool:7110): Fu-DEBUG: now 135 quirk entries
(fwupdtool:7110): Fu-DEBUG: loading quirks from /usr/local/share/fwupd/quirks.d/wac-intuos.quirk
[... 136 lines cut ...]
#6  0x00007ffff770cd9a in g_object_unref (_object=0x65d380) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./gobject/gobject.c:3183
        weak_locations = <optimized out>
        old_ref = 1
        object = 0x65d380 [FuDeviceList]
#7  0x000000000041e10a in fu_engine_finalize (obj=0x663020 [FuEngine]) at ../src/fu-engine.c:3441
        self = 0x663020 [FuEngine]
#8  0x00007ffff770cd9a in g_object_unref (_object=0x663020) at /build/glib2.0-prJhLS/glib2.0-2.48.2/./gobject/gobject.c:3183
        weak_locations = <optimized out>
[... cut ...] s = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x65ca84, description = 0x42f330 "Show devices that are not updatable", arg_description = 0x0}, {long_name = 0x42f354 "plugin-whitelist", short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_STRING_ARRAY, arg_data = 0x7fffffffe458, description = 0x42f368 "Manually whitelist specific plugins", arg_description = 0x0}, {long_name = 0x0, short_name = 0 '\000', flags = 0, arg = G_OPTION_ARG_NONE, arg_data = 0x0, description = 0x0, arg_description = 0x0}}
```

superm1, 2018-06-06:
To make reproducing this easier on your side I've done this whole set in a docker container.  It reliably reproduces with unifying for me.  You can pull this docker container and run with the instructions on the description.
https://hub.docker.com/r/superm1/fwupd-issue-test/

hughsie, 2018-06-06:
I can reproduce here, thanks for the container. Debugging...

hughsie, 2018-06-06:
If I recompile the sources (with no changes!) using the meson and ninja instructions in "Full Description" then the `/usr/local/libexec/fwupd/fwupdtool -v get-devices --plugin-whitelist=unifying` completes without any failures. :confused:

superm1, 2018-06-06:
:/

Looks like that's happening to me too.  The fwupdtool binary changes entry points and offsets too (by about 2k in diff).  I'm not sure why that would be.

superm1, 2018-06-06:
This may sound crazy but If you run this before the build it will reproduce again on newly built binaries:
`# apt purge libc6-dbg`

I snappshotted the container after installing `valgrind` (which pulled in `libc6-dbg`) but before running another rebuild.

superm1, 2018-06-06:
What's more - after removing `libc6-dbg` outside the container (such as in Ubuntu 18.04) I reproduce it there too now.

hughsie, 2018-06-06:
... stutters audibly

superm1, 2018-06-06:
At the fix I pushed or at how to reproduce it? :)

commit 2b593378 (2018-06-06): Clean up plugins as the last step of engine cleanup (Closes: #550)

Fixes a segfault that occurs during cleanup of USB plugins.

When g_module_close was called memory allocated by the plugin would
get freed leading to the finalize method for object class pointing
to garbage.
commit 3f9a1c18 (2018-06-06): Clean up plugins as the last step of engine cleanup (Closes: #550)

Fixes a segfault that occurs during cleanup of USB plugins.

When g_module_close was called memory allocated by the plugin would
get freed leading to the finalize method for object class pointing
to garbage.
closed 2018-06-06 by superm1