fwupd/fwupd #554: Get fwupd snap to not need devmode

superm1, 2018-06-07:
Per discussion at https://forum.snapcraft.io/t/approval-for-fwupd-classic-snap/5755/4 fwupd will not be approved as a classic snap and needs to be done as a properly confined snap.

Currently it's publishing to the edge channel in devmode.

Getting it to be a confined snap will require changes in `snapd` for a new interface (or changes to the existing fwupd interface in `snapd` used by [uefi-fw-tools](https://snapcraft.io/uefi-fw-tools)).

As a sample run of the types of things that need confinement rules:
```
[27552.796679] audit: type=1400 audit(1528331087.341:6754): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/home/supermario/src/fwupd/" pid=18327 comm="bash" requested_mask="r" denied_mask="r" fsuid=1000 ouid=1000
[27560.197817] audit: type=1400 audit(1528331094.752:6755): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/dmi/tables/smbios_entry_point" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27560.197820] audit: type=1400 audit(1528331094.752:6756): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/dmi/tables/DMI" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27560.199105] audit: type=1326 audit(1528331094.752:6757): auid=1000 uid=0 gid=0 ses=3 pid=18344 comm="fwupdtool" exe="/snap/fwupd/16/usr/lib/fwupd/fwupdtool" sig=0 arch=c000003e syscall=41 compat=0 ip=0x7fba7e0c75a7 code=0x7ffc0000
[27560.199108] audit: type=1400 audit(1528331094.752:6758): apparmor="ALLOWED" operation="create" profile="snap.fwupd.fwupdtool" pid=18344 comm="fwupdtool" family="netlink" sock_type="raw" protocol=15 requested_mask="create" denied_mask="create"
[27560.199110] audit: type=1400 audit(1528331094.752:6759): apparmor="ALLOWED" operation="setsockopt" profile="snap.fwupd.fwupdtool" pid=18344 comm="fwupdtool" family="netlink" sock_type="raw" protocol=15 requested_mask="setopt" denied_mask="setopt"
[27560.199207] audit: type=1326 audit(1528331094.752:6760): auid=1000 uid=0 gid=0 ses=3 pid=18344 comm="fwupdtool" exe="/snap/fwupd/16/usr/lib/fwupd/fwupdtool" sig=0 arch=c000003e syscall=49 compat=0 ip=0x7fba7e0c70f7 code=0x7ffc0000
[27560.199210] audit: type=1400 audit(1528331094.752:6761): apparmor="ALLOWED" operation="bind" profile="snap.fwupd.fwupdtool" pid=18344 comm="fwupdtool" family="netlink" sock_type="raw" protocol=15 requested_mask="bind" denied_mask="bind"
[27560.199213] audit: type=1400 audit(1528331094.752:6762): apparmor="ALLOWED" operation="getsockname" profile="snap.fwupd.fwupdtool" pid=18344 comm="fwupdtool" family="netlink" sock_type="raw" protocol=15 requested_mask="getattr" denied_mask="getattr"
[27560.199215] audit: type=1400 audit(1528331094.752:6763): apparmor="ALLOWED" operation="setsockopt" profile="snap.fwupd.fwupdtool" pid=18344 comm="fwupdtool" family="netlink" sock_type="raw" protocol=15 requested_mask="setopt" denied_mask="setopt"
[27560.199408] audit: type=1400 audit(1528331094.752:6764): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/bus/usb/devices/" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407397] kauditd_printk_skb: 152 callbacks suppressed
[27565.407398] audit: type=1400 audit(1528331099.964:6917): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/dev/drm_dp_aux0" pid=18344 comm="fwupdtool" requested_mask="wr" denied_mask="wr" fsuid=0 ouid=0
[27565.407775] audit: type=1400 audit(1528331099.964:6918): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407787] audit: type=1400 audit(1528331099.964:6919): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407794] audit: type=1400 audit(1528331099.964:6920): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/fw_class" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407802] audit: type=1400 audit(1528331099.964:6921): apparmor="ALLOWED" operation="capable" profile="snap.fwupd.fwupdtool" pid=18344 comm="fwupdtool" capability=21  capname="sys_admin"
[27565.407818] audit: type=1400 audit(1528331099.964:6922): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/fw_type" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407829] audit: type=1400 audit(1528331099.964:6923): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/fw_version" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407839] audit: type=1400 audit(1528331099.964:6924): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/capsule_flags" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407850] audit: type=1400 audit(1528331099.964:6925): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/last_attempt_status" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
[27565.407860] audit: type=1400 audit(1528331099.964:6926): apparmor="ALLOWED" operation="open" profile="snap.fwupd.fwupdtool" name="/sys/firmware/efi/esrt/entries/entry0/last_attempt_version" pid=18344 comm="fwupdtool" requested_mask="r" denied_mask="r" fsuid=0 ouid=0
```
I'm really not sure how to proceed with this.  I can't even get the previous fwupd interface used by uefi-fw-tools to connect.
```
$ snap connect fwupd:fwupdtool
error: snap "core" has no "fwupd" interface slots
```

I'm hoping that @timchen119 can assist with this.

timchen119, 2018-06-07:
@superm1 For connecting interface, you have to define slots and plugs in your snapcraft.yaml in order to use the fwupd interface as it's not a implicit interface exposed from the core snap. something like [this](https://git.launchpad.net/~lyoncore-team/lyoncore-snaps/+git/fwupd-snapcraft/tree/snapcraft.yaml#n25) in the snapcraft.yaml should work.

On the other hand, I'm not sure how things are going to work without declaring classic confinement on Ubuntu Classic (e.g. Desktop or Server), as /boot/efi is not bind mounted under confinement on Ubuntu Classic [currently](https://github.com/snapcore/snapd/blob/master/cmd/snap-confine/mount-support.c#L610).

hughsie, 2018-06-07:
> needs to be done as a properly confined snap

That's going to be slightly hard to do unless you can write to the actual ESP somehow.

superm1, 2018-06-07:
@timchen119 is checking with the snap folks to see what they advise for this scenario. Maybe the solution will be an interface to turn on bind mount to ESP, or maybe they'll approve classic after all because of this.

For now we should tell people doing UEFI update to use the non snapped version until there is a plan. The snapped version will support all the other plugins though.

superm1, 2018-06-11:
I've got some work that starts to bring daemon and real fwupdmgr into the snap too I've placed here.
https://github.com/hughsie/fwupd/tree/wip/superm1/snapped-daemon

Some functions work, but some need some fixes (such as GPG crypto engine isn't starting up properly right now).  When I've got all functions in place working in devmode I think we can better evaluate all the remaining interfaces that need to be in specialize confinement rules for snaps.

@timchen119 what is the deal with policy kit and snaps?  How do you register the daemon started with snap with system policy kit?

timchen119, 2018-06-12:
@superm1 The fwupd daemon should started by systemd with root permission automatically while booting. Is it now started on demand by regular user? Not sure if this topic related to your question: https://forum.snapcraft.io/t/should-we-use-polkit-for-local-auth/1206

superm1, 2018-06-12:
I guess I was originally assuming a policy kit problem, but maybe not.
I've got the daemon and client talking over dbus after doing a connect between plug and slot 
```
# snap connect fwupd:client fwupd:daemon
```

And sometimes I've had good behavior with it working, other times i've seen a dbus rejected error.  I don't know why sometimes it's worked.

The other thing i'm trying to figure out is what is making GPG fail in the snap.
```
$ fwupd.fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading?           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Failed to update metadata for lvfs: failed to import /root/snap/fwupd/x2/etc/pki/fwupd-metadata/GPG-KEY-Linux-Foundation-Metadata: Invalid crypto engine
```

```
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>  gpgme_debug: level=9
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>  gpgme_check_version: call: 0=(nil), req_version=(null), VERSION=1.6.0
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>  gpgme_check_version_internal: call: 0=(nil), req_version=(null), offset_sig_validity=60
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>  gpgme_new: enter: r_ctx=0xa35178
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>  gpgme-dinfo: gpgconf='/snap/fwupd/x2/usr/bin/gpgconf'
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_pipe: enter: filedes=0x7ffd2d8b46c0, inherit_idx=1 (GPGME uses it for reading)
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_pipe: leave: read=0x1c, write=0x1d
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_spawn: enter: path=0xa34ec0, path=/snap/fwupd/x2/usr/bin/gpgconf
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_spawn: check: path=0xa34ec0, argv[ 0] = /snap/fwupd/x2/usr/bin/gpgconf
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_spawn: check: path=0xa34ec0, argv[ 1] = --list-dirs
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_spawn: check: path=0xa34ec0, fd[0] = 0x1d -> 0x1
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_spawn: check: path=0xa34ec0, waiting for child process pid=1261
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x04ee>      gpgme:max_fds: call: 0=(nil), max fds=4096 (RLIMIT_NOFILE)
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>      _gpgme_io_close: enter: fd=0x1d
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>      _gpgme_io_close: leave: result=0
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_spawn: leave: result=0
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: enter: fd=0x1c, buffer=0x7ffd2d8b4720, count=1023
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 737973636f6e6664 69723a2f6574632f sysconfdir:/etc/
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 676e757067320a62 696e6469723a2f75 gnupg2.bindir:/u
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 73722f62696e0a6c 6962657865636469 sr/bin.libexecdi
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 723a2f7573722f6c 69622f676e757067 r:/usr/lib/gnupg
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 320a6c6962646972 3a2f7573722f6c69 2.libdir:/usr/li
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 622f7838365f3634 2d6c696e75782d67 b/x86_64-linux-g
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 6e752f676e757067 320a646174616469 nu/gnupg2.datadi
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 723a2f7573722f73 686172652f676e75 r:/usr/share/gnu
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 7067320a6c6f6361 6c656469723a2f75 pg2.localedir:/u
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 73722f7368617265 2f6c6f63616c650a sr/share/locale.
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 6469726d6e67722d 736f636b65743a2f dirmngr-socket:/
Jun 12 12:17:15 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:15 <0x0279>    _gpgme_io_read: check: 726f6f742f736e61 702f66777570642f root/snap/fwupd/
[... cut ...]
Jun 12 12:17:16 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:16 <0x0279>  engine.c:455: returning error: Invalid crypto engine
Jun 12 12:17:16 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:16 <0x0279>  gpgme_op_import:314: error: Invalid crypto engine <GPGME>
Jun 12 12:17:16 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:16 <0x0279>  gpgme_data_release: call: dh=0xa49400
Jun 12 12:17:16 supermario-XPS-13-9370 fwupd.fwupd[633]: GPGME 2018-06-12 12:17:16 <0x0279>  gpgme_release: call: ctx=0xa53dd0
```

timchen119, 2018-06-13:
@superm1 
For the policy on dbus issue the [fwupd interface code](https://github.com/snapcore/snapd/blob/master/interfaces/builtin/fwupd.go#L188) and apparmor errors in syslog may helps.

About the GPG issue, did you build on 18.04? Can you try adding `gnupg1` package into your stage packages in snapcraft.yaml as an experiment? The gnupg package changes to v2 as a default after Ubuntu 17.10 and the version update is the major difference between 16.04 and 18.04 I can think of. Not sure why gpgme needs both but I remembered I had issues on gpgconf and gpgme with gnupg2 when creating snap for core so I ended up include gnupg1 in the snap.

superm1, 2018-06-14:
@timchen119 Thanks for the pointers.  It turns out it was gpgconf.  gpgconf in 16.04 is compiled to look specifically for /usr/bin/gpg2 but that's not in place.  I did find out that if gpgconf isn't around it falls back to /usr/bin/gpg though.  So I did this and it's now working to refresh metadata.

```
$ fwupd.fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading?           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
```

Next problem is that policykit isn't registering.
```
# fwupd.fwupdmgr install ~/Downloads/firmware.cab --allow-reinstall 
Decompressing?         [***************************************]
Authenticating?        [***************************************]
Could not check for auth: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: Action org.freedesktop.fwupd.update-internal is not registered
```

As I can tell the system's policykit knows nothing of snaps.  So it seems like the only way to resolve it is to put the policykit file in the real `/usr/share/polkit-1/actions/` to me which is what the snapcraft snap seems to have done.

superm1, 2018-07-10:
I've rebased my branch onto master so it is now building in the fwupdate EFI application and the daemon.  The remaining problems need to be fixed in snapd or in the store.

- [ ] Policykit being loaded into host (allowing `fwupdmgr` to communicate with `fwupd`)
- [ ] Bind mounting of EFI system partition directory
- [ ] Adjusting all confinement rules to cover fwupd's usage
- [ ] Bind mounting of ESRT variables
- [ ] Automatically connecting the client plug to the daemon slot

superm1, 2018-09-12:
The snap has been converted to a classic snap until that list of remaining problems can be fixed in a released version of snapd.

The following commits will need to be reverted to start working on confined/devmode snap:
https://github.com/hughsie/fwupd/commit/c60012c784f1edd6e344d3b12298957788559aca
https://github.com/hughsie/fwupd/commit/b91d3d9bff3e130c1cc01a1bfba23dd61cb20a19
https://github.com/hughsie/fwupd/commit/25b49b79432782417d6522460e201986460ab420
https://github.com/hughsie/fwupd/commit/2da1aca6e38bd3ea2f781f93f5da3c9f107953e2
https://github.com/hughsie/fwupd/commit/05cc3f59563440174b41825a33fe69be758e9764
https://github.com/hughsie/fwupd/commit/3514a5ea84bdf01cb4e1e612023ba92fe8d70423

superm1, 2018-12-06:
the branch wip/superm1/onward-to-confined-snap has those reverted (they weren't simple reverts due to other changes needed).

hughsie, 2019-01-08:
> if you can think of any others missing feel free to chime in

Does `/sys/class/dmi` count? `/sys/class/tpm` as well?

superm1, 2019-01-08:
It looks like `/sys/class/dmi` has a few of them covered already but not all.
https://github.com/snapcore/snapd/blob/38e8503ef2e10bedf2372143e88a7da9022e2e1e/interfaces/builtin/fwupd.go#L61

So we would need to expand that to all of `/sys/firmare/dmi`  as well (since the whole table is now parsed)

TPM isn't yet covered, so yes that's another that is needed (particularly for TPM1.2)

superm1, 2019-01-08:
currently the snap uses `/boot/efi/EFI/ubuntu` for the EFI executable, but maybe that should actually be compiled as `/boot/efi/EFI/snap` or something like that too.

superm1, 2020-07-02:
I'm going to close this one, if there is interest in a confined snap, we'll need to see the work happen first in snapd.  Fwupd will be minor changed compared to that.

closed 2020-07-02 by superm1

superm1, 2020-07-21:
@woodrow-shen I saw your PR from here: https://github.com/snapcore/snapd/pull/8851 and was wondering are you working on bringing up the rest of the interfaces for fwupd to be able to convert to confined snap?  If so you should see https://github.com/fwupd/fwupd/issues/554#issuecomment-452382264 above for several other things that are needed for HW interaction on updates.

woodrow-shen, 2020-07-31:
@superm1 yes, I'm working on uefi-fw-tools to port the newer fwupd version based on 1.4.x branch, but my first goal is to make sure capsule updating can be smoothly done, so that's why I sent the PR which emphasize the fwupd interface to be strict to upstream. I know there are more HW components that need to be confined with fwupd, and let's keep these in my mind and solve them step-by-step. Also, currently, either ubuntu core or fwupd snap still has some limitations to be figured out:

- the confinement doesn't allow fwupd to access the raw disks
- /etc/os-release for ubuntu core is not "ubuntu" as ID
- snapd didn't implement policy kit as backend
so on.

Thus, the uefi-fw-tools still carries some patches from efivar and fwupd, and I don't know when we can fight against above issues, so this snap will move forward. Btw, the newer uefi-fw-tools based on 1.4 track just released to edge)