fwupd/fwupd #555: Flatpak should have access to home or xdg download

superm1, 2018-06-08:
In my experience with testing the flatpak I was finding that it's awkward to install cab files (or flat blobs for that matter) since they need to go into `$HOME/.var/org.freedesktop.fwupd` to be recognized by sandbox confinement policies.

I think some better directories should be chosen by default and error messages clarified if not using appropriate directories for the sandbox.

Since the normal model is to download from LVFS manually and then install my two thoughts were either `--filesystem=xdg-download` or `--filesystem=home`.

And then also adjust the error message for file not found specifically if running in sandbox mode (maybe `-Daemon=false` is enough).

hughsie, 2018-06-08:
Out of interest, how were you running fwupdtool from the flatpak bundle? I always just used sudo.

hughsie, 2018-06-08:
Cut sure, good idea to have `--filesystem=xdg-download`

superm1, 2018-06-08:
> I always just used sudo.

As I can tell that doesn't affect sandbox. I was trying with files in /tmp and unable to access them.

commit f8cb836d (2018-06-08): flatpak: Enable bind mount for xdg-download-dir (Closes: #555)

This makes it easier to install CAB files downloaded from LVFS
via your web browser.
closed 2018-06-08 by superm1