fwupd/fwupd #563: Updates for Dell XPS 13 9370 are not applied

arnomoonens, 2018-06-16:
I want to update the firmware of my Dell XPS 13 9370 to firmware version 0.1.4.0. However, after running `fwupdmgr update` and restarting, the updates are not applied (as seen by running `fwupdmgr get-devices`).

```shell
$ fwupdmgr get-devices
XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.2.1
  VersionLowest:        0.1.2.1
  Icon:                 computer
  Created:              2018-06-16

XPS 9370 Thunderbolt Controller
  DeviceId:             f3417bbb1147f2d88e4432a9f009e387a32e3bcb
  Guid:                 4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              28.00
  Icon:                 computer
  Created:              2018-06-16
```

```shell
$ efibootmgr -v
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002
Boot0000* Windows Boot Manager  HD(1,GPT,6275b35b-7a16-4bdf-a80d-ca3993a4550b,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...o................
Boot0002* Manjaro       PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-2C-29-1F)/HD(1,GPT,6275b35b-7a16-4bdf-a80d-ca3993a4550b,0x800,0x145000)/File(\EFI\Manjaro\grubx64.efi)
Boot0004* ubuntu        HD(1,GPT,6275b35b-7a16-4bdf-a80d-ca3993a4550b,0x800,0x145000)/File(\EFI\ubuntu\shimx64.efi)
```

```shell
$ efivar -l | grep fw

```

```shell
$ tree /boot
zsh: command not found: tree
```

- Operating system and version: Manjaro Linux 17.1.10 Hakoila (Kernel: x86_64 Linux 4.17.0-2-MANJARO)
- How did you install fwupd: `pacman`
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No

superm1, 2018-06-16:
Did you follow the arch directions to load bootloader to efi system partition?
Please share output of your ESP tree. It looks like you need to install tree.

https://github.com/hughsie/fwupd/wiki/LVFS-Triaged-Issue:-Arch:-Invalid-ESP-location

ArchangeGabriel, 2018-06-18:
You did not follow the instructions linked above by @superm1 for setting up fwupd: there is no fwupdate executable in you ESP.

ArchangeGabriel, 2018-06-18:
Sorry, the instructions you should follow are: https://wiki.archlinux.org/index.php/Fwupd#Installing_and_updating_fwupdate

arnomoonens, 2018-06-18:
I had already followed those instructions.
However, I just removed and installed `fwupd` and followed the instructions again and now it works. Thanks!

closed 2018-06-18 by arnomoonens