fwupd/fwupd #565: fwupd 1.0.8 fails to update 8bitdo SFC30 gamepad

Shamus397, 2018-06-19:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.0.8
daemon version: 1.0.8
compile-time dependency versions
        appstream-glib: 0.7.5
        gusb:   0.2.9
```

```shell
$ fwupdmgr get-devices
SFC30 Joystick
  DeviceId:             4efc400ea651f75299a787dbac70171613eee4cf
  Guid:                 f94d3231-f6e1-5ef3-a4a0-dc819d74ae54
  Guid:                 a52a1cc4-31ec-5191-8057-6743fcb23033
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x1235
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-19
```

Please answer the following questions:

- Operating system and version: Funtoo 1.2
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): portage (i.e., emerge fwupd)
- Have you tried rebooting? Yes (though for the life of me I can't figure out how this would help)
- Are you using an NVMe disk? No

This file shows what happened when I tried to update using fwupdmgr:
[8bitdo-update.txt](https://github.com/hughsie/fwupd/files/2113464/8bitdo-update.txt)

When I plugged the gamepad back in after unplugging it and turning it back on following the directions (unplug, turn off pad, hold L+R+Start for 3 seconds, wait for flashing lights, plug back in), nothing happened.  After attempting to do fwupdmgr update again, it did nothing; it seems that after putting the gamepad into firmware update mode that fwupd no longer recognized the gamepad. Further fwupd ignores the gamepad completely even after you unplug it, turn it off, and replug it back in (the only way to get fwupd to see it again is to do fwupdmgr clear-history after removing the gamepad and turning it off).

Here's the info from lsusb -v after plugging the gamepad in in firmware update mode (L+R+Start):
[8bitdo-usb.txt](https://github.com/hughsie/fwupd/files/2113465/8bitdo-usb.txt)

And here's the output from the fwupd daemon while trying all of the above.
[8bitdo-daemon.txt](https://github.com/hughsie/fwupd/files/2113463/8bitdo-daemon.txt)

hughsie, 2018-06-19:
What's Funtoo 1.2? Does the controller update if you use something like Fedora or Ubuntu?

hughsie, 2018-06-19:
> are you saying that fwupd requires one of those distributions in order to work properly?

No, I'm saying it's not a distro we test (https://travis-ci.org/hughsie/fwupd), and there could be thousand of things incorrectly configured on your system. Could you perhaps try the flatpak or snap bundle instead?

Shamus397, 2018-06-19:
"Thousands of things incorrectly configured on your system" seems a bit hyperbolic, and isn't very helpful. Nevertheless, I tried the flatpak (which I had to do some digging to find, here: https://github.com/hughsie/fwupd/wiki/fwupd-flatpak). It took a bit of effort to find out where to stuff the firmware cab for updating, since on my system, $XDG_DOWNLOAD_DIR is undefined and fwupd run thru flatpak couldn't see it (eventually, I found it could be discovered by the flatpak filesystem at /app by shoving the cab into /var/lib/flatpak/app/org.freedesktop.fwupd/current/active/files/).

So, at the end of the day, we have this (after putting the gamepad into firmware update mode):
```shell
# flatpak run org.freedesktop.fwupd get-devices
Fu-Message: 17:14:45.727: disabling plugin because: failed to startup uefi: UEFI firmware updating not supported
Fu-Message: 17:14:45.727: disabling plugin because: failed to coldplug amt: Unable to find a ME interface
Fu-Message: 17:14:45.729: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported, missing kernel support.
Fu-Message: 17:14:45.729: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
Fu-Message: 17:14:45.729: using plugins: colorhug, udev, steelseries, nitrokey, csr, thunderbolt, wacomhid, unifying, upower, dell, altos, dfu, ebitdo
8Bitdo  
  DeviceId:             5ec44b20be17233754d2a925e8fa8ee7ac91aef9
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 ba7f0ecd-1972-5bc0-b474-ffdc5fa17645
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x0483
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-19

# flatpak run org.freedesktop.fwupd install /app/8845f76a47170912c58875be458dfb3e2b15f62d-8Bitdo-SFC30_NES30_SFC30_SNES30-4.00.cab 
Fu-Message: 17:00:11.578: disabling plugin because: failed to startup uefi: UEFI firmware updating not supported
Fu-Message: 17:00:11.578: disabling plugin because: failed to coldplug amt: Unable to find a ME interface
Fu-Message: 17:00:11.580: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported, missing kernel support.
Fu-Message: 17:00:11.580: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
Fu-Message: 17:00:11.580: using plugins: colorhug, udev, steelseries, nitrokey, csr, thunderbolt, wacomhid, unifying, upower, dell, altos, dfu, ebitdo
Decompressing…         [***************************************]
No supported devices found
```
This is _identical_ to what happens when I run the non-flatpak version of fwuptool under the same circumstances (gamepad in firmware update mode).

hughsie, 2018-06-20:
The bootloader VID/PID is shared between all 8bitdo hardware, so you have to connect the controller in non-bootloader (runtime) mode, disconnect it, and then switch it to bootloader mode, then reconnect it. That way you'll have more GUIDs in the device (4 rather than 2) and it should match the firmware file.

Shamus397, 2018-06-20:
That doesn't work. To wit:

Put the gamepad into normal mode, connect it to the cable:
```shell
# flatpak run org.freedesktop.fwupd get-devices
Fu-Message: 12:39:31.715: disabling plugin because: failed to startup uefi: UEFI firmware updating not supported
Fu-Message: 12:39:31.715: disabling plugin because: failed to coldplug amt: Unable to find a ME interface
Fu-Message: 12:39:31.717: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported, missing kernel support.
Fu-Message: 12:39:31.717: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
Fu-Message: 12:39:31.717: using plugins: colorhug, udev, steelseries, nitrokey, csr, thunderbolt, wacomhid, unifying, upower, dell, altos, dfu, ebitdo
SFC30 Joystick
  DeviceId:             d10a323a7bac4b692835cb9f038f553535e36c26
  Guid:                 f94d3231-f6e1-5ef3-a4a0-dc819d74ae54
  Guid:                 a52a1cc4-31ec-5191-8057-6743fcb23033
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x1235
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-20
```

Disconnect gamepad from the cable, put it into "bootloader" mode, then reconnect it:
```shell
# flatpak run org.freedesktop.fwupd get-devices
Fu-Message: 12:40:48.958: disabling plugin because: failed to startup uefi: UEFI firmware updating not supported
Fu-Message: 12:40:48.958: disabling plugin because: failed to coldplug amt: Unable to find a ME interface
Fu-Message: 12:40:48.960: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported, missing kernel support.
Fu-Message: 12:40:48.960: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
Fu-Message: 12:40:48.960: using plugins: colorhug, udev, steelseries, nitrokey, csr, thunderbolt, wacomhid, unifying, upower, dell, altos, dfu, ebitdo
8Bitdo  
  DeviceId:             5ec44b20be17233754d2a925e8fa8ee7ac91aef9
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 ba7f0ecd-1972-5bc0-b474-ffdc5fa17645
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x0483
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-20
```

Attempt to update firmware:
```shell
flatpak run org.freedesktop.fwupd install /app/8845f76a47170912c58875be458dfb3e2b15f62d-8Bitdo-SFC30_NES30_SFC30_SNES30-4.00.cab 
Fu-Message: 12:41:01.124: disabling plugin because: failed to startup uefi: UEFI firmware updating not supported
Fu-Message: 12:41:01.124: disabling plugin because: failed to coldplug amt: Unable to find a ME interface
Fu-Message: 12:41:01.126: disabling plugin because: failed to coldplug synapticsmst: MST firmware updating not supported, missing kernel support.
Fu-Message: 12:41:01.126: disabling plugin because: failed to coldplug thunderbolt_power: missing kernel support for intel-wmi-thunderbolt
Fu-Message: 12:41:01.126: using plugins: colorhug, udev, steelseries, nitrokey, csr, thunderbolt, wacomhid, unifying, upower, dell, altos, dfu, ebitdo
Decompressing…         [***************************************]
No supported devices found
```

hughsie, 2018-06-20:
Right, fwupdtool probably won't work for this as it's not saving device state between invocations. I guess we should probably save this state somewhere when people are not using the daemon.

superm1, 2018-06-20:
In that case though the compiled daemon with funtoo, does the failure still happen if you follow that flow mentioned a few comments above?

Shamus397, 2018-06-20:
It fails exactly the same way with the daemon on Funtoo.

kylehawes, 2018-06-26:
I have this exact same problem on Ubuntu 18.04 with the SF30/SN30 Pro. After switching to bootloader mode fwupdmgr does not attempt to run the update, although it is still listed in get-devices.

hughsie, 2018-06-26:
Can somebody try git master please -- I think it's fixed now. If it's confirmed as fixed I'll cherry pick those commits back to the stable branch and prepare a release for next week.

Shamus397, 2018-06-26:
With commit 04d6965a146acbdbd5653c56f895bc30015905fb, no dice:
```shell
# fwupdmgr get-updates
SFC30 Joystick has firmware updates:
GUID:                    f94d3231-f6e1-5ef3-a4a0-dc819d74ae54
GUID:                    a52a1cc4-31ec-5191-8057-6743fcb23033
GUID:                    043c7673-bd45-5e2e-b8b9-c22df2dfec2a
GUID:                    b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
ID:                      com.8bitdo.sfc30.firmware
Update Version:          4.01
Update Name:             SFC30
Update Summary:          Firmware for the 8Bitdo SFC30 Game Controller
Update Remote ID:        lvfs
Update Checksum:         SHA1(78ef2663beaa952415c3719447b0d2ff43e837d8)
Update Location:         https://fwupd.org/downloads/fe066b57c69265f4cce8a999a5f8ab90d1c13b24-8Bitdo-SFC30_NES30_SFC30_SNES30-4.01.cab
Update Description:      Fixed input lag problem when used with other controllers.
ID:                      com.8bitdo.sfc30.firmware
Update Version:          4.00
Update Name:             SFC30
Update Summary:          Firmware for the 8Bitdo SFC30 Game Controller
Update Remote ID:        lvfs
Update Checksum:         SHA1(6c274a6f16a93488802671d2158a060c0e94d074)
Update Location:         https://fwupd.org/downloads/8845f76a47170912c58875be458dfb3e2b15f62d-8Bitdo-SFC30_NES30_SFC30_SNES30-4.00.cab
# fwupdmgr update
Downloading 4.01 for SFC30 Joystick...
Updating 4.01 on SFC30 Joystick...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Device SFC30 Joystick needs to manually be put in update mode: Unplug the controller, hold down L+R+START for 3 seconds until both LEDs are flashing then reconnect the controller.
# fwupdmgr update
# fwupdmgr get-updates
# fwupdmgr get-devices
8Bitdo  
  DeviceId:             2fcdf0313e61082c5fc990eed8a75c54843c380e
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 ba7f0ecd-1972-5bc0-b474-ffdc5fa17645
  Guid:                 f94d3231-f6e1-5ef3-a4a0-dc819d74ae54
  Guid:                 a52a1cc4-31ec-5191-8057-6743fcb23033
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x0483
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-26
```
And no love with fwupdtool (but this is expected):
```shell
# /usr/libexec/fwupd/fwupdtool get-devices
Fu-Message: using plugins: csr, ebitdo, altos, steelseries, upower, udev, amt, nitrokey, dfu, wacomhid, colorhug, unifying
SFC30 Joystick
  DeviceId:             33105ea6e4ea156d33cc4d1e2e90d76c7dd7448a
  Guid:                 f94d3231-f6e1-5ef3-a4a0-dc819d74ae54
  Guid:                 a52a1cc4-31ec-5191-8057-6743fcb23033
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x1235
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-26

# /usr/libexec/fwupd/fwupdtool install 8845f76a47170912c58875be458dfb3e2b15f62d-8Bitdo-SFC30_NES30_SFC30_SNES30-4.00.cab 
Fu-Message: using plugins: csr, ebitdo, altos, steelseries, upower, udev, amt, nitrokey, dfu, wacomhid, colorhug, unifying
Decompressing…         [***************************************]
Device SFC30 Joystick needs to manually be put in update mode: Unplug the controller, hold down L+R+START for 3 seconds until both LEDs are flashing then reconnect the controller.
# /usr/libexec/fwupd/fwupdtool install 8845f76a47170912c58875be458dfb3e2b15f62d-8Bitdo-SFC30_NES30_SFC30_SNES30-4.00.cab 
Fu-Message: using plugins: csr, ebitdo, altos, steelseries, upower, udev, amt, nitrokey, dfu, wacomhid, colorhug, unifying
Decompressing…         [***************************************]
No supported devices found
# /usr/libexec/fwupd/fwupdtool get-devices
Fu-Message: using plugins: csr, ebitdo, altos, steelseries, upower, udev, amt, nitrokey, dfu, wacomhid, colorhug, unifying
8Bitdo  
  DeviceId:             2fcdf0313e61082c5fc990eed8a75c54843c380e
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 ba7f0ecd-1972-5bc0-b474-ffdc5fa17645
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x0483
  Version:              2.70
  Icon:                 input-gaming
  Created:              2018-06-26
```
And yes, I followed all the proper steps in setting up the gamepad for firmware update.

Shamus397, 2018-06-27:
Just a simple question: why is this closed *before* being confirmed as working? And why can't I reopen it?

ricab, 2018-07-04:
Hello, I stumbled upon this while trying to work out the cause of [this issue](https://askubuntu.com/questions/1051536/appstreamcli-appstream-system-cache-was-updated-but-problems-were-found-metad?noredirect=1&lq=1), which I also recently started experiencing on ubuntu 16.04.

Running `/usr/lib/x86_64-linux-gnu/fwupd/fwupd --verbose` outputs the following at the end:

```
17:43:24:0460 Fu  FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: Error on line 265: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity - escape ampersand as &amp;
```

The corresponding line in `/var/cache/app-info/xmls/fwupd.xml`:
```
         <checksum filename="Firmware_SF30&SN30_Pro_V1.26.dat" target="content" type="sha1">3    265 ef2bdee8aca2a45b9f53b4d4cce9722523f57f8</checksum>
```

Which looks suspiciously related to SFC30 at this issue's title. And a google search for "Firmware_SF30" comes up with "8Bitdo" stuff...

Are the issues related? Is there a quick fix/workaround?

hughsie, 2018-07-04:
Ubuntu needs to backport this fix into 16.04: https://github.com/hughsie/appstream-glib/commit/6048520484101df5d33f3c852c10640e630d20cf

ricab, 2018-07-04:
I see, thanks for the clarification.

Shamus397, 2018-07-05:
I can't find either 4577907 or fa1bbd6 in the master branch. Will these fixes be getting merged to master any time soon? :-)

JusterK, 2018-07-05:
> The bootloader VID/PID is shared between all 8bitdo hardware, so you have to connect the controller in non-bootloader (runtime) mode, disconnect it, and then switch it to bootloader mode, then [...]

@hughsie For me it's the opposite. Non-bootloader mode gives me 4 GUIDs and bootloader mode only 2. Also after a few tries I get 5 GUIDs. This happens on fedora 28 with stock fwupd (1.0.8) and master which seems to have  4577907

` % LD_PRELOAD="/usr/local/lib64/libfwupd.so.2" fwupdmgr --version
client version: 1.1.0
daemon version: 1.1.0
checkout info:  1.0.8-104-gba10348
compile-time dependency versions
        appstream-glib: 0.7.9
        gusb:   0.3.0
        efivar: 35`

fmarier, 2018-07-08:
> Ubuntu needs to backport this fix into 16.04: hughsie/appstream-glib@6048520

I filed https://bugs.launchpad.net/ubuntu/+source/appstream-glib/+bug/1780520 and backported that fix to my PPA. It does solve the problem, thanks!

JusterK, 2018-07-08:
So I just gave it try again. Plug in the controller, fwupdmgr update, unplug, put controller in bootloader mode, replug, fwupdmgr update -> nothing. But when calling fwupdmgr install /path/to/file.cab it updated the firmware without complaints. I guess it was fixed after all, but a hint to actually call fwupdmgr install after replugging the controller would be nice. Anyway thanks for the fix!

robert-ancell, 2018-08-15:
@hughsie in https://launchpad.net/bugs/1780442 the reporter suggests this fix doesn't work until the cache is replaced - Is there something server side that can be done to get clients to redownload the cache or is another fix required?

hughsie, 2018-08-16:
I'm not sure what "until the cache is replaced" means; can you get the reporter to explain a bit more here? Thanks.

ricab, 2018-08-16:
The cache containing the un-escaped "&" is not updated when there is no newer version, so updating appstream-glib is not enough to fix the issue: the file has to be erased so that a new one is produced. I suppose the ubuntu package should do this in a post step.

ricab, 2018-08-16:
I tried it with `fwupdmgr refresh --force` and `appstreamcli refresh-cache --force`, but they failed because of the problematic ampersand. BTW, sorry that this bug got hijacked with this matter, I though it was related when I first commented :S

hughsie, 2018-08-16:
Can you tell me how `fwupdmgr refresh` failed please.

ricab, 2018-08-16:
It failed with a timeout. If I understand properly, I don't think there is much more to be done on your side. Perhaps there would be a command that could update the file without it being previously deleted, but it would still need to go into the package's post step...

hughsie, 2018-08-16:
I think 068d34307a6d779492c4b0898e76e8490f702e62 is the commit that switched fwupd to just ignore corrupt XML files, which corresponds to anything >= 1.0.0. I assume the version of fwupd you're using is 0.something? I could always backport that commit into an older branch if that's a better idea.

ricab, 2018-08-16:
I am not sure. I know that the issue can be fixed with the proposed package + delete and produce cache file. I don't know if there would be a better approach.

ricab, 2018-08-16:
Also, I confirm your suspicions on the version
```
$ fwupdmgr --version
client version:	0.8.3
daemon version:	0.8.3
```

ricab, 2018-08-16:
Perhaps @superm1 has an opinion, he opened https://launchpad.net/bugs/1780442 and provided the *partially fixing* appstream-glib package.

hughsie, 2018-08-16:
@robert-ancell I think fe1c4de5a4b178cae3e4e9325ea84ff85c4b1be3 is what you want to include with the fwupd in xenial (that commit is on the 0_8_X branch) -- yell if a tarball would make it easier.

superm1, 2018-08-16:
I think the right thing to do is backport that commit that @hughsie put in 0_8_X to the 0.8.3 fwupd package in 16.04 rather than postinst magic.  @ricab this should be verification-done on the appstream-glib package, but a new task is needed for fwupd in the same bug to handle that part of it.

superm1, 2018-09-27:
So anyone affected by this bug in Ubuntu-land, to avoid being blocked by Ubuntu's lack of an SRU with this fix there is now a snap available.
https://snapcraft.io/fwupd

Install that and you'll be able to upgrade succesfully.

closed 2018-09-27 by superm1

kalikiana, 2018-12-24:
I'm happy to report that OpenSUSE Tumbleweed-land also benefits from the snap'ed `fwupdatemgr` - note that `sudo ln -s /var/lib/ca-certificates/ca-bundle.pem /etc/ssl/certs/ca-certificates.crt` is required to make downloads work.