fwupd/fwupd #583: build of fwupd 1.0.8 fails with: Unknown variable "colorhug_pkcs7_signature"

maxim27, 2018-07-03:
I am trying to build fwupd 1.0.8, but it eventually fails with the following error:

`src/meson.build:22:2: ERROR:  Unknown variable "colorhug_pkcs7_signature"`

no other errors occured. I have no idea what is the problem here. Also google does not find anything usefull, besides this bug report: https://bugs.gentoo.org/659698

any ideas?

hughsie, 2018-07-03:
Is the gcab binary installed? We've fixed that recently on master.

maxim27, 2018-07-03:
yes gcab-1.1 is installed

hughsie, 2018-07-03:
Can you try with the 1_0_X branch please.

superm1, 2018-07-03:
Please share the full configuration line in use for meson and output too.

maxim27, 2018-07-04:
Ok i finaly managed to do the build with gcab-1.0 but I got the same error again.

Here is the full build log (created with open build service):
https://file.io/lrRLsw

superm1, 2018-07-04:
That link is 404 for me.

maxim27, 2018-07-04:
Sorry the serves i was using deleted the file after the first download, i have uploaded again: https://justpaste.it/4uy0y

superm1, 2018-07-10:
I did a test with the 1_0_X branch and your build command and confirm it's fixed now.
```
# /usr/bin/meson --buildtype=plain --prefix=/usr --libdir=/usr/lib64 --libexecdir=/usr/lib --bindir=/usr/bin --sbindir=/usr/sbin --includedir=/usr/include --datadir=/usr/share --mandir=/usr/share/man --infodir=/usr/share/info --localedir=/usr/share/locale --sysconfdir=/etc --localstatedir=/var --sharedstatedir=/var/lib --wrap-mode=nodownload . build -Dtests=false
The Meson build system
Version: 0.45.1
Source dir: /home/supermario/src/fwupd
Build dir: /home/supermario/src/fwupd/build
Build type: native build
Project name: fwupd
Native C compiler: cc (gcc 7.3.0 "cc (Ubuntu 7.3.0-16ubuntu3) 7.3.0")
Build machine cpu family: x86_64
Build machine cpu: x86_64
Program git found: YES (/usr/bin/git)
Compiler for C supports arguments -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Wincompatible-pointer-types-discards-qualifiers -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-error=cpp -Wno-unknown-pragmas -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wnull-dereference -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wwrite-strings: SOME
WARNING: -Wl,-z,relro looks like a linker argument, but has_argument
and other similar methods only support checking compiler arguments.
Using them to check linker arguments are never supported, and results
are likely to be wrong regardless of the compiler you are using.

Compiler for C supports argument -Wl,-z,relro: YES
WARNING: -Wl,-z,now looks like a linker argument, but has_argument
and other similar methods only support checking compiler arguments.
Using them to check linker arguments are never supported, and results
are likely to be wrong regardless of the compiler you are using.

Compiler for C supports argument -Wl,-z,now: YES
Found pkg-config: /usr/bin/pkg-config (0.29.1)
Native dependency gio-2.0 found: YES 2.56.1
Native dependency gmodule-2.0 found: YES 2.56.1
Native dependency gio-unix-2.0 found: YES 2.56.1
Native dependency gudev-1.0 found: YES 232
Native dependency appstream-glib found: YES 0.7.7
Native dependency gusb found: YES 0.2.11
Native dependency sqlite3 found: YES 3.22.0
Native dependency libarchive found: YES 3.2.2
Native dependency json-glib-1.0 found: YES 1.4.2
Native dependency valgrind found: YES 3.13.0
Native dependency libsoup-2.4 found: YES 2.62.1
Native dependency polkit-gobject-1 found: YES 0.105
Native dependency udev found: YES 237
Native dependency gnutls found: YES 3.5.18
Library gpgme found: YES
Library gpg-error found: YES
Library m found: YES
Native dependency uuid found: YES 2.31.1
Native dependency libgcab-1.0 found: YES 1.1
Program gcab found: YES (/usr/bin/gcab)
Native dependency bash-completion found: YES 2.8
Native dependency cairo found: YES 1.15.10
Library fontconfig found: YES
Library freetype found: YES
Native dependency libelf found: YES 0.170
Native dependency fwup found: YES 10
Native dependency efivar found: YES 34
Native dependency libsmbios_c found: YES 2.4.1
Dependency efivar found: YES (cached)
Dependency umockdev-1.0 found: NO
Native dependency systemd found: YES 237
Configuring config.h using configuration
Configuring fwupd.conf using configuration
Configuring vendor.conf using configuration
Configuring fwupdtool using configuration
Configuring org.freedesktop.fwupd.service using configuration
Configuring fwupd-offline-update.service using configuration
Configuring fwupd.service using configuration
Program g-ir-scanner found: YES (/usr/bin/g-ir-scanner)
Configuring fwupd-version.h using configuration
Program g-ir-compiler found: YES (/usr/bin/g-ir-compiler)
Native dependency gobject-introspection-1.0 found: YES 1.56.1
Program vapigen found: YES (/usr/bin/vapigen)
[... 19 lines cut ...]
/usr/include/stdlib.h:147: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '     char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:147: syntax error, unexpected ')', expecting ',' or ';' in '     char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:158: syntax error, unexpected identifier, expecting ',' or ';' in 'extern _Float32x strtof32x (const char *__restrict __nptr,' at 'strtof32x'
/usr/include/stdlib.h:159: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '       char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:159: syntax error, unexpected ')', expecting ',' or ';' in '       char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:165: syntax error, unexpected CHAR, expecting identifier or '(' or '*' in '       char **__restrict __endptr)' at 'char'
/usr/include/stdlib.h:165: syntax error, unexpected ')', expecting ',' or ';' in '       char **__restrict __endptr)' at ')'
/usr/include/stdlib.h:233: syntax error, unexpected identifier in '         _Float32 __f)' at '_Float32'
[... cut ...]
g-ir-scanner: link: cc -o /home/supermario/src/fwupd/build/tmp-introspect_aawk9zi/Fwupd-2.0 /home/supermario/src/fwupd/build/tmp-introspect_aawk9zi/Fwupd-2.0.o -L. -Wl,-rpath,. -Wl,--no-as-needed -lfwupd -lgio-2.0 -lgobject-2.0 -lglib-2.0 -lsoup-2.4 -ljson-glib-1.0 -L/home/supermario/src/fwupd/build/libfwupd -Wl,-rpath,/home/supermario/src/fwupd/build/libfwupd -lgio-2.0 -lgobject-2.0 -Wl,--export-dynamic -lgmodule-2.0 -pthread -lglib-2.0
[177/177] Linking target plugins/thunderbolt/tbtfwucli.

```