fwupd/fwupd #587: Confusion about minimal commands needed to update to the latest firmware

ChaiTRex, 2018-07-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	0.8.3
daemon version:	0.8.3

$ dpkg -l fwupd
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                         Version                     Architecture                Description
+++-============================================-===========================-===========================-=============================================================================================
ii  fwupd                                        0.8.3-0ubuntu3              amd64                       Firmware update daemon
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

Please answer the following questions:

- Operating system and version: Ubuntu 16.04.4 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): By installing Ubuntu 16.04 (uses an APT package)
- Have you tried rebooting? N/A
- Are you using an NVMe disk? N/A
- Is secure boot enabled (only for the UEFI plugin)? N/A

---

I'm a bit confused by what's necessary to get the very latest firmware updates and apply them. With APT (Ubuntu's package manager), I have to do so in two steps: updating the latest information on all packages and downloading and applying anything new.

The description of `fwupdmgr` commands in [this repository's `README.md`](https://github.com/hughsie/fwupd#readme) seems similar:

>     # fwupdmgr refresh
> This will download the latest metadata from LVFS.
> ⋮
> 
>     # fwupdmgr update
> This will download and apply all updates for your system.

Yet the description of the last one seems to indicate that it might not need the `refresh` command. What minimal command sequence can I use to update to the very latest firmware available for my system (even if my system doesn't yet know about an update that was just published, it should get that new update) if I don't care about status messages (other than one telling me to reboot if needed)?

* All four commands in the readme.
* `refresh` followed by `update`.
* Just `update`.
* Something else?

superm1, 2018-07-10:
On it's own the daemon will not refresh metadata.  The `fwupdmgr refresh` command is needed to make that happen.

_However_, if you are using gnome-software and it's running in your session, gnome-software **will** periodically refresh fwupd remote metadata for you.

I could see the argument that if someone calls `update` and the metadata is very stale (maybe more than a week old?) we should just combine the actions in `refresh` with `update`, but that is not how it behaves today.

Are you trying to come up with something to cron?
Newer versions of `fwupdmgr` will prompt you for reboot in the case of a UEFI update, so if you are trying to come up with a scriptable solution you'll need to take those into account and provide the argument to disable showing the reboot prompt.  The version you have installed (0.8.x) I don't believe will do this however.

ChaiTRex, 2018-07-10:
I just manually run an update script that updates all my software and does some minor housekeeping tasks like trimming the SSD and rebuilding some indexes (like `locate`'s). I want it to also get the very latest firmware updates but make the script shorter if I can (currently, I'm running all four commands in the readme).

superm1, 2018-07-10:
Sounds good thanks.

closed 2018-07-10 by superm1