fwupd/fwupd #590: Error when running apt update

nikolayanchev, 2018-07-11:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	0.8.3
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out
```

```shell
$ efibootmgr -v
BootCurrent: 0003
Timeout: 1 seconds
BootOrder: 0003,0002,0004,0006,0005,0001,0000
Boot0000* Windows Boot Manager	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....h...............
Boot0001* Windows Boot Manager	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0002* Hard Drive	BBS(HD,,0x0)..GO..NO........o.S.a.m.s.u.n.g. .S.S.D. .8.5.0. .E.V.O. .2.5.0.G.B....................A...........................>..Gd-.;.A..MQ..L.2.S.P.1.S.N.G.A.3.4.8.2.0.7. .R. . . . ........BO
Boot0003* ubuntu	HD(1,GPT,a0b1ddfa-7691-41df-8a2c-bc73b7dc7d21,0x800,0x100000)/File(\EFI\UBUNTU\SHIMX64.EFI)
Boot0004* CD/DVD Drive	BBS(CDROM,,0x0)..GO..NO........o.H.L.-.D.T.-.S.T. .D.V.D.R.A.M. .G.U.E.1.N....................A...........................>..Gd-.;.A..MQ..L.L.K.G.J.N.2.5.I.2.2. .9. . . . . . . . ........BO
Boot0005* neon	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)
Boot0006* Fedora	VenHw(99e275e7-75a0-4b37-a2e6-c5385e6c00cb)

```

```shell
$ efivar -l | grep fw
Doesn't output anything
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Ubuntu 16.04
- How did you install fwupd : i think it came with the OS
- Have you tried rebooting? Yes
- Are you using an NVMe disk?  i don't know, probably not
- Is secure boot enabled (only for the UEFI plugin)? i think not

=======================================================
when i run apt update i get 

```
AppStream system cache was updated, but problems were found: Metadata files have errors: /var/cache/app-info/xmls/fwupd.xml
Reading package lists... Done
E: Problem executing scripts APT::Update::Post-Invoke-Success 'if /usr/bin/test -w /var/cache/app-info -a -e /usr/bin/appstreamcli; then appstreamcli refresh-cache > /dev/null; fi'
E: Sub-process returned an error code
```

I have poked around a bit and found there is an xml error  in the file /var/cache/app-info/xmls/fwupd.xml on line 265 
`<checksum filename="Firmware_SF30&SN30_Pro_V1.26.dat" target="content" type="sha1">3ef2bdee8aca2a45b9f53b4d4cce9722523f57f8</checksum>`
the & symbol should be escaped otherwise this isn't a valid xml.

hughsie, 2018-07-11:
You need to ask your distro to either ship appstream-glib 0.7.6 or newer, or to backport the fix to older versions: https://github.com/hughsie/appstream-glib/commit/6048520484101df5d33f3c852c10640e630d20cf

hughsie, 2018-07-11:
Here's the 16.04 bug: https://bugs.launchpad.net/ubuntu/+source/appstream-glib/+bug/1780442

nikolayanchev, 2018-07-11:
@hughsie ok, thanks

closed 2018-07-11 by nikolayanchev

superm1, 2018-07-11:
@robert-ancell can you please help finish driving that SRU through?   I think ideally we'd like to see 0.7.6+ come in, but if that's not possible at least get that patch in.

SRU bug I started was here: https://bugs.launchpad.net/ubuntu/+source/appstream-glib/+bug/1780442