fwupd/fwupd #592: 1.1.0: uefi plugin may fail to build in cross environments

tgurr, 2018-07-12:
While efi-cc and efi-ld can be passed via meson configure options, objcopy and readelf can't. On Exherbo we have our build tools prefixed with the host triplet to allow native cross compilation, so we run into a build failure when objcopy instead of x86_64-pc-linux-gnu-objcopy is found/called.

In `meson.build` I see:

```
  objcopy = find_program ('objcopy')
  readelf = find_program ('readelf')
```
and later on the `readelf` command used in `plugins/uefi/efi/meson.build`.

I'm not really familiar with the meson build system but the other regular compiler settings (C compiler/gcc) are detected my meson without such problems (probably not via a plain find_program?)

Though we sure have to set efi-cc and efi-ld for the uefi part additonally.

So it would be nice if the build system would detect the objcopy and readelf commands correctly or meson options flags get added for allowing them to be specified manually for the uefi plugin usage as well.

superm1, 2018-07-12:
Compiler is configured through CC environment variable. I don't think that can be done for objcopy and readelf but if I'm wrong please let me know.

So I think we need more meson configuration options for those.

superm1, 2018-07-12:
I looked into this some, and from what I can tell you should be building using a [meson cross file](http://mesonbuild.com/Cross-compilation.html) and overrides.
This feature is supported in meson 0.43 (which is minimum for fwupd now).

http://mesonbuild.com/Release-notes-for-0-43-0.html#can-override-executables-in-the-cross-file

superm1, 2018-07-12:
FWIW fwupd CI has a sample of how to properly use a meson cross file you may consider to reference.
https://github.com/hughsie/fwupd/blob/master/contrib/ci/debian_s390x.sh

superm1, 2018-07-17:
As this should be done with a cross file, I'm going to close this issue.

closed 2018-07-17 by superm1