fwupd/fwupd #597: Arch failure in LVFS telemetry: read only ESP?

superm1, 2018-07-16:
Saw this come in:

```
BootTime=1531676634, CompileVersion(com.redhat.efivar)=35, CompileVersion(org.freedesktop.appstream-glib)=0.7.7, CompileVersion(org.freedesktop.fwupd)=1.1.0, CompileVersion(org.freedesktop.gusb)=0.3.0, CpuArchitecture=x86_64, DistroId=arch, ESPMountPoint=/efi, FirmwareId=724, Flags=299, Guid=5ffdbc0d-f340-441c-a803-8439c8c0ae10, KernelVersion=4.17.5-1-ARCH, MachineId=81d45bb5722db99dd1d642fa85b3fcc1c885f060d22c1f8a1506e3f421ec0127, Plugin=uefi, RuntimeVersion(com.dell.libsmbios)=2.4, RuntimeVersion(org.freedesktop.fwupd)=1.1.0, SecureBoot=Disabled, UEFIUXCapsule=Disabled, UpdateError=Failed to create file /efi/EFI/arch/fw/fwupd-5ffdbc0d-f340-441c-a803-8439c8c0ae10.cap.56PXLZ: Read-only file system, UpdateState=failed, VersionNew=0.2.8.1, VersionOld=0.2.7.1
```

I notice that they are using a different ESP mount point than we previously acccounted for (`/efi`) but I don't think that's the root cause of this problem.  To help prevent this kind of failure I think we should do a `g_access` test to make sure that we can write to the ESP location selected at plugin initialization too.

hughsie, 2018-07-16:
> I don't think that's the root cause of this problem

Why is the ESP read-only then?

superm1, 2018-07-16:
Well I should say _it doesn't prove that's the root cause of the problem_. 

I suspect the root cause of the problem is actually that because they used a non-standard path which means https://github.com/hughsie/fwupd/blob/master/data/meson.build#L54 doesn't get filled in with that path.

Not sure how to fix that though.  Maybe systemd unit needs to source `uefi.conf` file and add it to the list of paths?  Not sure if that's possible to do early on.

In any case I don't think we should let them proceed with the plugin if the directory is read only no matter the reason, it's just going to lead to a failure.

ArchangeGabriel, 2018-07-16:
Because the user specified `ro` in their `fstab`? It’s a common practice to have `/boot` and/or the ESP not mounted/mounted ro to mitigate some potential issues. fwupd should issue an error message in this case, telling the user to remount their ESP with rw rights.

ArchangeGabriel, 2018-07-16:
Ah, the systemd thing could be another explanation indeed.

hughsie, 2018-07-16:
> because they used a non-standard path

Maybe I'm being dumb, but I thought we had `/efi` in `rw_directories`...

> In any case I don't think we should let them proceed with the plugin

If we return FALSE from setup(), then the plugin gets disabled and the only way of finding out what's wrong is by looking in the journal. If we detect it at firmware flash time then we can return an error to the application, e.g. fwupdmgr or gnome-software. It also means if the user re-mounts r/w at runtime (after fwupd has initialized) or even remounts r/o we don't lie to the user.

Can fwupd just remount the `/efi` path `rw` and then back to `ro` when done? I'd rather have it "just work" than force the user to learn about this thing.

superm1, 2018-07-16:
> Maybe I'm being dumb, but I thought we had /efi in rw_directories...
 
IIRC `/boot/efi`, `/boot`, and `/esp` were the ones we put by default.

> If we return FALSE from setup(), then the plugin gets disabled and the only way of finding out what's wrong is by looking in the journal.

Which is the same thing that happens if they have secure boot enabled and are missing binaries too.

> Can fwupd just remount the /efi path rw and then back to ro when done? I'd rather have it "just work" than force the user to learn about this thing.

Not with the systemd unit `ReadWriteDirectories`/`ReadWritePaths` in place.  That's potentially systemd blocking the write instead.

hughsie, 2018-07-16:
> IIRC /boot/efi, /boot, and /esp were the ones we put by default.

Nope, `/boot/efi`, `/boot` and `/efi` -- Unless a neuron went pop, I've never seen `/esp` mentioned before...

> Which is the same thing that happens if they have secure boot enabled and are missing binaries too.

Maybe that's a bug. It seems wrong to me pretending that there are no devices updatable without some kind of messaging to the user.

hughsie, 2018-07-16:
> So I know we talked about this in the past with regard to removing the UPDATEABLE flag if we know there is going to be a failure and you preferred to disable plugin instead in this situation.

Can a man change his mind? :) Either way, I'm not going to die on this hill; after all, the average arch linux user is going to be comfortable with the journal and the Ubuntu-ers of this world are unlikely to poke around and change the r/w defaults in the fstab.

It does seem a shame we can't just fix things (the r/o->r/w thing) for the user in this case.

superm1, 2018-07-16:
> Can a man change his mind? :)

OK, so my thoughts then.

I agree we should at least try to correct the ro/rw bit if we find it's wrong at install time.

Regarding the `UPDATEABLE` flag and things that cause the plugin setup to fail - how about we replace those instances with removing the `UPDATEABLE` flag and put a `NonUpdateableReason` key in place.  That key can then be exposed up to `fwupdmgr`.  We might want to revert the behavior that hides non-updateable devices too though in that case.

hughsie, 2018-07-16:
> and put a NonUpdateableReason key in place

I think we should just clear UPDATABLE and pre-fill the "UpdateError" key; it's normally used to convey the error for the flash we just tried, but it seems logically valid to use it for the error you would get if you were able to update.

superm1, 2018-07-16:
Well now i'm extra confused.  We do _check_ for read only ESP and reject it from the list.
https://github.com/hughsie/fwupd/blob/master/plugins/uefi/fu-uefi-common.c#L290

hughsie, 2018-07-17:
If we're checking for a `/ro` mount and we still manage to get `Read-only file system` maybe that suggests the filesystem is somehow read only, but the mount is `r/w` -- e.g. if the VFAT was damaged somehow and rejecting writes? I'm grabbing at straws here.

hughsie, 2018-07-17:
Okay, lets re-open if another one pops up. Maybe we can make an LVFS issue asking them to email us for more information?

closed 2018-07-17 by hughsie

superm1, 2018-07-17:
OK I've turned on a known issue for LVFS for it that should re-direct people here to re-open this bug for more investigation.
https://github.com/hughsie/fwupd/wiki/LVFS-Triaged-Issue:-UEFI-Read-only-file-system

hughsie, 2018-07-17:
Much appreciated.

0xdc, 2018-12-12:
Hi there,

I came across this issue with my /efi mounted as read-only.

Remounting with `mount -o remount,rw /efi` and trying again still gave read-only errors as described. However, after I gave a restart to fwupd.service and it started working. Does fwupd cache read-only status on startup?

superm1, 2018-12-12:
Yes it does. So that would be expected.