fwupd/fwupd #612: Build with '"gusb" not found'

CottonEaster, 2018-07-28:
Having managed to identify and install three dependencies I needed to build `fwupd`, the build system complains that `gusb` is not found, even though I've installed the only thing in my repositories that seems to answer to that, viz., `gir1.2-gusb-1.0`.

- Operating system and version: Linux Mint 19 x64 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): git
- Have you tried rebooting? no
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

CottonEaster, 2018-07-29:
I managed to identify all the missing packages. Or my system they were:
`
sudo apt install libgudev-1.0-dev libappstream-glib-dev gir1.2-gusb-1.0 libgusb-dev sqlite3 libsqlite3-dev libjsonrpc-glib-1.0-dev libpolkit-gobject-1-dev libgpgme-dev gcab efivar libefivar-dev libelf-dev libefiboot-dev libsmbios-dev gnutls-bin gobject-introspection libgirepository1.0-dev help2man gnu-efi`

However, the `ninja` part of the build fails thusly:

`/usr/bin/x86_64-linux-gnu-ld: cannot open linker script file /home/<username>/Downloads/Kernel: No such file or directory`

CottonEaster, 2018-07-29:
Moreover:

```
sudo fwupdmgr get-devices
fwupdmgr: /usr/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.0.8' not found (required by fwupdmgr)
fwupdmgr: /usr/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.0.7' not found (required by fwupdmgr)
fwupdmgr: /usr/lib/x86_64-linux-gnu/libfwupd.so.2: version `LIBFWUPD_1.1.0' not found (required by fwupdmgr)

```

hughsie, 2018-08-05:
How did you run meson?

CottonEaster, 2018-08-05:
I am sorry but I have given up on all this ('all this' being: building from source, and the flatpack method). I am using the version from my repository instead, even though it will not reboot the computer automatically.

hughsie, 2018-08-05:
Okay, fingers crossed Mint updates the distro version at some point.

closed 2018-08-05 by hughsie

superm1, 2018-08-09:
FWIW It's actually the comma in the folder that causes the failure to build in second comment, not the space.