fwupd/fwupd #614: Lenovo T480s EFI scheduled firmware update never occurs

Cirromulus, 2018-08-01:
Hello there!
I am having troubles with a fresh fedora 26 installation on a T480s, NVMe disk with secureboot enabled (now disabled for troubleshooting).
I noticed a firmware update on the fedora gui update manager, but after clicking "update", the notification just came back. When manually calling `fwupdmgr update`, it seems to interrupt itself of some sort:
```shell
$ fwupdmgr update
Downloading 0.1.25 for 20L7001VGE System Firmware...
Updating 0.1.25 on 20L7001VGE System Firmware...
Decompressing…         [  -                                     ]
Scheduling…            [      \                                 ]
$
```
Initially it did not create an entry in efiboot, so I ran `sudo dnf reinstall fwupdate-efi` (as suggested in #246), which then upon rerun created the entry. My Problem is, that somehow the Lenovo UEFI Bios does not want to boot the Firmware Updater. It is active and (manually) first in Bootorder<sup>1</sup>. It does not appear in the Boot override screen (F12), and always boots the Fedora instead. I think that somehow it does not find the Updater, or it is broken.

<sup>1</sup><sub>Some Lenovo Thinkpads wont boot an Entry not occuring in Bootorder.</sub>

```shell
$ fwupdmgr --version
client version:	0.9.9
daemon version:	0.9.9
compile-time dependency versions
	appstream-glib:	0.7.6
	gusb:	0.2.11
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceID:             cd010000-0082-8098-a07d-6104e8c4b11e
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  DeviceVendor:         Lenovo
  DeviceVendorId:       TBT:0x0109
  Version:              12.02
  Created:              2018-08-01

20L7001VGE Device Firmware
  DeviceID:             UEFI-676af093-2a5c-4238-9c29-db8063a33532-dev0
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3460
  VersionLowest:        0.0.1
  Created:              2018-08-01

20L7001VGE Device Firmware
  DeviceID:             UEFI-4d254d6e-cd67-477b-97d5-bc3048af45c4-dev0
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Created:              2018-08-01

20L7001VGE System Firmware
  DeviceID:             UEFI-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-dev0
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Description:          <p>Lenovo ThinkPad T480s System Firmware N22ET48W, version 1.25</p>
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.14
  VersionLowest:        0.0.1
  Created:              2018-08-01
  Modified:             2018-08-01

USB Billboard Device   
  DeviceID:             usb:00:01:05
  Guid:                 33bdfda8-516f-5874-9ec2-dfc952756b33
  Guid:                 b9ceddca-6450-53b2-acea-0e25b68aa91e
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x17EF
  Version:              0.1
[... 194 lines cut ...]
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub
│   └── splash.xpm.gz
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
├── initramfs-0-rescue-7a1a1141978e432ca505021562f81561.img
├── initramfs-4.12.12-300.fc26.x86_64.img
├── initramfs-4.16.11-100.fc26.x86_64.img
├── memtest86+-5.01
├── System.map-4.12.12-300.fc26.x86_64
├── System.map-4.16.11-100.fc26.x86_64
├── vmlinuz-0-rescue-7a1a1141978e432ca505021562f81561
├── vmlinuz-4.12.12-300.fc26.x86_64
└── vmlinuz-4.16.11-100.fc26.x86_64
```
Thanks for your help.

exploide, 2018-08-03:
Don't know if this is causing your problem, but please note that Fedora 26 is end of life and hence you are using a rather old version of fwupd. Maybe it works with a more recent version. (BTW, Fedora, as a cutting edge distribution, is not meant to be used much longer than when the next release appears)

Cirromulus, 2018-08-03:
Thanks for your reply. I installed the OS just a month ago and already outdated, that is pretty cutting edge 😄
I will do a dist-upgrade next week and post the results.

exploide, 2018-08-04:
Haha, all right :laughing:  But I think your installation iso was already outdated when you installed the system. Fedora 28 is released since May. Maybe you didn't download it from https://getfedora.org but however. Generally, a new Fedora release approaches approx. every 6 months. And when release n is out, release n-2 will be end of life short after. If you want to upgrade to F27 and maybe F28 afterwards, see https://fedoraproject.org/wiki/DNF_system_upgrade

hughsie, 2018-08-04:
I think Lenovo ought to have put a requirement on fwupdate 11 on the update. Another thing to check is that you've not set the BootOrder to `locked` in the BIOS.

superm1, 2018-08-04:
@hughsie What happens if fwupdate requirement is set to 11 and you use 1.1.x? I don't know for sure we thought about that when merging.

hughsie, 2018-08-04:
>  if fwupdate requirement is set to 11 and you use 1.1.x?

That is a good question. We should probably add an automatic provide of fwupdate 11 to 1.1.0

Cirromulus, 2018-08-06:
Confirmed. Dist-Upgrade to F28, firmware upgrade was done direct in upgrade-reboot. Now clean.
Thanks for your time.

closed 2018-08-06 by Cirromulus

hughsie, 2018-08-06:
I've set the requirement on the firmware manually now. Thanks.