fwupd/fwupd #615: Automatic reboot does not work; Mint 19

CottonEaster, 2018-08-02:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**INSERT OUTPUT HERE**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
`0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-a4b51dca-8f97-4310-8821-3330f83c9135-0`

```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```
$ tree /boot | grep .cap
[no output]
$  tree /boot | grep fwupx64.efi
[no output]

Please answer the following questions:

- Operating system and version: Mint 19 x64 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting?
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

```
$  fwupdmgr update
Downloading RQR24.05_B0029 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/4511b9b0d123bdbe8a2007233318ab215a59dfe6-Logitech-Unifying-RQR24.05_B0029.cab
Downloading…           [***************************************]
Updating RQR24.05_B0029 on Unifying Receiver...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Restarting device…     [***************************************]Downloading 0.1.27 for 20KHCTO1WW System Firmware...
Fetching firmware https://fwupd.org/downloads/769be80680bf15ee51a3782e664041900c45de76-Lenovo-ThinkPad-X1Carbon6th-SystemFirmware-1.27.cab
Downloading…           [***************************************]
Updating 0.1.27 on 20KHCTO1WW System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: Y
GDBus.Error:org.freedesktop.DBus.Error.AccessDenied: Rejected send message, 2 matched rules; type="method_call", sender=":1.1248" (uid=1000 pid=32611 comm="fwupdmgr update ") interface="org.freedesktop.systemd1.Manager" member="Reboot" error name="(unset)" requested_reply="0" destination="org.freedesktop.systemd1" (uid=0 pid=1 comm="/sbin/init splash ")
```

hughsie, 2018-08-02:
This was fixed a while ago in 1.0.7

closed 2018-08-02 by hughsie

hughsie, 2018-08-02:
Use a snap or flatpak?

superm1, 2018-08-03:
FWIW if you're having difficulty compiling packages on the system, it can be done in docker using the contrib scripts as mentioned in the wiki.
https://github.com/hughsie/fwupd/wiki/Compilation

CottonEaster, 2018-08-03:
Ah, you mean to refer to this [webpage](https://github.com/hughsie/fwupd/tree/master/contrib). The procedure that you describe requires that one find the `generate_docker.py` script (which turns out to be in `contrib/ci`). And one must installer docker. And then (though perhaps my python installation is messed up) I get a load of errors terminating with

```
No such file or directory: 'docker'
```

(even though I did install docker).

superm1, 2018-08-03:
Sounds like you don't really have docker installed if you're getting that error. Note the package in the apt repositories provided by ubuntu that mint uses is not called docker. I believe it's docker-io or something like that.

CottonEaster, 2018-08-03:
Thanks. The package was called `docker.io`. But now I have yet another problem:

```
$ sudo docker run -t -v `pwd`:/build fwupd-ubuntu-x86_64
docker: Error response from daemon: OCI runtime create failed: container_linux.go:348: starting container process caused "exec: \"./contrib/ci/ubuntu.sh\": stat ./contrib/ci/ubuntu.sh: no such file or directory": unknown.
```

superm1, 2018-08-03:
Did you run the container build command from top level directory of the checkout?

CottonEaster, 2018-08-03:
No, I did not run the command from the root of the git folder. But I have now. I got the following error.

```
Fu:ERROR:../src/fu-self-test.c:1729:fu_common_firmware_builder_func: assertion failed (error == NULL): failed to build firmware: bwrap: No permissions to creating new namespace, likely because the kernel does not allow non-privileged user namespaces. On e.g. debian this can be enabled with 'sysctl kernel.unprivileged_userns_clone=1'.
 (FwupdError, 0)
-------

Full log written to /build/build/obj-x86_64-linux-gnu/meson-logs/testlog.txt
FAILED: meson-test 
/usr/bin/python3 -u /usr/bin/meson test --no-rebuild --print-errorlogs
ninja: build stopped: subcommand failed.
dh_auto_test: cd obj-x86_64-linux-gnu && LC_ALL=C.UTF-8 MESON_TESTTHREADS=8 ninja test returned exit code 1
make[1]: *** [debian/rules:75: override_dh_auto_test] Error 1
make[1]: Leaving directory '/build/build'
make: *** [debian/rules:32: build] Error 2
dpkg-buildpackage: error: debian/rules build subprocess returned exit status 2
debuild: fatal error at line 1152:
dpkg-buildpackage -rfakeroot -us -uc -ui failed
```
Now, `sysctl kernel.unprivileged_userns_clone=1` itself produced an error ('no such file or directory'). So I tried

```
echo 'kernel.unprivileged_userns_clone=1' > /etc/sysctl.d/00-local-userns.conf
service procps restart
```
But even after a reboot I still received the aforementioned error.

CottonEaster, 2018-08-03:
`--privileged` has to go between `run` and `-t`, or at least the command didn't run when I put that argument in various other places.

When the command did run, it created various debs.

[image]

Which one do I install, please?

superm1, 2018-08-03:
Okay we'll need to update documentation to include that. All but fwupd-tests and the template package are needed.

CottonEaster, 2018-08-03:
I had to remove older versions of the packages first. Also, the packages needs to be installed in a particular order. Also, the final package, `fwupd`, had an unsatisfiable dependency.

This is too much for an end user to have to do. Can't you push an update, or a flatpack, to Mint, please?

superm1, 2018-08-03:
Sorry we're not in control of what mint puts in repositories. You'll need to talk to them.

There is a flatpak available as documented in the wiki.

https://github.com/hughsie/fwupd/wiki/fwupd-flatpak

CottonEaster, 2018-08-03:
Thank you. (I am new to flatpacks; I said above I could not find the pack in Mint's flatpack system; I did not know one could download such a pack from a website and install it - and I didn't know where to get such a file.)

But, am I right in thinking that, now, I have to preface every fwup command with `sudo flatpak run org.freedesktop.fwupd`?

CottonEaster, 2018-08-03:
I can't manage to get the `refresh` or `get-updates` commands, but only the`get-devices` command., to run.

Do you think you could message the Mint people about getting the newer version into the repository? I'm sure they'll take the request more seriously from you; and you can point out the rebooting bug, and the difficulty of installing your package in any other way. There is an IRC channel at, I think, `#linuxmint-devs` (or possibly `#linuxmint-dev`). Or you could try `https://github.com/linuxmint/mintinstall/issues`.