fwupd/fwupd #616: Post-reboot 'update found' message

CottonEaster, 2018-08-02:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

The post-reboot 'update found' message (1) does not tell the user what is happening - is the update going to be installed? Is there some sort of problem? (2) does not adjust itself to hiDPI, and consequently is almost too small to read.

```shell
$ fwupdmgr get-devices
**INSERT OUTPUT HERE**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
`0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-a4b51dca-8f97-4310-8821-3330f83c9135-0`

```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```
$ tree /boot | grep .cap
[no output]
$  tree /boot | grep fwupx64.efi
[no output]

Please answer the following questions:

- Operating system and version: Mint 19 x64 Cinnamon
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting?
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2018-08-03:
You're missing the output from several of those commands that were in the template.  I suspect that this means the update didn't run at bootup on your machine for some reason but those commands are needed to confirm.

CottonEaster, 2018-08-03:
> You're missing the output from several of those commands that were in the template. 

Apologies. Here you go.

```
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             16324df8d782cbc6d3396541199c8fc397182242
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.05_B0029
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  Created:              2018-08-03

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-03

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.8
  VersionLowest:        0.1.8
  Icon:                 audio-card
  Created:              2018-08-03

20KHCTO1WW System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.27
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-08-03

X1 Carbon Thunderbolt Controller
  DeviceId:             74a7c8a4eaeb4ebe00e809c0eab60b420a5e452c
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              30.00
  Icon:                 computer
  Created:              2018-08-03
```

```
 efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,001A,001C,0019,0018,001B,001D,001E,001F,0024
Boot0000* ubuntu	HD(1,GPT,d3d8b2bf-16e6-470b-8c07-d628691e14de,0x800,0x106000)/File(\EFI\ubuntu\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
[... 314 lines cut ...]
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.15.0-29-generic
├── initrd.img-4.17.4-041704-generic
├── initrd.img-4.17.9-041709-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── retpoline-4.15.0-29-generic
├── retpoline-4.17.4-041704-generic
├── retpoline-4.17.9-041709-generic
├── System.map-4.15.0-29-generic
├── System.map-4.17.4-041704-generic
├── System.map-4.17.9-041709-generic
├── vmlinuz-4.15.0-29-generic
├── vmlinuz-4.17.4-041704-generic
└── vmlinuz-4.17.9-041709-generic

6 directories, 298 files
```

The update did run - I know this because it was a very noisy (it makes many a beep) BIOS update.

hughsie, 2018-08-10:
Can you try with 1.0.8 please? Or git master if you're feeling brave.

CottonEaster, 2018-08-10:
No, I cannot, I am afraid. For, as documented in another, long thread on this tracker (which my github fu is insufficient to relocate), getting a newer version installed proved exceptionally difficult - with build failures and flatpak problems.

superm1, 2018-08-13:
Since the commits for this are in 1.0.8+ we know I believe we should close this then.

closed 2018-08-13 by superm1