fwupd/fwupd #619: Lenovo X1 Carbon 5th Gen BIOS update reboot on update

megian, 2018-08-05:
The update configuration is done and also found at boot time.

```
Found update fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4-0
File \EFI\arch\fw\fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4.cap searched
```

The update is starting with "wait ..." 1 second later the system is reset.

```shell
$ fwupdmgr --version
client version:	1.1.0
daemon version:	1.1.0
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
20HR002MMZ System Firmware
  DeviceId:             1cbe298fc17877b6883e85560778ac0812f0a385
  Guid:                 798ffd60-f10e-4ac4-8939-c8beabfe55b4
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.31
  VersionLowest:        0.1.14
  Icon:                 computer
  Created:              2018-08-05
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             6461040534f2259a0439361986adf73979fd836e
  Guid:                 c35736d2-9e47-4578-93e9-68d5b04ea77e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              182.29.3287
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-05

UEFI Device Firmware
  DeviceId:             6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
  Guid:                 74997a6b-1adf-4b12-b994-401f06ea8c72
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.19
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-05

ThinkPad X1 Carbon Thunderbolt Controller
  DeviceId:             3089ddc42fafce0092931e7578d018c46732d213
  Guid:                 89d9d1e6-9e4c-5f07-b5c1-603da3d61835
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              15.00
  Icon:                 computer
  Created:              2018-08-05
```

```shell
$ efibootmgr -v
BootNext: 0000
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux Firmware Updater	HD(1,GPT,eafbb25c-84f3-544a-b6f4-2cf27ee33e4a,0x800,0x113000)/File(\EFI\arch\fwupdx64.efi)
Boot0001* arch	HD(1,GPT,eafbb25c-84f3-544a-b6f4-2cf27ee33e4a,0x800,0x113000)/File(\EFI\arch\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4-0
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

```
$ ls -lh EFI/arch/fw/fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4.cap
-rwxr-xr-x 1 root root 9.0M Aug  5 13:45 EFI/arch/fw/fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4.cap
```

Please answer the following questions:

- Operating system and version: Arch Linux (latest updates)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? No

joakim-tjernlund, 2018-08-06:
hmm, this actually matches my experience on Lenovo Thinkpad T480s better than issue #620 
I do have my ESP mounted on /boot though. Not sure where yours is?

megian, 2018-08-08:
My partition table looks like:

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048         1128447   550.0 MiB   EF00  
   2         1128448         4098047   1.4 GiB     8300  

The EFI system partition (ESP) is mounted at /boot/efi. Filesystem FAT32.

Because you can found "Please wait while we install a system update" messages also from Windows users. I'm not sure, if this is relay related to fwupd or just an bug in the firmware package itself.

joakim-tjernlund, 2018-08-08:
Since you ESP is on /boot/efi it will probably work if you downgrade to sys-apps/fwupd-1.0.8

joakim-tjernlund, 2018-08-08:
well, it did work for me(sort of), see issue 620 where I messed with the older fwupd-1.0.8 and then
upgraded again, suddenly it worked.

Using fwupdmgr -v install --allow-reinstall 2745a26ac883b9c171522bb16e1c628632932b20-Lenovo-ThinkPad-T480s-SystemFirmware-1.25.cab I have recreated the problem again and so far I just
fails as first described in this report. I installed sys-apps/fwupdate-12 and copied its
 ./EFI/gentoo/fwupx64.efi to ./gentoo/fwupdx64.efi and it still fails.

joakim-tjernlund, 2018-08-08:
Maybe there is VFAT/FAT problem? I created my ESP with:
mkfs.vfat -F 32 -n BOOT

hughsie, 2018-08-08:
What does `cat /etc/mtab | grep efi` say?

joakim-tjernlund, 2018-08-08:
After bootup
cat /etc/mtab | grep efi 
efivarfs /sys/firmware/efi/efivars efivarfs ro,relatime 0 0

but I manually change that before upgrading:
cat /etc/mtab | grep efi 
efivarfs /sys/firmware/efi/efivars efivarfs rw,relatime 0 0

hughsie, 2018-08-08:
can you do the same with `grep nvme0n1p1` pls.

joakim-tjernlund, 2018-08-08:
cat /etc/mtab | grep nvme0n1p1
/dev/nvme0n1p1 /boot vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro 0 0

joakim-tjernlund, 2018-08-08:
I changed to same as you(except for iocharset=ascii as mount did not like that):
```
cat /etc/mtab | grep efi
efivarfs /sys/firmware/efi/efivars efivarfs rw,nosuid,nodev,noexec,relatime 0 0
se-jocke-lx ~ # cat /etc/mtab | grep nvme0n1p1
/dev/nvme0n1p1 /boot vfat rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=winnt,errors=remount-ro 0 0
```

But it did not help, error still there.

joakim-tjernlund, 2018-08-08:
downgrading to 1.0.8 I get:
```
(fwupdmgr:8295): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         [  -                                    ](fwupdmgr:8295): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Decompressing…         [***************************************]
(fwupdmgr:8295): Fwupd-DEBUG: Emitting ::status-changed() [waiting-for-auth]
Authenticating…        [   -                                   ](fwupdmgr:8295): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Authenticating…        [***************************************]
(fwupdmgr:8295): Fwupd-DEBUG: Emitting ::status-changed() [scheduling]
Scheduling…            [            /                          ](fwupdmgr:8295): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Scheduling…            [***************************************]
UEFI firmware update failed: {error #0} libfwup.c:1081 get_paths(): could not find fwup on ESP: No such file or directory	{error #1} libfwup.c:1219 set_up_boot_next(): could not find paths for shim and fwup: No such file or directory
```

Is that a known problem?

joakim-tjernlund, 2018-08-08:
After the this I upgraded fwupd again to 1.1.0, reinstalled: 
fwupdmgr -v install --allow-reinstall 2745a26ac883b9c171522bb16e1c628632932b20-Lenovo-ThinkPad-T480s-SystemFirmware-1.25.cab
and now it WORKED.
```
efibootmgr  -v
BootNext: 0000
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0023,0000
Boot0000* Linux Firmware Updater	HD(1,GPT,078adc23-858e-452b-89dd-88d310315494,0x800,0x80000)/File(\EFI\gentoo\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```
here is my ESP after upgrade:
```
s -lR .
.:
total 3
drwx------ 4 root root  512 Aug  8 16:52 ./
drwx------ 5 root root 1536 Jan  1  1970 ../
drwx------ 2 root root  512 Aug  7 12:32 BOOT/
drwx------ 3 root root  512 Aug  7 16:44 gentoo/

./BOOT:
total 120
drwx------ 2 root root    512 Aug  7 12:32 ./
drwx------ 4 root root    512 Aug  8 16:52 ../
-rwx------ 1 root root 121344 Aug  7 12:32 BOOTX64.EFI*

./gentoo:
total 67
drwx------ 3 root root   512 Aug  7 16:44 ./
drwx------ 4 root root   512 Aug  8 16:52 ../
drwx------ 2 root root   512 Aug  8 17:41 fw/
-rwx------ 1 root root 66963 Aug  6 15:26 fwupdx64.efi*

./gentoo/fw:
total 18395
drwx------ 2 root root     512 Aug  8 17:41 ./
drwx------ 3 root root     512 Aug  7 16:44 ../
-rwx------ 1 root root 9417280 Aug  8 17:40 fwupdate-Ihi5q8.cap*
-rwx------ 1 root root 9417280 Aug  8 17:41 fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf.cap*
```
I also noted that the BIOS EFI found both *.cap files this time. I diffed them and they are identical.
.cap file name/length problem?

superm1, 2018-08-08:
Maybe fwupd 1.1.x is building the EFI variable in /sys differently. Can you binary diff those?

joakim-tjernlund, 2018-08-08:
not really fwupd 1.0.8 fails before it gets to the /sys efi variables.
I just noted that even if my ESP is /boot, fwupdate installs everything in /boot/efi/EFI/
How can I make/rebuild fwupdate to use /boot/EFI instead ?

hughsie, 2018-08-08:
> How can I make/rebuild fwupdate to use /boot/EFI instead

I really think it's better if you just stick to 1.1.x now.

joakim-tjernlund, 2018-08-08:
but 1.1.x does not work and I figured I could test with 1.0 to see where the 1.1.x bug lies.

joakim-tjernlund, 2018-08-09:
built fwupd from master instead and now it actually WORKED!
Time for a new release ?

hughsie, 2018-08-09:
> Time for a new release ?

I think so. There the Titan Ridge PR i'd like to get in, but after that we should be good for a new tarball.

superm1, 2018-08-09:
Closed by https://github.com/hughsie/fwupd/commit/4ab6116ea08798ffabdc932a2e694db4090ffd9a

closed 2018-08-09 by superm1

joakim-tjernlund, 2018-08-09:
The Titan Ridge PR could be the same problem as this one.

superm1, 2018-08-09:
Titan Ridge PR is for supporting new Thunderbolt hardware.  Not for a UEFI problem.

superm1, 2018-08-09:
@joakim-tjernlund I agree it's potentially the same root cause that I fixed by 4ab6116

megian, 2018-08-09:
Can confirm that I have Thunderbold enabled in the BIOS and the output of fwupdate.

# fwupdate --info

Information for the update status entry 0:
  Information Version: 7
  Firmware GUID: {c35736d2-9e47-4578-93e9-68d5b04ea77e}
  Capsule Flags: 0x00008010
  Hardware Instance: 0
  Update Status: Unknown
  Capsule File Path: 

Information for the update status entry 1:
  Information Version: 7
  Firmware GUID: {74997a6b-1adf-4b12-b994-401f06ea8c72}
  Capsule Flags: 0x00000000
  Hardware Instance: 0
  Update Status: Unknown
  Capsule File Path: 

Information for the update status entry 2:
  Information Version: 7
  Firmware GUID: {798ffd60-f10e-4ac4-8939-c8beabfe55b4}
  Capsule Flags: 0x00000000
  Hardware Instance: 0
  Update Status: Unknown
  Capsule File Path: 

Information for the update status entry 3:
  Information Version: 7
  Firmware GUID: {ux_capsule}
  Capsule Flags: 0x00000000
  Hardware Instance: 0
  Update Status: Unknown
  Capsule File Path:

megian, 2018-08-15:
Update did work with the version 1.1.1. Unfortunately the BIOS update purged all the NVRAM boot entries and prevented the OS booting, thanks Lenovo :( Although it's booting from ESP:/boot/bootx64.efi. So you may check your grub installation before updating the BIOS.

hughsie, 2018-08-15:
> Unfortunately the BIOS update purged all the NVRAM boot entries 

What did it do exactly?

megian, 2018-08-15:
The boot entries Boot0000 and Boot0001 are gone after the BIOS upgrade, while all BIOS settings are preserved. Boot0001 "arch" was my primary boot entry.

```
Boot0000* Linux Firmware Updater	HD(1,GPT,eafbb25c-84f3-544a-b6f4-2cf27ee33e4a,0x800,0x113000)/File(\EFI\arch\fwupdx64.efi)
Boot0001* arch	HD(1,GPT,eafbb25c-84f3-544a-b6f4-2cf27ee33e4a,0x800,0x113000)/File(\EFI\arch\grubx64.efi)
```

Currently it looks like this:
```
$ efibootmgr  --verbose
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

superm1, 2018-08-15:
Those old entries should hopefully be in the journal log for fwupd to recover from.  (before boot they're all saved for this purpose in case of problems).

megian, 2018-08-15:
They are in the log. But while not able to boot the system, the workaround copy ESP:/arch/grubx64.efi to ESP:/boot/bootx64.efi was the fastest way to get the system booting again and fix it.

superm1, 2018-08-15:
Would you care to please share what was there previously?  It may be of assistance to debugging this.

megian, 2018-08-17:
All the information of the past, I have, are in the first entry of this issue. What kind of information do you need?