fwupd/fwupd #623: Lenovo X1 Carbon 5th gen not showing up as supported

wenottingham, 2018-08-06:
According to the website (https://fwupd.org/lvfs/device/798ffd60-f10e-4ac4-8939-c8beabfe55b4) my laptop is supported. But it's not showing up as having an updatable device available.

```shell
$ fwupdmgr --version
[notting@nostromo: ~]$ fwupdmgr --version
client version:	1.0.8
daemon version:	1.0.8
compile-time dependency versions
	appstream-glib:	0.7.8
	gusb:	0.3.0
	fwupdate:	10
	efivar:	34

```

```shell
[notting@nostromo: ~]$ fwupdmgr get-devices
[notting@nostromo: ~]$ fwupdmgr get-devices --show-all-devices
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.6.29
  VersionBootloader:    11.6.29
  Icon:                 computer
  Created:              2018-07-25

HD Graphics 620 (ThinkPad X1 Carbon 5th Gen)
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-07-25

```

```shell
[notting@nostromo: ~]$ efibootmgr -v
BootCurrent: 0000
Timeout: 2 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E
Boot0000* Fedora	HD(1,GPT,1eea94c1-a535-4bfe-ab06-f91a60037079,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
[notting@nostromo: ~]$ efivar -l  | grep fw
(nothing)
```

```shell
[root@nostromo ~]# find /boot -name fwup*
/boot/efi/EFI/fedora/fwupx64.efi
/boot/efi/EFI/fedora/fwupia32.efi
```

- Operating system and version

Fedora 28

- How did you install fwupd

included in distro

- Have you tried rebooting?

No, because it's presumably before the point where that matters.

- Are you using an NVMe disk?

Yes.

- Is secure boot enabled (only for the UEFI plugin)?

Yes.

wenottingham, 2018-08-06:
system info:

```
[root@nostromo id]# fwupdmgr hwids 
Computer Information
--------------------
BiosVendor: LENOVO
BiosVersion: N1MET46W (1.31 )
BiosMajorRelease: 01
BiosMinorRelease: 1f
Manufacturer: LENOVO
Family: ThinkPad X1 Carbon 5th
ProductName: 20HR0CTOUS
ProductSku: LENOVO_MT_20HR_BU_Think_FM_ThinkPad X1 Carbon 5th
EnclosureKind: a
BaseboardManufacturer: LENOVO
BaseboardProduct: 20HR0CTOUS

Hardware IDs
------------
{06385e28-6179-5313-94a6-eb959073caa0}   <- Manufacturer + Family + ProductName + ProductSku + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{3cabbc5f-ed31-5404-ab0e-1cda9ddad5ec}   <- Manufacturer + Family + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{3533f7ab-a2ba-5579-8b3b-6fe72403ecdd}   <- Manufacturer + ProductName + BiosVendor + BiosVersion + BiosMajorRelease + BiosMinorRelease
{e07a98b7-4140-53d7-95ff-ae9b8c7672f3}   <- Manufacturer + Family + ProductName + ProductSku + BaseboardManufacturer + BaseboardProduct
{9268b4b2-6a15-5916-8bec-c082b31823f7}   <- Manufacturer + Family + ProductName + ProductSku
{6798f114-dbd0-59a7-a89b-5ae1163d40e6}   <- Manufacturer + Family + ProductName
{c987a9c9-14ff-509b-b29f-1721af73bad5}   <- Manufacturer + ProductSku + BaseboardManufacturer + BaseboardProduct
{3fb92698-0841-52a5-85c2-11f633e675d7}   <- Manufacturer + ProductSku
{2c09a1cf-55f7-587e-9e41-90dda56951db}   <- Manufacturer + ProductName + BaseboardManufacturer + BaseboardProduct
{dd5cbf43-604f-5557-ac8d-94051f5a4a9d}   <- Manufacturer + ProductName
{663e0a44-980d-58e8-b00f-dcd3badd9daf}   <- Manufacturer + Family + BaseboardManufacturer + BaseboardProduct
{595a0c1a-9819-52f6-8ea7-e0924c073f64}   <- Manufacturer + Family
{e093d715-70f7-51f4-b6c8-b4a7e31def85}   <- Manufacturer + EnclosureKind
{89465eae-068c-57fd-9fd0-c4553f32942e}   <- Manufacturer + BaseboardManufacturer + BaseboardProduct
{6de5d951-d755-576b-bd09-c5cf66b27234}   <- Manufacturer

```

hughsie, 2018-08-06:
You have hardware that identifies as `3ec3df3a-2290-56e5-9d2f-eda62e9ab50b` -- that hardware doesn't seem to be supported yet. "ThinkPad X1 Carbon 5th Gen" does match, but there does seem to be more than one model that's called that.

wenottingham, 2018-08-07:
Hm. The BIOS that's on Lenovo's site for it (https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-x-series-laptops/thinkpad-x1-carbon-type-20hr-20hq/downloads/ds120390) appears to be the same one LVFS is offering.

hughsie, 2018-08-07:
I'm being an idiot, sorry. I was looking at the udev-created device not the (missing) uefi-created device. You seem to be missing all the ESRT-derived devices altogether. Can you tell me what the output of `sudo fwupdate -l` is please? If that produces plausible GUIDs, can you also get me the `--verbose` log of fwupd also?

wenottingham, 2018-08-07:
```
openat(AT_FDCWD, "/sys/firmware/efi/esrt/entries/", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = -1 ENOENT (No such file or directory)
```
(from `sudo fwupdate -l`). I will assume that's the problem.

wenottingham, 2018-08-07:
... Oops.

Apparently firmware updating from UEFI was disabled in the BIOS.  Flipped those settings and now things look proper:
```
[notting@nostromo: ~]$ fwupdmgr get-updates
20HR0CTOUS System Firmware has firmware updates:
GUID:                    798ffd60-f10e-4ac4-8939-c8beabfe55b4
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
ID:                      com.lenovo.ThinkPadN1MET.firmware
Update Version:          0.1.34
Update Name:             ThinkPad X1 Carbon 5th
Update Summary:          Lenovo ThinkPad X1 Carbon 5th System Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(011ee3631a43d02eb05940eab45dfd2a750a4e12)
Update Location:         https://fwupd.org/downloads/2b2f0a343bdc7a2861e20779c3f12f9bd3c46d5f-Lenovo-ThinkPad-X1Carbon5th-SystemFirmware-1.34.cab
Update Description:      Lenovo ThinkPad X1 Carbon 5th System Firmware
                         
                         CHANGES IN THIS RELEASE
...
```

closed 2018-08-07 by wenottingham