fwupd/fwupd #624: Thinkpad P51 and CRYPTO_LUKS not working

brandsimon, 2018-08-06:
```shell
$ fwupdmgr --version
client version:	1.1.0
daemon version:	1.1.0
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
20HHCTO1WW System Firmware
  DeviceId:             2ceb88afd526670d958f8626ae9ebc2596968f79
  Guid:                 ed5cd219-4b49-4105-8952-31fe3e09dde4
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.15
  VersionLowest:        0.1.14
  Icon:                 computer
  Created:              2018-08-06
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             bd1e082e3bd067233e618414bb1020eb5eaa7e0c
  Guid:                 bec0a796-5688-499f-b208-9fc1d1e734a6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-06

UEFI Device Firmware
  DeviceId:             8425785da201bc11fe069e55a6f7d3ac3ec109ee
  Guid:                 da12576b-eb03-4efa-bafe-4c977515bbe9
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.7
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-06
```

```shell
$ efibootmgr -v
BootNext: 0002
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0017,0000,0018,001A,001B,001C,001D,0019,001E,0020,0021,0022,001F,0027,0002
Boot0000  Windows Boot Manager	HD(1,GPT,c608d1c2-5255-47b5-bb38-cca2ab703986,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...e................
Boot0001* arch	HD(2,GPT,0fcda1df-def9-40ad-b0d8-66c6135f5751,0x3c00800,0x200000)/File(\EFI\arch\grubx64.efi)
Boot0002* Linux Firmware Updater	HD(2,GPT,0fcda1df-def9-40ad-b0d8-66c6135f5751,0x3c00800,0x200000)/File(\EFI\arch\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A  NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001B  ATA HDD2	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C  ATA HDD3	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f604)
Boot001D  ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f603)
Boot001E  ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001F  USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0022  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0023* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0024* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
Boot0025* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0026* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0027* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
Boot0000  Windows Boot Manager	HD(1,GPT,c608d1c2-5255-47b5-bb38-cca2ab703986,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...e................
Boot0002* Linux Firmware Updater	HD(2,GPT,0fcda1df-def9-40ad-b0d8-66c6135f5751,0x3c00800,0x200000)/File(\EFI\arch\fwupdx64.efi)
```

```shell
$ find /boot/efi/EFI/arch/
/boot/efi/EFI/arch/
/boot/efi/EFI/arch/grubx64.efi
/boot/efi/EFI/arch/fw
/boot/efi/EFI/arch/fw/fwupd-ed5cd219-4b49-4105-8952-31fe3e09dde4.cap
/boot/efi/EFI/arch/fwupdx64.efi
```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

Hi, I have problems updating my firmware.
I use grub with cryptoluks. The above commands are executed after `fwupdmgr update` and before a reboot. If I reboot, I just see my grub screen, no update gets executed.
To force an update, I edited my grub.cfg with the following:

```
menuentry "fwupx64" {
	load_video
	set gfxpayload=keep
	insmod part_gpt
	set root=(hd1,gpt2)
	chainloader (hd1,gpt2)/EFI/arch/fwupdx64.efi
}
```

If I select this after an reboot, I see the screen where it is searching for ...cab. After that, the laptop asks me for my credentials (fingerprint/passphrase), and then it says, that it will update now. The update message only appears for a few seconds, then the laptop is rebooting.
When I start my arch linux, no update was executed.

I am not sure if it is due to grub/crypto_luks, but I'd like to update my firmware anyway.

patricius972, 2018-08-07:
Try this: 

dowgrade to the 1.0.8 arch package, restart the systemd service and do a refresh update (dont reboot), it should fail somehow., but verify that it create a cap file. Then reupdate to 1.1.0, restart service update and it will create different cap but should work with the efi file. Reboot, and see.

hughsie, 2018-08-07:
> expecting it in a particular filename

I very much doubt it.

brandsimon, 2018-08-07:
Thank you very much for your help, time and for providing such a great tool! :-)

I did what @patricius972 recommended and it worked.
The file `/boot/efi/EFI/arch/fw/fwup-$GUID.cap` was indeed created.
After I rebooted, I saw grub again and I choose the fwupd entry (starting fwupdx64.efi)
It found both `*.cap` files! Then I provided my credentials, got the screen, where the firmware got updated. It beeped a lot (which made me a bit worried). After the reboot, the firmware was updated (confirmed via fwupdmgr and UEFI).

Thank you a lot, I'd like to provide you any additional data if needed.

RedL0tus, 2018-08-09:
I've met the same problem with my ThinkPad P50. Mine can boot to `fwupdx64.efi` correctly but the `Please wait while we install a system update` message only last less than a second and then my computer reboots (Of course the upgrade failed). After trying @patricius972 's method now it works.

closed 2018-08-10 by hughsie

0vximcmlxxv, 2019-12-23:
I am on Lenovo Thinkpad T470 with OEL 7.6 and LUKS Encryption enabled. I was not able to use fwupdmgr (1.0.8) to update the Lenovo BIOS and Firmware using the Lenovo provided "cab" files. I tried various options like using fwupdmgr, fwupdate, changing various BIOS settings etc, but was not successful.

But following the instructions from @patricius972 of running:
- systemctl daemon-reload
- fwupdmgr refresh
- fwupdmgr update and
- rebooting the system
resulted in successfully updating the BIOS from 1.38 to 1.60 and Firmware from 1.25 to 1.34.
(Note: I was already on fwupdmgr 1.0.8 version, hence I did not have to go through downgrading fwupdmgr version)

@patricius972 and @RedL0tus thanks for your comments and advice.