fwupd/fwupd #625: /sys/firmware/efi/efivars: Read-only file system

joakim-tjernlund, 2018-08-07:
Gentoo always mount efivars readonly so fwupdmgr fails with:
Error opening file “/sys/firmware/efi/efivars/fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0-0abba7dc-e516-4167-bbf5-4d9d1c739416”: Read-only file system
Unless one manually remonts efivars rw.

Could fwupdmgr remount efivvars automatically instead?

superm1, 2018-08-08:
That seems like it's going to be generally problematic.  For example, `efibootmgr` is going to fail as well if efivars is mounted read only all the time.

hughsie, 2018-08-08:
Why does it mount it ro? That seems like a bug. We should probably detect that and not offer any updates in that case.

joakim-tjernlund, 2018-08-08:
It is intentional as there is a possibility to "brick/shoot your in the foot" , say some script/admin
does rm -rf in efivars

There is a similar RFE for efibootmgt too:
https://github.com/rhboot/efibootmgr/issues/76

hughsie, 2018-08-08:
No, that's not possible anymore. Files in a rw efivars mount are immutable by default.

superm1, 2018-08-08:
Been that way for a long time too now.

https://github.com/torvalds/linux/commit/ed8b0de5a33d2a2557dce7f9429dca8cb5bc5879

closed 2018-08-09 by hughsie

ArchangeGabriel, 2018-08-09:
@superm1 Some news travel slowly… People who bricked their laptop because of this made a lot of noise, and the immediate reaction from people was to switch to `ro`. News of the proper fix did not spread well and configs last long.

superm1, 2018-08-09:
@ArchangeGabriel makes sense to me.  If you're aware of anywhere else that still does `RO`, please notify them.  I've mentioned this in the recent [OpenRC bug report](https://github.com/OpenRC/openrc/issues/238) on the topic.

At least from fwupd side it will show a very obvious `UpdateError` now that this is the reason that you can't do an update.

commit fd99902b (2018-08-09): uefi: Test for read only efivarfs (Closes: #625)