fwupd/fwupd #627: Dell XPS 15 9560 fwupd can't write cab to boot

kaputtnik, 2018-08-08:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.1.0
daemon version: 1.1.0
compile-time dependency versions
        appstream-glib: 0.7.7
        gusb:   0.3.0
        efivar: 35
```

```shell
$ fwupdmgr get-devices
XPS 15 9560 TPM 2.0
  DeviceId:             cc2b222929056ab3e9a208df9f86b7d9a74ae514
  Guid:                 bc891321-25aa-54f6-86e7-633d065b8d5d
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              1.3.2.8
  Icon:                 computer
  Created:              2018-08-08

XPS 15 9560 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.9.4
  VersionLowest:        0.1.9.4
  Icon:                 computer
  Created:              2018-08-08

XPS 9560 Thunderbolt Controller
  DeviceId:             e9de4ca4554591ec0991e88ed15f05a9c3c6a573
  Guid:                 72533768-6a6c-5c06-994a-367374336810
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2018-08-08
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 2 seconds
BootOrder: 0001,0003
Boot0000* Windows Boot Manager  HD(1,GPT,d1851ce5-3079-49bd-bdc3-2edd154de628,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* grub  HD(1,GPT,31832684-9a3f-409c-b0fe-1ef6d20e510c,0x800,0xff801)/File(\EFI\grub\grubx64.efi)

```

```shell
$ efivar -l | grep fw
no output
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── arch
│   │   └── fw
│   └── grub
│       └── grubx64.efi
[...]
```

Please answer the following questions:

- Operating system and version: Arch LInux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

I got my trouble in getting fwupd work.
After installation I got only thunderbold as device. I fixed it with setting OverrideESPMountPoint=/boot in uefi.conf
After that I tried to update the firmware:
```fwupdmgr update -v
ignoring XPS 15 9560 TPM 2.0 [cc2b222929056ab3e9a208df9f86b7d9a74ae514] as not updatable
Downloading 0.1.10.1 for XPS 15 9560 System Firmware...
(fwupdmgr:9287): Fu-DEBUG: 06:56:23.570: creating path /home/nik/.cache/fwupdmgr
(fwupdmgr:9287): Fu-DEBUG: 06:56:23.608: skpping download as file already exists
Updating 0.1.10.1 on XPS 15 9560 System Firmware...
(fwupdmgr:9287): Fwupd-DEBUG: 06:56:23.612: Emitting ::status-changed() [decompressing]
Entpacken …            [ -                                     ](fwupdmgr:9287): Fwupd-DEBUG: 06:56:23.688: Emitting ::status-changed() [idle]
Entpacken …            [***************************************]
(fwupdmgr:9287): Fwupd-DEBUG: 06:56:23.784: Emitting ::status-changed() [waiting-for-auth]
Authenticating…        [  -                                    ](fwupdmgr:9287): Fwupd-DEBUG: 06:56:23.786: Emitting ::status-changed() [idle]
Authenticating…        [***************************************]
(fwupdmgr:9287): Fwupd-DEBUG: 06:56:23.813: Emitting ::status-changed() [scheduling]
Einplanen …            [  -                                    ](fwupdmgr:9287): Fwupd-DEBUG: 06:56:23.818: Emitting ::status-changed() [idle]
Einplanen …            [***************************************]
Can't write file »/boot/EFI/arch/fw/fwupd-34578c72-11dc-4378-bc7f-b643866f598c.cap.Q6R8MZ« File system is read-only. (translated by me)```

I got this error before, when fwupd tried to set the directories, so I made them.
It's enough free space on /boot and with root, I have permissions to write on boot-partition.

```# journalctl -u fwupd -b 0
-- Logs begin at Thu 2018-08-02 07:44:48 CEST, end at Wed 2018-08-08 07:01:03 CEST. --
Aug 08 06:44:13 venom systemd[1]: Starting Firmware update daemon...
Aug 08 06:44:13 venom fwupd[3042]: disabling plugin because: failed to startup dell_esrt: UEFI firmware already supported
Aug 08 06:44:13 venom fwupd[3042]: disabling plugin because: failed to startup redfish: no SMBIOS table
Aug 08 06:44:14 venom fwupd[3042]: using plugins: altos, dfu, dell, amt, steelseries, ebitdo, csr, upower, colorhug, udev, wacomhid, unifying, nitrokey, thunderbolt, synapticsmst, thunderbolt_power, uefi
Aug 08 06:44:14 venom fwupd[3042]: Daemon ready for requests
Aug 08 06:44:14 venom systemd[1]: Started Firmware update daemon.
Aug 08 06:44:20 venom fwupd[3042]: dfu failed to add USB device 1d6b:0003: 1d6b:0003 is not supported: USB error on device 1d6b:0003 : Entity not found [-5]
Aug 08 06:44:20 venom fwupd[3042]: dfu failed to add USB device 1d6b:0002: 1d6b:0002 is not supported: USB error on device 1d6b:0002 : Entity not found [-5]```

superm1, 2018-08-08:
Is the EFI system partition mounted rw?

hughsie, 2018-08-08:
`fwupd-34578c72-11dc-4378-bc7f-b643866f598c.cap.Q6R8MZ` -- _perhaps_ this is because `g_file_set_contents` is doing a tmpcreate+atomic_rename rather than just writing the file? I don't see why that would be different on VFAT but maybe it's worth testing.

kaputtnik, 2018-08-08:
I made a log with strace, maybe it is helpfull.
[fwupd.log](https://github.com/hughsie/fwupd/files/2270469/fwupd.log)

It seems, that ´protectsystem=full´ in the service file is the problem. When I comment it out, it looks like it works:
```fwupdmgr update     
ignoring XPS 15 9560 TPM 2.0 [cc2b222929056ab3e9a208df9f86b7d9a74ae514] as not updatable
No upgrades for XPS 9560 Thunderbolt Controller, current is 21.00: 21.00=same
Downloading 0.1.10.1 for XPS 15 9560 System Firmware...
Updating 0.1.10.1 on XPS 15 9560 System Firmware...
Decompressing?         [***************************************]
Authenticating?        [***************************************]
Scheduling?            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]:
```

Is it secure to run the update without this option?

hughsie, 2018-08-08:
Maybe we need to make:

    ReadWritePaths=/var/lib/fwupd /etc/fwupd/remotes.d -/boot/efi -/boot -/efi

into:

    ReadWritePaths=/var/lib/fwupd /etc/fwupd/remotes.d -/boot/efi -/boot -/efi -/boot/EFI

superm1, 2018-08-08:
Vfat is case insensitive though I thought

hughsie, 2018-08-08:
> Vfat is case insensitive

Sure, but is systemd when it applies the paths? I'd be surprised if it was. Do you think its worth adding `/boot/EFI` to the list of allowed ESP paths?

superm1, 2018-08-08:
@kaputtnik can you please try to add that path and see if it helps you?

kaputtnik, 2018-08-08:
sorry, I already updated the firmware.

Is it possible to make fwupd create the folders again?

superm1, 2018-08-08:
Yes you can manually download the cab file and install it using the install argument and `--allow-reinstall`

superm1, 2018-08-08:
Sorry to clarify - download manually from fwupd.org using your browser. Then install using

`fwupdmgr install --allow-reinstall file.cab`

kaputtnik, 2018-08-08:
Thank you.
After adding -/boot/EFI in fwupd.service, it runs without an error.

```
fwupdmgr install Downloads/9c9702786bd6883845c0a1c49216a9af735efe70-firmware_XPS9560_Precision5520_1.10.1.cab --allow-reinstall -v
(fwupdmgr:13376): Fwupd-DEBUG: 20:08:01.798: Emitting ::status-changed() [decompressing]
Decompressing?         [ -                                     ](fwupdmgr:13376): Fwupd-DEBUG: 20:08:01.877: Emitting ::status-changed() [idle]
Decompressing?         [***************************************]
(fwupdmgr:13376): Fwupd-DEBUG: 20:08:01.967: Emitting ::status-changed() [waiting-for-auth]
Authenticating?        [  -                                    ](fwupdmgr:13376): Fwupd-DEBUG: 20:08:01.968: Emitting ::status-changed() [idle]
Authenticating?        [***************************************]
(fwupdmgr:13376): Fwupd-DEBUG: 20:08:01.996: Emitting ::status-changed() [scheduling]
Scheduling?            [    \                                  ](fwupdmgr:13376): Fwupd-DEBUG: 20:08:02.104: Emitting ::status-changed() [idle]
Scheduling?            [***************************************]
(fwupdmgr:13376): Fwupd-DEBUG: 20:08:02.104: Emitting ::device-changed(f49999b6e3af5b4bf8ad1f2ed2922b76df457271)
(fwupdmgr:13376): Fwupd-DEBUG: 20:08:02.104: Emitting ::changed()
```
```
tree /boot/EFI 
/boot/EFI
├── arch
│   ├── fw
│   │   └── fwupd-34578c72-11dc-4378-bc7f-b643866f598c.cap
│   └── fwupdx64.efi
[...]
```

kaputtnik, 2018-08-08:
I think this can be closed. It should be pretty unique, that the efi-folder is written in caps. To be honest, I don't know, why mine is.

But I still do not understand, why I had to add `OverrideESPMountPoint=/boot` in uefi.conf. It should be in paths, right?

superm1, 2018-08-08:
Well if you broke it by the configuration you adopted and we can fix it via the systemd unit we may as well do that right?

Can you please do the following:
1. Add the extra path into systemd unit like you did.
2. Modify the ExecStart line to also add a `-v` to the daemon invocation
3. Remove your path override
4. Retry the update again

If that still fails please share the `journalctl -u fwupd.service` output.

kaputtnik, 2018-08-09:
You are right.

After adding `/boot/EFI`, I can remove the path override and it still works.
Adding this fixes the whole issue.

The last question was just for me, to understand it better: /boot was in the allowed paths before. Why did I have to add the path override though?

superm1, 2018-08-09:
Well that's good, thanks for that confirmation.

That part is still confusing to me too. Can you share all of your output of your `mount` command so we can look at the whole picture of mount options and maybe we can make sense of it

kaputtnik, 2018-08-09:
```
mount | grep boot /dev/nvme0n1p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro)
```

Do you need the whole output or is this enough?

superm1, 2018-08-09:
Nothing looks out of the ordinary on that.  I tried on my side with no changes to master, formatted my ESP and mounted exactly like you (same mount options and at `/boot`) with no failure.  It created the whole directory structure, put the files in the right place and everything.

Is it possible that you had a dirty vfat table from a bad shutdown and so it was mounted read only at some point maybe?  I don't think we should add `-/boot/EFI` unless we can properly characterize this (eg reproduce it).

superm1, 2018-08-09:
OK I understand what's happening here.

It's because of `ProtectSystem=full` and what it _specifically_ is doing.

Quoting from the documentation
> Takes a boolean argument or the special values "full" or "strict". If true, mounts the /usr and /boot directories read-only for processes invoked by this unit. If set to "full", the /etc directory is mounted read-only, too. If set to "strict" the entire file system hierarchy is mounted read-only, except for the API file system subtrees /dev, /proc and /sys (protect these directories using PrivateDevices=, ProtectKernelTunables=, ProtectControlGroups=). This setting ensures that any modification of the vendor-supplied operating system (and optionally its configuration, and local mounts) is prohibited for the service. It is recommended to enable this setting for all long-running services, unless they are involved with system updates or need to modify the operating system in other ways. If this option is used, ReadWritePaths= may be used to exclude specific directories from being made read-only. This setting is implied if DynamicUser= is set. For this setting the same restrictions regarding mount propagation and privileges apply as for ReadOnlyPaths= and related calls, see below. Defaults to off.

So `/boot` is a special cased path when this option is used.  Normally when adding the file it's added to `$ESP/EFI/$OS/fw` and `$ESP/EFI/$OS/fwupdx64.efi`.  

So when it fails for you, adding `-/boot/EFI` to the list works because that directory is then something that is treated just like `/boot/efi` is  (a sub directory).

Now as for what to do, I'm not sure yet.  It's ideal to keep `ProtectSystem` as this is a long running service and that's safer, but this is actually writing to `/boot` so maybe that option should be removed.  I'll experiment some and see what I can come up with.

superm1, 2018-08-09:
OK so this commit fixes the problem IIF `/boot/EFI` is created in advance.

```
diff --git a/data/meson.build b/data/meson.build
index 51384d47..69df5b48 100644
--- a/data/meson.build
+++ b/data/meson.build
@@ -40,7 +40,7 @@ if get_option('systemd')
   rw_directories += join_paths (localstatedir, 'lib', 'fwupd')
   rw_directories += join_paths (default_sysconfdir, 'fwupd', 'remotes.d')
   if get_option('plugin_uefi')
-    rw_directories += ['-/boot/efi', '-/boot', '-/efi']
+    rw_directories += ['-/boot/efi', '-/boot', '-/efi', '-/boot/EFI']
   endif
 
   dynamic_options = []
diff --git a/plugins/uefi/fu-uefi-common.c b/plugins/uefi/fu-uefi-common.c
index f87f15d6..1a519976 100644
--- a/plugins/uefi/fu-uefi-common.c
+++ b/plugins/uefi/fu-uefi-common.c
@@ -283,7 +283,7 @@ fu_uefi_check_esp_path (const gchar *path, GError **error)
                             "%s was not mounted", path);
                return FALSE;
        }
-       if (g_unix_mount_is_readonly (mount)) {
+       if (g_strcmp0 (path, "/boot") != 0 && g_unix_mount_is_readonly (mount)) {
                g_set_error (error,
                             FWUPD_ERROR,
                             FWUPD_ERROR_NOT_SUPPORTED,
```

If /boot/EFI isn't created in advance fwupd will start up properly but then fail like this when installing:

```
$ fwupdmgr install a3b9d8b67f3c047d54860a35c4f67b8691323886-Signed_1152921504627700266.cab --allow-reinstall
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
Failed to create '/boot/EFI/ubuntu/fw': Read-only file system
```

hughsie, 2018-08-09:
> `if (g_strcmp0 (path, "/boot") != 0 && g_unix_mount_is_readonly (mount)) {`

Can you expand a bit on that? I'm not sure I get that chunk.

superm1, 2018-08-09:
> Can you expand a bit on that? I'm not sure I get that chunk.

Basically when `ProtectSystem=full` is set, the process sees `/boot` as read only from systemd sandboxing.  Even if it's in `ReadWritePaths`, it is still treated as Read Only.  So it would need to be special cased in this instance to not fail the read only check.

I'm personally leaning on we drop `ProtectSystem=full`.

superm1, 2018-08-09:
The other option is to add a special case to check for if set to `/boot` see if `/boot/EFI` exists.  If not, fail with an `UpdateError` saying that `/boot/EFI` doesn't exist.

ArchangeGabriel, 2018-08-09:
Is there a case where you need write permissions for `/boot` itself and not a subdirectory?

If we assume they are three standard ESP mount points as `/boot`, `/boot/efi` and `/efi`, you should only need write permissions for the `EFI` subdir of those directories, right?

superm1, 2018-08-09:
Yes, If you don't make the directory structure `/boot/EFI` in advance, the sandboxed process won't have permission to create it.

I even tested creating it outside the sandbox after fwupd started up and it still doesn't get access until you restart the process and it existed in advance of starting.

ArchangeGabriel, 2018-08-09:
Can you boot an EFI system without an `EFI` folder on your ESP?

kaputtnik, 2018-08-09:
Afaik it is not problematic to rename EFI to efi.
So maybe it is enough to inform, that EFI makes these problems and to recommend to rename EFI to efi?

hughsie, 2018-08-09:
> it is not problematic to rename EFI.

I'm kinda erring on this -- I mean, are we going to support `/boot/UEFI_PARTITION` and any other weird thing? My gut instinct is we support enough ways to do this already, and if you've chosen to put your system together in a way that's different from everyone else it's okay if we just show a warning telling the user it's not going to work.

commit 825fd28a (2018-08-09): uefi: Handle /boot properly (Closes: #627)

/boot is a special cased directory when using ProtectSystem=full

Due to this, it's marked read only even if it's listed in ReadWritePaths.

Allow folks to use this for their ESP, but they need to create /boot/EFI
in advance of starting fwupd.
commit 9bdbbc5c (2018-08-09): uefi: Handle /boot properly (Closes: #627)

/boot is a special cased directory when using ProtectSystem=full

Due to this, it's marked read only even if it's listed in ReadWritePaths.

Allow folks to use this for their ESP, but they need to create /boot/EFI
in advance of starting fwupd.
closed 2018-08-09 by superm1