fwupd/fwupd #631: _FORTIFY_SOURCE redefined while compiling on Gentoo with Portage

RedL0tus, 2018-08-08:
It's a problem with compiling.

Please answer the following questions:

- Operating system and version: Gentoo Base System release 2.6
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Portage
- Have you tried rebooting? No. It's a problem with compiling.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

The compiling failed on Gentoo at: 
```
[166/184] gcc -c ../fwupd-1.1.0/plugins/uefi/efi/fwupdate.c -o plugins/uefi/efi/fwupdate.o -Og -g3 -Wp,-D_FORTIFY_SOURCE=2 --param=ssp-buffer-size=4 -fexceptions -Wall -Werror -Wextra -std=gnu11 -fpic -fshort-wchar -ffreestanding -fno-strict-aliasing -fno-stack-protector -
fno-stack-check -fno-merge-constants -Wsign-compare -Wno-missing-field-initializers -grecord-gcc-switches '-DDEBUGDIR="/usr/lib64/debug"' -isystem /usr/include/efi -isystem /usr/include/efi/x86_64 -mno-red-zone -mno-sse -mno-mmx -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI
FAILED: plugins/uefi/efi/fwupdate.o 
gcc -c ../fwupd-1.1.0/plugins/uefi/efi/fwupdate.c -o plugins/uefi/efi/fwupdate.o -Og -g3 -Wp,-D_FORTIFY_SOURCE=2 --param=ssp-buffer-size=4 -fexceptions -Wall -Werror -Wextra -std=gnu11 -fpic -fshort-wchar -ffreestanding -fno-strict-aliasing -fno-stack-protector -fno-stack-
check -fno-merge-constants -Wsign-compare -Wno-missing-field-initializers -grecord-gcc-switches '-DDEBUGDIR="/usr/lib64/debug"' -isystem /usr/include/efi -isystem /usr/include/efi/x86_64 -mno-red-zone -mno-sse -mno-mmx -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI
<command-line>: error: "_FORTIFY_SOURCE" redefined [-Werror]
<built-in>: note: this is the location of the previous definition
cc1: all warnings being treated as errors
```

The problem seems to be Gentoo's GCC has built-in parameter that sets `-D_FORTIFY_SOURCE` to another value, and the build system set that value again.

My solution is to add `-U_FORTIFY_SOURCE` before `-D_FORTIFY_SOURCE=2`.
I believe there will be better solution so I opened this issue.

Link to Gentoo Bugzilla: https://bugs.gentoo.org/661068

hughsie, 2018-08-08:
I guess it is slightly odd to unconditionally set _FORTIFY_SOURCE -- @superm1 any ideas here? My gut instinct is just to remove it from the `meson.build` file.

superm1, 2018-08-08:
`compiler.has_flags` maybe? If that doesn't work then yeah I agree remove it.

hughsie, 2018-08-08:
> compiler.has_flags maybe

I don't think that'll work; it's not really a flag it's just a #define. I'll remove it.

closed 2018-08-08 by hughsie