fwupd/fwupd #633: Update monitor firmware

paulmenzel, 2018-08-08:
Do you know if there is a common standard to update monitor firmware? For example, the monitor Dell U2718Q currently can only [be updated using an EXE file for MIcrosoft Windows](https://www.dell.com/support/home/de/de/debsdt1/Drivers/DriversDetails?driverid=DV984).

superm1, 2018-08-08:
Depends on what this particular firmware is actually updating. Many monitors are little computers now with USB hubs, MST controllers, embedded controllers. So it can possibly be across a variety of buses.

superm1, 2018-08-09:
Just looking at the update executable I can see a DLL file that is supposed to be used for a USB bridge to I2C, so I anticipate this update is installed over USB.  I suppose you may be able to reverse engineer it would be to capture the update process from Windows using Wireshark.

hughsie, 2018-09-12:
@superm1 on the assumption I might have found some time to reverse engineer this, I've got two questions:

* Could Dell loan me U2718Q for a few months?
* On the assumption we succeed and the flash can be done via fwupd, would Dell be able to upload the firmware to the LVFS?

If the second point is a "probably not" then I guess there's little point doing the RE work.

Happy to switch to a less public forum if required.