fwupd/fwupd #636: Maybe get rid of vim header in all source files?

hughsie, 2018-08-09:
Every source file has: ` -*- Mode: C; tab-width: 8; indent-tabs-mode: t; c-basic-offset: 8 -*-` at the top. This was cargo-culted from my other projects where co-maintainers used vim. I don't think they're super useful, and I was going to suggest getting rid of them in new files. Comments?

superm1, 2018-08-09:
No concerns, I don't use em. To can lift then from all files in my mind. Shouldn't make backporting any harder since it's a comment and top of the file.

closed 2018-08-09 by hughsie

ArchangeGabriel, 2018-08-09:
Note that if you are ever interested into this, there is a way to provide project-wide config for all files (including different config for different types of file) for both vim and emacs in a dedicated file instead of putting things in each file. ;)