fwupd/fwupd #646: Support for Huion tablets

ozeszty, 2018-08-10:
I noticed that there's no Huion entry on [Vendor Status page](https://fwupd.org/vendorlist), and since they started supporting DIGImend, it might be worth reaching out to them. Currently firmware update requires windows: https://github.com/DIGImend/digimend-kernel-drivers/issues/181

hughsie, 2018-08-10:
I've added them here: https://fwupd.org/lvfs/vendorlist#huion

Can you open a bug against DIGImend perhaps? We need to get someone from Huion aware of the issue.

closed 2018-08-10 by hughsie

ozeszty, 2018-08-10:
Firmware update is out of DIGImend's scope, you want it's community to draw Huion's attention to fwupd?

hughsie, 2018-08-10:
Ahh, sorry. I thought DIGImend was provided by Huion. I guess it's more likely to be noticed by somebody at Huion there.

ozeszty, 2018-08-11:
@spbnick (project lead) probably has some contacts in Huion.