fwupd/fwupd #649: T480s not recognized

xaitax, 2018-08-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.1.1
daemon version:	1.0.6
checkout info:	1.1.0-89-g7343ed4f
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             16324df8d782cbc6d3396541199c8fc397182242
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2018-08-10

Thinkpad T480s Thunderbolt Controller
  DeviceId:             a8a0631cf1c7269fb1406b30575e2f4d2dff3872
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              12.02
  Icon:                 computer
  Created:              2018-08-10

```

Please answer the following questions:

- Operating system and version: 
Linux Mint 19
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
from source
- Have you tried rebooting?
yes
- Are you using an NVMe disk?
Not that I know of ...
- Is secure boot enabled (only for the UEFI plugin)?
no

xaitax, 2018-08-10:
Another output:

```shell
inxi -Fxz
System:    Host: w00t Kernel: 4.17.13-041713-generic x86_64 bits: 64 gcc: 8.2.0
           Desktop: Cinnamon 3.8.8 (Gtk 2.24.32) Distro: Linux Mint 19 Tara
Machine:   Device: laptop System: LENOVO product: 20L7CTO1WW v: ThinkPad T480s serial: N/A
           Mobo: LENOVO model: 20L7CTO1WW v: SDK0J40709 WIN serial: N/A
           UEFI [Legacy]: LENOVO v: N22ET46W (1.23 ) date: 06/19/2018
Battery    BAT0: charge: 50.6 Wh 96.5% condition: 52.5/57.0 Wh (92%) model: SMP 01AV480 status: N/A
           hidpp__0: charge: N/A condition: NA/NA Wh model: Logitech Wireless Mouse M335 status: Discharging
CPU:       Quad core Intel Core i7-8550U (-MT-MCP-) arch: Kaby Lake rev.10 cache: 8192 KB
           flags: (lm nx sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx) bmips: 15936
           clock speeds: max: 4000 MHz 1: 800 MHz 2: 800 MHz 3: 800 MHz 4: 800 MHz 5: 800 MHz 6: 800 MHz
           7: 800 MHz 8: 800 MHz
Graphics:  Card: Intel UHD Graphics 620 bus-ID: 00:02.0
           Display Server: x11 (X.Org 1.19.6 ) drivers: modesetting (unloaded: fbdev,vesa)
           Resolution: 1920x1080@60.01hz
           OpenGL: renderer: Mesa DRI Intel UHD Graphics 620 (Kabylake GT2)
           version: 4.5 Mesa 18.3.0-devel Direct Render: Yes
Audio:     Card Intel Sunrise Point-LP HD Audio driver: snd_hda_intel bus-ID: 00:1f.3
           Sound: Advanced Linux Sound Architecture v: k4.17.13-041713-generic
Network:   Card-1: Intel Ethernet Connection (4) I219-V driver: e1000e v: 3.2.6-k bus-ID: 00:1f.6
           IF: eth0 state: down mac: <filter>
           Card-2: Intel Wireless 8265 / 8275 driver: iwlwifi bus-ID: 3d:00.0
           IF: wlan0 state: up mac: <filter>
Drives:    HDD Total Size: 1024.2GB (26.5% used)
           ID-1: /dev/nvme0n1 model: SAMSUNG_MZVLB1T0HALR size: 1024.2GB
Partition: ID-1: / size: 893G used: 230G (28%) fs: ext4 dev: /dev/dm-1
           ID-2: /boot size: 704M used: 91M (14%) fs: ext4 dev: /dev/nvme0n1p1
           ID-3: swap-1 size: 25.27GB used: 0.00GB (0%) fs: swap dev: /dev/dm-2
RAID:      No RAID devices: /proc/mdstat, md_mod kernel module present
Sensors:   System Temperatures: cpu: 39.0C mobo: N/A
           Fan Speeds (in rpm): cpu: 0
Info:      Processes: 284 Uptime: 11 min Memory: 1945.4/23637.2MB Init: systemd runlevel: 5 Gcc sys: 7.3.0
           Client: Shell (bash 4.4.191) inxi: 2.3.56

xaitax, 2018-08-10:
Meaning, I am on FW version 1.23 and according to firmware.xml.gz there is an update to 1.25. Which is also listed here:
https://fwupd.org/lvfs/device/ebfe8df8-dee7-4692-a721-cbcf5095c5cf

hughsie, 2018-08-10:
> client version:	1.1.1
> daemon version:	1.0.6

You need to either restart your computer or restart the fwupd service if you've just done a major upgrade.

xaitax, 2018-08-10:
Thanks, did all this now. For some reason he didn't overwrite the apt version of fwupd from build/src/fwupd

Regardless, unfortunately still the same :/

```
 xaitax  ~  sudo fwupdmgr --version
client version:	1.1.1
daemon version:	1.1.1
checkout info:	1.1.0-89-g7343ed4f
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	efivar:	34
 xaitax  ~  sudo fwupdmgr get-devices
Unifying Receiver
  DeviceId:             16324df8d782cbc6d3396541199c8fc397182242
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2018-08-10

Thinkpad T480s Thunderbolt Controller
  DeviceId:             a8a0631cf1c7269fb1406b30575e2f4d2dff3872
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              12.02
  Icon:                 computer
  Created:              2018-08-10

 xaitax  ~  sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

 xaitax  ~  sudo fwupdmgr get-updates
No upgrades for Unifying Receiver, current is RQR12.07_B0029: RQR12.07_B0029=same
 xaitax  ~  sudo fwupdmgr update
No upgrades for Unifying Receiver, current is RQR12.07_B0029: RQR12.07_B0029=same
 xaitax  ~  
```

hughsie, 2018-08-10:
What does `fwupdtool get-plugins` report? What's the output from `fwupd --verbose` when run as root? Thanks.

xaitax, 2018-08-10:
Ran all of it with root now:

```
root@w00t /usr/lib/fwupd $ l
total 604K
drwxr-xr-x   3 root   root   4,0K Aug 10 14:57 .
drwxr-xr-x 161 xaitax xaitax  12K Aug 10 14:57 ..
drwxr-xr-x   2 root   root   4,0K Aug 10 14:57 efi
-rwxr-xr-x   1 root   root   242K Aug 10 14:44 fwupd
-rwxr-xr-x   1 root   root    94K Aug 10 14:44 fwupdate
-rwxr-xr-x   1 root   root   242K Aug 10 14:44 fwupdtool

root@w00t /usr/lib/fwupd $ ./fwupdtool get-plugins
altos
amt
colorhug
csr
dell
dell_esrt
dfu
ebitdo
nitrokey
redfish
steelseries
synapticsmst
thunderbolt
thunderbolt_power
udev
uefi
unifying
upower
wacomhid

 root@w00t /usr/lib/fwupd $ ./fwupd --verbose
13:06:31:0857 Fu  Verbose debugging enabled (on console 1)
13:06:31:0857 As  run FuEngine:load
13:06:31:0857 Fu  loading config values from /etc/fwupd/daemon.conf
13:06:31:0858 Fwupd looking for os-release at /etc/os-release
13:06:31:0858 As  run AsStore:load-installed{/usr/share/fwupd/metainfo}
13:06:31:0858 As  adding existing file: /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs-testing.metainfo.xml
13:06:31:0858 As  adding existing file: /usr/share/fwupd/metainfo/org.freedesktop.fwupd.remotes.lvfs.metainfo.xml
13:06:31:0859 As  Emitting ::changed() [load-installed]
13:06:31:0859 Fu  using config path of /etc/fwupd
13:06:31:0859 Fu  loading config from /etc/fwupd/remotes.d/lvfs.conf
13:06:31:0859 Fu  loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
13:06:31:0859 Fu  loading config from /etc/fwupd/remotes.d/fwupd.conf
13:06:31:0859 Fu  loading config from /etc/fwupd/remotes.d/vendor.conf
13:06:31:0859 Fu  using config path of /usr/lib/fwupd
13:06:31:0859 Fu  ordering lvfs=fwupd+1
13:06:31:0859 Fu  ordering lvfs-testing=lvfs+1
13:06:31:0859 As  run FuEngine:load-smbios
13:06:31:0859 As  run FuEngine:load-hwids
13:06:31:0859 Fu  smbios property Manufacturer=LENOVO
13:06:31:0859 Fu  smbios property EnclosureKind=a
13:06:31:0859 Fu  smbios property Family=ThinkPad T480s
13:06:31:0859 Fu  smbios property ProductName=20L7CTO1WW
13:06:31:0859 Fu  smbios property ProductSku=LENOVO_MT_20L7_BU_Think_FM_ThinkPad T480s
13:06:31:0859 Fu  smbios property BiosVendor=LENOVO
13:06:31:0859 Fu  smbios property BiosVersion=N22ET46W (1.23 )
13:06:31:0859 Fu  smbios property BiosMajorRelease=01
13:06:31:0859 Fu  smbios property BiosMinorRelease=17
13:06:31:0859 Fu  smbios property BaseboardManufacturer=LENOVO
13:06:31:0859 Fu  smbios property BaseboardProduct=20L7CTO1WW
13:06:31:0859 Fu  {8dbe4d19-425e-575c-94c5-95f48c4817cb}   <- LENOVO&ThinkPad T480s&20L7CTO1WW&LENOVO_MT_20L7_BU_Think_FM_ThinkPad T480s&LENOVO&N22ET46W (1.23 )&01&17
13:06:31:0859 Fu  {01360fe9-d014-56d2-9abd-dabd95a58785}   <- LENOVO&ThinkPad T480s&20L7CTO1WW&LENOVO&N22ET46W (1.23 )&01&17
13:06:31:0859 Fu  {d111cf67-4fa7-5ba6-8653-f64e29284847}   <- LENOVO&20L7CTO1WW&LENOVO&N22ET46W (1.23 )&01&17
13:06:31:0860 Fu  {3e99442b-0a06-5579-a9d4-e103994f8c04}   <- LENOVO&ThinkPad T480s&20L7CTO1WW&LENOVO_MT_20L7_BU_Think_FM_ThinkPad T480s&LENOVO&20L7CTO1WW
13:06:31:0860 Fu  {e4d5ce60-d07c-52ac-bd0b-d15b21a3d37c}   <- LENOVO&ThinkPad T480s&20L7CTO1WW&LENOVO_MT_20L7_BU_Think_FM_ThinkPad T480s
13:06:31:0860 Fu  {f63bdf26-79bc-5f04-9fb8-d5cf0e5c95a6}   <- LENOVO&ThinkPad T480s&20L7CTO1WW
13:06:31:0860 Fu  {33b1ff56-cffd-5685-899d-517b941e7899}   <- LENOVO&LENOVO_MT_20L7_BU_Think_FM_ThinkPad T480s&LENOVO&20L7CTO1WW
[... 251 lines cut ...]
13:06:36:0851 Fu  Daemon ready for requests
13:06:36:0851 Fu  emit added from udev: 8de6c7959053fd5798006dcc63590d33fa5e51cb
13:06:36:0851 Fu  performing device_added() on thunderbolt_power
13:06:36:0851 Fu  ::added 8de6c7959053fd5798006dcc63590d33fa5e51cb
13:06:36:0886 Unifying [HID] device->host:20 01 02 00 00 f9 4f 00 00 00 00 00 00 00 00 
13:06:37:0050 Fu  registering D-Bus service
###################################################################################### @4991ms FuEngine:load 4991ms
#                                                                                      @0003ms FuEngine:load-quirks 1ms
#                                                                                      @0013ms FuEngine:load-plugins 6ms
#                                                                                      @0016ms FuEngine:setup 2ms
#                                                                                      @0015ms FuEngine:setup{upower} 1ms
#####################################################################################  @4990ms FuEngine:coldplug 4974ms
#                                                                                      @0019ms FuEngine:coldplug{udev} 3ms
#                                                                                      @0019ms FuPluginUdev:coldplug{pci} 3ms
#                                                                                      @0027ms FuEngine:coldplug{unifying} 8ms
############################################################################           @4454ms FuEngine:coldplug{thunderbolt} 4426ms
                                                                            #########  @4990ms FuEngine:coldplug{synapticsmst} 535ms
13:06:37:0062 Fu  FuMain: acquired name: org.freedesktop.fwupd
13:06:41:0890 Unifying [HID] device->host:20 01 02 00 00 f6 4f 00 00 00 00 00 00 00 00 
13:06:46:0896 Unifying [HID] device->host:20 01 02 00 00 f8 5f 00 00 00 00 00 00 00 00 
13:06:51:0901 Unifying [HID] device->host:20 01 02 00 00 f9 3f 00 00 00 00 00 00 00 00 
13:06:56:0906 Unifying [HID] device->host:20 01 02 00 00 f9 3f 00 00 00 00 00 00 00 00 
```

Last line just stays the repeats

hughsie, 2018-08-10:
> 13:06:31:0873 Fu  disabling plugin because: failed to startup uefi: kernel efivars support missing: /sys/firmware/efi/efivars

Does `/sys/firmware/efi/efivars` exist for you? Are you running a custom kernel or something?

ArchangeGabriel, 2018-08-10:
Are you booting in UEFI mode and not BIOS/Legacy mode?

xaitax, 2018-08-10:
Thanks a lot for your prompt replies, Richard!

It doesn't exist here:
```
 root@w00t  /usr/lib/fwupd  l /sys/firmware/
total 0
drwxr-xr-x  5 root root 0 Aug 10 15:26 .
dr-xr-xr-x 13 root root 0 Aug 10 15:26 ..
drwxr-xr-x  5 root root 0 Aug 10 15:26 acpi
drwxr-xr-x  3 root root 0 Aug 10 15:26 dmi
drwxr-xr-x 26 root root 0 Aug 10 15:26 memmap
```
Not a custom kernel, no. Grabbing the last mainlines usually from here:
http://kernel.ubuntu.com/~kernel-ppa/mainline/
Since Mint (19) is based on Ubuntu 18.04.

Also:
```
xaitax $ dpkg -l | grep efi
ii  automake                                          1:1.15.1-3ubuntu2                             all          Tool for generating GNU Standards-compliant Makefiles
ii  efibootmgr                                        15-1                                          amd64        Interact with the EFI Boot Manager
ii  efitools                                          1.4.2+git20140118-0ubuntu2                    amd64        useful tools for manipulating UEFI secure boot platforms
ii  efivar                                            34-1                                          amd64        Tools to manage UEFI variables
ii  gnu-efi                                           3.0.4-2                                       amd64        Library for developing EFI applications
ii  libefiboot-dev:amd64                              34-1                                          amd64        Development headers for libefiboot
ii  libefiboot1:amd64                                 34-1                                          amd64        Library to manage UEFI variables
ii  libefivar-dev:amd64                               34-1                                          amd64        Development headers for libefivar
ii  libefivar1:amd64                                  34-1                                          amd64        Library to manage UEFI variables
```

When I ran `meson build` he first complained about missing debs (such as header files and efi stuff) but I installed them all and it ran through smoothly in the end.

xaitax, 2018-08-10:
@ArchangeGabriel
"Legacy Only" and Secure Boot off.

hughsie, 2018-08-10:
Legacy only isn't going to work as that turns off all UEFI (and thus UpdateCapsule) support. You need to boot in UEFI mode to update, sorry.

closed 2018-08-10 by hughsie