fwupd/fwupd #655: Logitech Unifying Receiver Bricked After Failed Firmware Update

sagie-d, 2018-08-12:
I was running Live Ubuntu 18.04 LTS, and the Ubuntu Software Tool suggested me to update the Unifying Receiver's firmware, so I let it do that (I clicked the "update" button in the app's GUI).
The app reported an error, and the mouse connected to it (which I was using) stopped responding, so I went to the terminal to check out the matter:

```
ubuntu@ubuntu:~$ fwupdmgr --version
client version:    1.0.6
daemon version:    1.0.6
compile-time dependency versions
    appstream-glib:    0.7.4
    gusb:    0.2.11
    fwupdate:    10
    efivar:    34

ubuntu@ubuntu:~$ fwupdmgr get-devices

                 . . .
   ( output for irrelevant devices snipped )
                 . . .

Unifying [bootloader-texas]
  DeviceId:             868156f608e395b80eab11486389a2103e52a265
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.xx_Bxxxx
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  Created:              2018-08-11

________________________________________________

Devices that were not updated correctly:

 • Unifying Receiver (RQR24.01_B0023 → RQR24.05_B0029)
```

I tried to restart the update, no dice:
```
ubuntu@ubuntu:~$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

ubuntu@ubuntu:~$ sudo fwupdmgr update
No upgrades for Unifying [bootloader-texas], current is RQR24.xx_Bxxxx: RQR24.05_B0029=older
```
I tried to enable testing firmwares update (via ``/etc/fwupd/remotes.d/lvfs-testing.conf``), again - no dice:
```
ubuntu@ubuntu:~$ sudo fwupdmgr refresh --verbose
(fwupdmgr:7684): Fu-DEBUG: 22:28:37.121: creating path /home/ubuntu/.cache/fwupdmgr
(fwupdmgr:7684): Fwupd-DEBUG: 22:28:37.121: looking for os-release at /etc/os-release
(fwupdmgr:7684): Fu-DEBUG: 22:28:37.122: downloading https://cdn.fwupd.org/downloads/firmware-testing.xml.gz to /home/ubuntu/.cache/fwupdmgr/lvfs-testing-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
(fwupdmgr:7684): Fu-DEBUG: 22:28:37.754: progress: 50%
Downloading…           [*******************                    ](fwupdmgr:7684): Fu-DEBUG: 22:28:37.754: progress: 100%
Downloading…           [***************************************]
(fwupdmgr:7684): Fu-DEBUG: 22:28:37.755: downloading https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc to /home/ubuntu/.cache/fwupdmgr/lvfs-testing-metadata.xml.gz.asc
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc
(fwupdmgr:7684): Fu-DEBUG: 22:28:37.847: progress: 100%

(fwupdmgr:7684): Fu-DEBUG: 22:28:37.920: creating path /home/ubuntu/.cache/fwupdmgr
(fwupdmgr:7684): Fu-DEBUG: 22:28:37.920: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/ubuntu/.cache/fwupdmgr/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:7684): Fu-DEBUG: 22:28:38.015: progress: 16%
Downloading…           [******                                 ](fwupdmgr:7684): Fu-DEBUG: 22:28:38.015: progress: 34%
Downloading…           [*************                          ](fwupdmgr:7684): Fu-DEBUG: 22:28:38.104: progress: 52%
Downloading…           [********************                   ](fwupdmgr:7684): Fu-DEBUG: 22:28:38.104: progress: 71%
Downloading…           [***************************            ](fwupdmgr:7684): Fu-DEBUG: 22:28:38.106: progress: 89%
Downloading…           [**********************************     ](fwupdmgr:7684): Fu-DEBUG: 22:28:38.106: progress: 100%
Downloading…           [***************************************]
(fwupdmgr:7684): Fu-DEBUG: 22:28:38.107: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.asc to /home/ubuntu/.cache/fwupdmgr/lvfs-metadata.xml.gz.asc
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
(fwupdmgr:7684): Fu-DEBUG: 22:28:38.200: progress: 100%

ubuntu@ubuntu:~$ sudo fwupdmgr update --verbose
No upgrades for Unifying [bootloader-texas], current is RQR24.xx_Bxxxx: RQR24.03_B0027=older, RQR24.05_B0029=older

```
I tried disconnecting and reconnecting the receiver, turning off and on the paired mouse, etc, to no avail.
I switched to Windows, the Logitech Unifying Software shows the unifying receiver, but it is not functional. It shows firmware version 000.000.00000, no devices paired (although it used to be paired with a mouse), and the "pair a new device" button is disabled.
The only action possible is "update firmware", but it fails with an error.

Tried restarting the Live Ubuntu, and restarting the update, but everything looks exactly the same, except for the "Devices that were not updated correctly" section of the get-devices output, which is now gone:

```
ubuntu@ubuntu:~$ sudo fwupdmgr get-devices

                 . . .
   ( output for irrelevant devices snipped )
                 . . .

Unifying [bootloader-texas]
  DeviceId:             868156f608e395b80eab11486389a2103e52a265
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.xx_Bxxxx
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  Created:              2018-08-11

ubuntu@ubuntu:~$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

ubuntu@ubuntu:~$ sudo fwupdmgr update
No upgrades for Unifying [bootloader-texas], current is RQR24.xx_Bxxxx: RQR24.05_B0029=older

                    ( enabled /etc/fwupd/remotes.d/lvfs-testing.conf )

ubuntu@ubuntu:~$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware-testing.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware-testing.xml.gz.asc

Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

ubuntu@ubuntu:~$ sudo fwupdmgr update
No upgrades for Unifying [bootloader-texas], current is RQR24.xx_Bxxxx: RQR24.03_B0027=older, RQR24.05_B0029=older
```

Why did the update fail?
Is there some way I can revive the receiver?

Thanks in advance.

sagie-d, 2018-08-12:
I don't recall being offered a choice to report anything.

Perhaps I should mention - at the time I did the update I had two unifying receivers connected, each of them connected to a different device - not sure if this made a difference.

I managed to revive the bricked receiver, by the way, by running SecureDFU in Windows, after reading about the tool in one of the other issues, but I think fwupd should've detected the nonfunctional state the receiver was in, and reattempt the firmware programming when I ran the update command.

hughsie, 2018-09-12:
We do that in the recently released 1.0.9 and 1.1.2.

closed 2018-09-12 by hughsie

btoconnor, 2018-11-02:
Thanks for the quick reply. It's hard for me to boot into a LiveUSB image now because I don't actually have a keyboard that I can use - I need to hit keys in order to get the boot menu, and I don't have a spare keyboard accessible.

I just ran `fwupdmgr get-devices` and it showed the following:

```
Devices that were not updated correctly:

 • Unifying [bootloader-nordic] (RQR12.01_B0019 → RQR12.07_B0029)

Upload report now? (Requires internet connection) [Y|n]: y
```

So I went forward with the crash upload.  I'm not sure if you see that or if it's helpful.  I have access to a macbook that can SSH into the Ubuntu machine, it that helps with anything.

superm1, 2018-11-02:
I think since this is Ubuntu installing snap will be easiest to get new version.

1. `# sudo apt remove fwupd`
2. `# sudo snap install fwupd --classic`

btoconnor, 2018-11-02:
Thanks.  I ran `sudo snap install fwupd --classic` . and now `--version` is outputting this:
```
$ fwupdmgr --version
client version:	1.0.6
daemon version:	1.1.3
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34```

Does this look correct?  If so, what are the steps to update the firmware?  I'm very new to this software package after only discovering it today after the failed update.

superm1, 2018-11-02:
It looks to me like you didn't remove the deb version before installing the snap.
PATH is adjusted when snap is installed to include `/snap/bin/` but it's priority is below `/usr/bin/`.

So either run `/snap/bin/fwupdmgr` or remove the snap, remove the deb, reinstall the snap.

btoconnor, 2018-11-02:
Okay:

```
$ /snap/bin/fwupdmgr --version
client version:	1.1.3
checkout info:	1.1.3
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.3
```

It seems like I can figure out how to make 1.1.3 the canonical version after my keyboard is restored, so for now I'll run it from `/snap/bin/`.

Is it possible to walk me through the steps to fix the firmware?

Thanks again for your prompt replies.

btoconnor, 2018-11-02:
Okay, tried that, but got a weird error during the `update` command:

```
$ /snap/bin/fwupdmgr update
Downloading RQR12.07_B0029 for Unifying Receiver...
Fetching firmware https://fwupd.org/downloads/938fec082652c603a1cdafde7cd25d76baadc70d-Logitech-Unifying-RQR12.07_B0029.cab
Downloading?           [***************************************]
Decompressing?         [***************************************]
Authenticating?        [***************************************]
Failed to obtain auth
```

superm1, 2018-11-02:
Can you try running it as sudo?

btoconnor, 2018-11-02:
Yes! Thank you! That worked (of course I needed sudo, sigh).

The keyboard is now functional again. I ran `get-devices` afterwards just to see what the status was, and it prompted me to send in a report saying something failed.  I uploaded it, not sure if it was helpful, but the keyboard definitely works again.

Thank you a ton for your prompt help, much appreciated.

superm1, 2018-11-02:
Phew, that's great to hear, thanks!

@hughsie is it worth bumping up the logitech stuff from this scenario on LVFS to require newer fwupd client?  Maybe 1.0.9 or later?

hughsie, 2018-11-02:
> is it worth bumping up the logitech stuff from this scenario on LVFS to require newer fwupd 

I don't think so; only a very small number of failures has been reported. The old code mostly works, but the new code is much more tolerant of failures.

AnessZurba, 2018-11-21:
Hi @hughsie, 
Thanks for helping people out here :)
I ran into a similar issue today, I'm running the latest version of fwupd 

`$ /snap/bin/fwupdmgr --version
client version:	1.1.4
checkout info:	1.1.4
compile-time dependency versions
	appstream-glib:	0.7.9
	gusb:	0.3.0
	efivar:	35
daemon version:	1.1.4
`
whenever I try to flash a firmware to my unifying receiver I get the following error:

`$ /snap/bin/fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading?           [*                                      ] Less than one mDownloading?           [**                                     ] Less than one mDownloading?           [***                                    ] Less than one mDownloading?           [*****                                  ] Less than one mDownloading?           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

$ /snap/bin/fwupdmgr update
Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
`

What can I do to solve this?

Thanks in advance.

hughsie, 2018-11-21:
@AnessZurba what U00000 number is laser etched onto the USB plug?

AnessZurba, 2018-11-21:
`M/N:C-U0007`

actorius, 2018-11-24:
Have the save problem, but i can't get listed receiver on 
`sudo fwupdmgr get-devices
`
lsusb
`Bus 002 Device 005: ID 046d:aaaa Logitech, Inc. 
`

alxpl, 2018-12-28:
Hello Richard,

I just got a standalone U0012 (046d:c52b) and I see this error message in F29 with fwupdmgr-1.1.4-1:
`Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.03_B0009]
`

```
Unifying Receiver
  DeviceId:             fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.03_B0009
  Icon:                 preferences-desktop-keyboard
  Created:              2018-12-28
```

Unless I am mistaken, the firmware version is more recent than that uploaded to LVFS by Logitech, right?

Happy new year!

didierm, 2019-07-11:
Reproducible with latest F30 fwupd v1.2.9 : 
```
# rpm -q fwupd
fwupd-1.2.9-1.fc30.x86_64

# fwupdmgr --version
client version:	1.2.9
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.9

# fwupdmgr get-topology
○
├─ Latitude 7200 2-in-1 Thunderbolt Controller 0f210aff1e51d7265027ad39878ac7604ee331f8
├─ Latitude 7200 2-in-1 System Firmware      b3e092475802acdd28548bc8ee1c25877fd48a4c
├─ PC SN520 NVMe WDC 512GB                   6cc1c2d37a98d92020a9c80f9cc15d88a0656384
└─ Unifying Receiver                         0e83f7258bea62fcd41de1c03fac7f9ce2a8b714

# fwupdmgr get-devices
...
Unifying Receiver
  DeviceId:             0e83f7258bea62fcd41de1c03fac7f9ce2a8b714
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4 <- HIDRAW\VEN_046D&DEV_C52B
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.09_B0030
  VersionBootloader:    BOT01.04_B0016
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-07-09

# fwupdmgr update
...
No releases found for device: Not compatible with bootloader version: failed predicate [BOT01.0[0-3]_* regex BOT01.04_B0016]
```