fwupd/fwupd #656: UEFI update on Thinkpad T480 fails

itoijala, 2018-08-12:
I am trying to update the UEFI firmware of my Thinkpad T480 from 1.14 to 1.16.
`fwupdmgr update` succeeds and asks to reboot.
Upon reboot, the UEFI updater is started and outputs the path of the capsule file.
Then the Lenovo logo flashes on the screen and the computer reboots back to Linux, without updating the firmware.
UEFI reports that the version is still 1.14.
After the reboot, the capsule file has been deleted. The output below is from a run of `fwupdmgr` before rebooting. The only difference is the presence of the capsule.

```shell
$ fwupdmgr --version
client version:	1.1.0
daemon version:	1.1.0
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
T480 Thunderbolt Controller
  DeviceId:             523177911f1fb95f5cf3e4f71fa4e5ce70c172b5
  Guid:                 fe328c2d-5300-597b-a21d-991116db8ab6
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2018-08-12

20L5CTO1WW System Firmware
  DeviceId:             48477ba14bd9bd0c97323a07caa54b7766f917c2
  Guid:                 dc7f0308-1ef8-4774-9ba5-89a58c4d731c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.14
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2018-08-12
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             5decfff65e3391259bca4003a090e59bf50e3bea
  Guid:                 dd31d983-cf3b-4c84-acc9-70caa94e827b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              184.50.3460
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-12

UEFI Device Firmware
  DeviceId:             c19580db344e418cf2b23297bcf46d75c06e432a
  Guid:                 b87a926d-189c-49a1-b1ff-921099b1de89
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.9
  VersionLowest:        0.1.9
  Icon:                 audio-card
  Created:              2018-08-12
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,001C,0017,0018,0019,001A,001B,001D,001E,001F,0024
Boot0000* Windows Boot Manager	HD(1,GPT,f2b839f5-aaee-7042-ba68-1317b0acf076,0x800,0x200000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...M................
Boot0001* GRUB	HD(1,GPT,f2b839f5-aaee-7042-ba68-1317b0acf076,0x800,0x200000)/File(\EFI\GRUB\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019  NVMe1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001A  NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001B  ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001C  USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
```

```shell
$ tree /boot/efi
/boot/efi
└── EFI
    ├── arch
    │   ├── fw
    │   │   └── fwupd-dc7f0308-1ef8-4774-9ba5-89a58c4d731c.cap
    │   └── fwupdx64.efi
    ├── Boot
    │   └── bootx64.efi
    ├── GRUB
    │   └── grubx64.efi
    └── Microsoft
```

Please answer the following questions:

- Operating system and version:
Arch Linux, latest
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman
- Have you tried rebooting?
Yes, UEFI update requires reboot. Multiple attempts made.
- Are you using an NVMe disk?
Yes
- Is secure boot enabled (only for the UEFI plugin)?
No

hughsie, 2018-08-12:
This is a known bug with 1.1.0 -- you need to either use 1.0.8, or git master, or the 1.1.1 that we're planning to release on Monday or Tuesday.

hughsie, 2018-08-13:
Okay, 1.1.1 done just now. I'll update Fedora now.

closed 2018-08-13 by hughsie

gkorland, 2018-12-17:
It still doesn't work with 1.1.2 on Ubuntu 18.10
```