fwupd/fwupd #660: ESP detection doesn't work with autofs

superm1, 2018-08-15:
This bug was reported in Debian from the 1.1.1 release.

> My ESP is mounted via autofs so that it's unmounted when not in use:
> 
>     $ grep /boot/efi /etc/fstab
>     PARTLABEL=joyeux-esp /boot/efi vfat umask=0077,x-systemd.automount,x-systemd.idle-timeout=10 0 2
> 
> This leads to:
> 
>     # /usr/lib/fwupd/fwupdate --info 
>     Unable to determine EFI system partition location, override using --esp-path
> 
>     # /usr/lib/fwupd/fwupdate --info --esp-path /boot/efi
>     ESP specified was not valid: /boot/efi has an invalid type, expected vfat|ntfs|exfat
> 
> fwupd seems to be looking through existing mounts for one of the correct
> type, which won't work where the ESP is mounted via autofs, since it's
> not mounted yet...
>

hughsie, 2018-08-15:
I'm not sure what we should do here. I'm not terribly happy parsing fstab. Ideas?

superm1, 2018-08-15:
It seems that there should be a matching systemd unit (and systemd does parsing of /etc/fstab).
https://www.freedesktop.org/software/systemd/man/systemd.automount.html

I see two solutions:
1. Add `autofs` to the supported filesystem types.  Users who do this will have to use the ESP override and then it should work just fine for them.
2. "1" and "Simple" parsing for this scenario as a fallback if it's not mounted.  That should mean look for:
a. uncommented line
b. one of the 3 different supported strings in the second field of `/etc/fstab` 
c. `x-systemd.automount` in the fourth.

hughsie, 2018-08-15:
> Add autofs to the supported filesystem types

If that works, that seems like the pragmatic choice to me.

reversesh3ll, 2018-08-20:
I can confirm this is a bug on Arch using systemd-boot as well.

superm1, 2018-08-20:
Can you please test this?

https://github.com/hughsie/fwupd/pull/661

commit 25b49b6a (2018-08-20): uefi: Add 'autofs' to supported filesystems (Closes: #660)

systemd-automount will unmount the ESP when not in use for some
people.  This causes automatic ESP detection to fail.

In this case the ESP will need to be added to the conf file and
then this commit will let it keep working.
closed 2018-08-20 by superm1