fwupd/fwupd #665: Lenovo T470s fw update does not work

voxik, 2018-08-16:
gnome-software shows firmware update available for my system from 0.1.20 to 0.1.28. Trying the update from g-s does not appear to work. I tried the fwdupdmgr directly:

~~~
$ fwupdmgr get-history 
20HGS22D0W System Firmware
  DeviceId:             84149fb78009cf27cc1dd5520911f46f8792dbfe
  Guid:                 7a176688-0960-47ba-931b-7829849e8347
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|needs-reboot
  Version:              0.1.20
  Created:              2018-08-16
  Modified:             2018-08-16
  UpdateState:          needs-reboot
  
  [Release]
  RemoteId:             lvfs
  Version:              0.1.28
  Checksum:             SHA1(d3f93f01734f37be902cbfd4362d6dd230d20d12)
  TrustFlags:           none
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.1.0
  CompileVersion(org.freedesktop.fwupd): 1.1.0
  SecureBoot:           Disabled
  CompileVersion(org.freedesktop.gusb): 0.3.0
  CompileVersion(org.freedesktop.appstream-glib): 0.7.10
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.11
  ESPMountPoint:        /boot/efi
  KernelVersion:        4.17.14-200.fc28.x86_64
  UEFIUXCapsule:        Disabled
  RuntimeVersion(com.dell.libsmbios): 2.4
  BootTime:             1534415094
  CompileVersion(com.redhat.efivar): 35

$ fwupdmgr update
Downloading 0.1.28 for 20HGS22D0W System Firmware...
Updating 0.1.28 on 20HGS22D0W System Firmware...
Rozbaluje se firmware… [***************************************]
Ověřuje se…          [***************************************]
Plánuje se…           [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: n 
~~~

But after restart, there briefly just flashes some "wait ... install" message or something like that and it appears to do nothing more, just restarts again and boot. So someghing is wrong with fwupd or the update itself? Not really sure.

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.1.0
daemon version:	1.1.0
compile-time dependency versions
	appstream-glib:	0.7.10
	gusb:	0.3.0
	efivar:	35
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20HGS22D0W System Firmware
  DeviceId:             84149fb78009cf27cc1dd5520911f46f8792dbfe
  Guid:                 7a176688-0960-47ba-931b-7829849e8347
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.7
  Icon:                 computer
  Created:              2018-08-16
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             cbede50560aaa37366326cd54923421517427f39
  Guid:                 e9124c4a-fdff-42e5-b2ad-f745db345953
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              182.29.3287
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-16

UEFI Device Firmware
  DeviceId:             03a6930fc4c833443a7502dde379b98b961c8c57
  Guid:                 e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.18
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-16

T470s Thunderbolt Controller
  DeviceId:             9448a212fc859a93ea8be9821db8335c2d9c1b46
  Guid:                 96c3f95f-611a-56f8-be95-0f912cb84657
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              04.00
  Icon:                 computer
  Created:              2018-08-16
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0017,0018,0019,001A,001B,001C,001D,001E,0023
Boot0000* Fedora	HD(1,GPT,3ae6cbcb-6c23-4e5a-8346-97ee1ebc70f8,0x800,0x64000)/File(\EFI\fedora\shim.efi)
Boot0001* Fedora	HD(1,GPT,3ae6cbcb-6c23-4e5a-8346-97ee1ebc70f8,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
/boot
├── config-4.17.14-200.fc28.x86_64
├── config-4.17.3-200.fc28.x86_64
├── config-4.17.6-200.fc28.x86_64
├── efi [error opening dir]
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── chain.c32
│   ├── ifcpu.c32
│   ├── ifcpu64.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2 [error opening dir]
├── initramfs-0-rescue-a304c0039bd04cda8d93ac10281d7620.img
├── initramfs-4.17.14-200.fc28.x86_64.img
├── initramfs-4.17.3-200.fc28.x86_64.img
├── initramfs-4.17.6-200.fc28.x86_64.img
├── loader
│   └── entries [error opening dir]
├── lost+found [error opening dir]
├── memtest86+-5.01
├── System.map-4.17.14-200.fc28.x86_64
├── System.map-4.17.3-200.fc28.x86_64
├── System.map-4.17.6-200.fc28.x86_64
├── vmlinuz-0-rescue-a304c0039bd04cda8d93ac10281d7620
├── vmlinuz-4.17.14-200.fc28.x86_64
├── vmlinuz-4.17.3-200.fc28.x86_64
├── vmlinuz-4.17.6-200.fc28.x86_64
└── 7b3927c214f04d6fa095ec183c1b4323
    └── 4.11.0-2.fc26.x86_64

8 directories, 77 files
```

Please answer the following questions:

- Operating system and version: Fedora release 29 (Rawhide)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

hughsie, 2018-08-16:
Known bug with 1.1.0, fixed in 1.1.1 -- can you try the build in rawhide pls? https://koji.fedoraproject.org/koji/buildinfo?buildID=1135756

superm1, 2018-08-16:
I'm gonna mark this as duplicate of https://github.com/hughsie/fwupd/issues/656

If the 1.1.1 build fails, holler and we can re-open.

closed 2018-08-16 by superm1

voxik, 2018-08-17:
@hughsie nice, the 1.1.1 works like a charm.

BTW a bit OT, but would it be possible to improve g-s to somehow give a hint, that the update was actually scheduled for next restart? Also, I updated the bios from 0.1.20 to 0.1.28, but only the changelog for the latest version was displayed. It would be helpful to be able to review changes in all version. Please forgive me my laziness that I don't open a separate ticket for these and thx a lot for working on this.

hughsie, 2018-08-17:
The ChangeLog merging is work in progress; it should all be working by Fedora 29.