fwupd/fwupd #667: Unescaped ampersand

paulmenzel, 2018-08-16:
With fwupd 0.8.3-0ubuntu3, `/usr/lib/x86_64-linux-gnu/fwupd/fwupd` shows the error below.

    Fu-WARNING **: FuMain: failed to load AppStream data: Failed to parse /var/cache/app-info/xmls/fwupd.xml file: Error on line 265: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity - escape ampersand as &amp;

The problem seems to be `filename="Firmware_SF30&SN30_Pro_V1.26.dat"`.

superm1, 2018-08-16:
This is fixed with commit https://github.com/hughsie/fwupd/commit/fe1c4de5a4b178cae3e4e9325ea84ff85c4b1be3 on the `0_8_X` branch (and similar commits in later versions) which needs to be backported into Ubuntu.

superm1, 2018-08-16:
See https://bugs.launchpad.net/ubuntu/+source/appstream-glib/+bug/1780442 and https://github.com/hughsie/fwupd/issues/565 for related comments.

closed 2018-08-16 by superm1