fwupd/fwupd #669: Secure boot is enabled, but shim isn't installed to the EFI system partition

gdamjan, 2018-08-17:
```shell
$ fwupdmgr --version
client version: 1.1.1
daemon version: 1.1.1
compile-time dependency versions
        appstream-glib: 0.7.10
        gusb:   0.3.0
        efivar: 35
```

```shell
$ fwupdmgr get-devices
20HQS0LV00 System Firmware
  DeviceId:             1cbe298fc17877b6883e85560778ac0812f0a385
  Guid:                 798ffd60-f10e-4ac4-8939-c8beabfe55b4
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.33
  VersionLowest:        0.1.14
  Icon:                 computer
  Created:              2018-08-17
  Modified:             2018-08-17
  UpdateState:          failed
  UpdateError:          Secure boot is enabled, but shim isn't installed to the EFI system partition

UEFI Device Firmware
  DeviceId:             6461040534f2259a0439361986adf73979fd836e
  Guid:                 c35736d2-9e47-4578-93e9-68d5b04ea77e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              182.10.1196
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-17

UEFI Device Firmware
  DeviceId:             6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
  Guid:                 74997a6b-1adf-4b12-b994-401f06ea8c72
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.19
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-17

ThinkPad X1 Carbon Thunderbolt Controller
  DeviceId:             0ce8788ee1c567edffe7dccfda4d6fa7d219d776
  Guid:                 89d9d1e6-9e4c-5f07-b5c1-603da3d61835
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              15.00
  Icon:                 computer
  Created:              2018-08-17
```

```shell
$ efibootmgr -v
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0000,0017,0018,0019,001A,001B,0023,001D,001E
Boot0000* SecureBoot linux      HD(1,GPT,b0a0807d-0592-40e9-adac-3bb724e9e305,0x800,0x80000)/File(\EFI\Secure\secure-boot-linux.efi)
Boot0002* SecureBoot 4.18.1+    HD(1,GPT,b0a0807d-0592-40e9-adac-3bb724e9e305,0x800,0x80000)/File(\EFI\Secure\secure-boot-4.18.1+.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D  Other CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD     VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0020* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0021* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0023* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4-0
```

```shell
$ tree /boot
/boot/
├── EFI
│   ├── arch
│   │   └── fw
│   │       ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │       └── fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4.cap
│   ├── Secure
│   │   ├── secure-boot-4.18.1+.efi
│   │   └── secure-boot-linux.efi
├── initramfs-4.18.1+.img
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── vmlinuz-4.18.1+
└── vmlinuz-linux
```

- Operating system and version: ArchLinux (up-to-date as of 2018-08-17)
-  installed with `pacman`
- Have you tried rebooting? `no`
- Are you using an NVMe disk? `yes`
- Is secure boot enabled (only for the UEFI plugin)? `yes`

```shell
$ sudo fwupdmgr update
Downloading 0.1.34 for 20HQS0LV00 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating 20HQS0LV00 System Firmware from 0.1.33 to 0.1.34…     ]
Scheduling…            [***************************************]
Secure boot is enabled, but shim isn't installed to the EFI system partition
```

I need to clarify that I use my own keys for UEFI secure boot, and sign a combined kernel+initramfs+command-line image. I've already signed the fwupdx64.efi with my keys. Do I need to install fwupdx64.efi.signed in /boot and activate it with efibootmgr myself?

```shell
/usr/lib/fwupd/efi/
├── fwupdx64.efi
└── fwupdx64.efi.signed
```

hughsie, 2018-08-17:
fwupd is looking for `shim.efi` on the ESP. We've not tested at all without shim, but in the self-signed case I suppose it should work, right? In this atypical case I suppose a config option is probably the best thing to have.

superm1, 2018-08-17:
Agree

gdamjan, 2018-08-17:
I can confirm that manually doing this:
```shell
sudo sbsign --key /etc/secure-boot/db.key --cert /etc/secure-boot/db.crt -o  /boot/EFI/arch/fw/fwupdx64.efi /usr/lib/fwupd/efi/fwupdx64.efi
sudo efibootmgr --create-only -L fwupd -l /EFI/arch/fw/fwupdx64.efi  -d /dev/nvme0n1
sudo efibootmgr --bootnext 0004
```
and rebooting, the bios update was done. Obviously the sbsign part would need to be custom, I don't know how fwupd could be made to handle this scenario. 

Does `fwupdx64.efi` change between releases? If not I can use `sudo efibootmgr --bootnext 0004` whenever I want to update.

ps.
unfortunately, my custom boot image was replaced with the default Lenovo image.
In the Lenovo Flash efi program there was an option to not update the BIOS image.

ftiede, 2018-08-17:
I can confirm that a manually signed fwupdx64.efi on ESP similar to what @gdamjan described is sufficient to perform a successful BIOS update of a recent Dell Latitude system running Gentoo Linux. Shim was not required to perform this task and I was about to propose a modified plugins/uefi/fu-uefi-bootmgr.c in a similar issue.
If possible fwupd should check if its ESP/fwupdx64.efi is verifiably signed (there's the sbverify tool which can do that, probably needs an exec() call for that) before erroring out on missing shim.

As for the signing part, this would have to be done during installation/upgrade of fwupd and would require manual system-owner interaction anyway (unless the keys are unprotected and always available, which IMHO renders the whole point of Secure Boot void), so that would be a part for the distribution package managers to alert the user for necessary binary signing after fwupd has been installed or updated.

ArchangeGabriel, 2018-08-17:
I seem to remember there was some warning regarding people using Secure Boot that they needed to sign the executable themselves in order to use fwupdate. Since this dependency was merged, things have changed and there should indeed be some way to manually sign the executable for fwupd usage.

ftiede, 2018-08-19:
My tests showed that sbverify can indeed check and verify an efi loader's signature which includes the loader for fwupd. Unfortunately it does not use the hardware's key store to do so but rather checks against a supplied certificate list. Also, sbverify uses only 0 (success) and 1 (any error) as return value, which makes it hard to understand why sbverify failed, as it returns 1 if the file is no efi loader, not at all signed or not verifiably signed.
So to be useful, fwupd would need to extract the system's keychain and start verifying its boot loader against all keys until either a valid signature was found or the chain is exhausted. And it would possibly need to parse sbverify's output to break of early, f.ex. if the loader is not signed at all.

If the system's keys were replaced according to Secure Boot standards that would still be just one check (against Secure Boot's db certificates), but there's no guarantee the owner did adhere to the standard and did not use Secure Boot's KEK or even PK certificate to sign fwupd, which is possible and would still yield bootable results.
If the system's keys were not replaced then probably shim is required to boot fwupd anyway, so no need in that case.

I for myself am considering to remove the fault part from fwupd's checks and just replace it with a warning to make sure fwupdx64.efi is validly signed otherwise the update would fail to load during reboot but otherwise just proceed as if Secure Boot was inactive.

gdamjan, 2018-08-19:
urgh, just now I notice the firmware update removed my custom UEFI boot entries. This didn't happen with the UEFI updater extracted from Lenovos .iso image.

where/how do we report these issues?

superm1, 2018-08-19:
My take is that just people will not be doing self signed keys, so given the above findings we should just have an override in the config file to ignore missing shim.

superm1, 2018-08-19:
I'm not saying users shouldn't change their keys in recommending this solution. It's just the realty that most won't and I'd like to balance making things easy for the majority while also supporting those that will.

I guess my question is why adopt a solution that signs the EFI application directly rather than signing shim like all the distros do and then signing other binaries with an inner key like all distros do. I'd think that the features provided by Mok are ideal, especially if features such as extending the PCRs land in shim and can be chain loaded to other bootloaders used in the stack such as fwup and grub2 later.

If I'm not mistaken the patches that are not in upstream kernel but adopted by a few distros for secure boot in the kernel are dependent upon using shim too aren't they?

ftiede, 2018-08-20:
I don't think it's fwupd's task to sign itself for Secure Boot, much like it is not GrUB's task to do so for the OS.
If a system owner uses Secure Boot and has replaced the system's key with his own, it's also the owner's responsibility to make sure, fwupd is also signed.

Alas, according to my recent test, fwupd doesn't need to be signed for every firmware update it runs, it just needs to be signed when being installed or updated itself, something the distro's package manager can and should warn about, but that's out of fwupd's runtime scope anyway.

superm1, 2018-08-20:
We used to require runtime installation of the bootloader but it doesn't work for everyone.

A few ways it can (and did) break:
1. Switch secure boot on and off
2. ESP not mounted at install time
3. ESP mounted read only at install time.
4. ESP in non standard location.
5. ESP moved since installation.

There's probably other scenarios that can break too. That's why we do it at runtime now.

In the model that involves shim, fwupd is still signed but with a different key. Shim has a public cert embedded and fwupd is signed with a private key.

Anyway this is veering off topic. I think of we just add a Boolean configuration variable that is `IgnoreMissingShim` (other names welcome) it will allow people to make their own decisions whether they want shim in the picture or not.

closed 2018-08-23 by hughsie

gdamjan, 2018-10-06:
* fwupd 1.1.2 on arch
* setting `RequireShimForSecureBoot=false` in `/etc/fwupd/uefi.conf`
* signing /usr/lib/fwupd/efi/fwupdx64.efi` to `/usr/lib/fwupd/efi/fwupdx64.efi.signed`

updating with my own secure boot works (there are other problems though).

ftiede, 2018-10-08:
- fwupd 1.1.2 on Gentoo
- setting `RequireShimForSecureBoot=false` in `/etc/fwupd/uefi.conf`
- signed `/sr/lib/fwupd/efi/fwupdx64.efi` to `/usr/lib/fwupd/efi/fwupdx64.efi.signed` after installing fwupd two weeks ago

Ran `fwupdmgr refresh; fwupdmgr update` today, fwupd picked up a new firmware blob, successfully downloaded it, scheduled itself for installation on next reboot with the UEFI firmware and ran completely uneventful, all updates were successfully installed, no errors reported at all.

physkets, 2019-03-30:
> updating with my own secure boot works (there are other problems though).

What other problems do you face? Does it wipe boot entries or other custom UEFI config?

gdamjan, 2019-04-01:
> What other problems do you face? Does it wipe boot entries or other custom UEFI config?

yes, from what I've seen it wipes the boot entries and the custom boot image. 
I haven't noticed if it would make some other changes in the BIOS

physkets, 2019-04-04:
That seems bad; Nothing of the sort happens when I update the firmware manually, from the UEFI. Shouldn't this capsule update mechanism do the same thing?

superm1, 2019-04-04:
That sounds like a vendor bug; see https://github.com/hughsie/fwupd/issues/1058