fwupd/fwupd #670: Check Free Space on EFI Partition before reboot

mrdrogdrog, 2018-08-19:
I tried to update my thinkpad T480s's UEFI which failed several times.
Everything ran fine until the reboot.
The fwupd efi boot crashed after some seconds, showed an error message for a split second  and bootet back into my linux. I was able to read the error message using my phones camera.

The error message:
There wasn't enough space on my EFI-Partition.

So I moved the Windows Bootloader to a temporary place and tried again. Now everything is okay.

Please check if there is enough free space on the EFI-Partition before rebooting.
and
Please insert a 5-second-wait before rebooting in case of an error.

Thx :)

hughsie, 2018-08-20:
> enough free space

How do we know the amount of free space the EFI updates needs? Do yo know if it decompresses to the ESP or anything like that? I had assumed that if we write the .cap file then no further space was required.

superm1, 2018-08-20:
This may not apply to Lenovo, but at least for Dell a "BIOS recovery" file is created that allows going back to the old BIOS in the event of a failure.  Maybe they're doing something similar?

If we're going to add in some safety checks for size I think the biggest size SPI part currently seen (32M) is a good starting point.

Regarding automatically rebooting, it's a little unclear which actually performed the reboot here - fwupd or the vendor binary.
Can you please show the error message screenshot you saw?  We can see who showed what message so we can know if control returned to fwupd such that increasing a reboot timeout would help.

hughsie, 2018-08-20:
@superm1 so we could just call `statfs()` in `fu_uefi_check_esp_path()` -- right? If so would @mrdrogdrog like to work on a patch for this issue? We're happy to help if you get stuck.

superm1, 2018-08-20:
Yeah that seems reasonable to me.

closed 2018-09-12 by hughsie