fwupd/fwupd #680: UpdateError: Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf

k4rtik, 2018-08-22:
This might be more of a support issue than a bug, but I have not found any solution online so reporting here for help.

I followed the instructions to install Arch on my XPS 9370 with Windows 10 dual boot. Specifically, I followed the instruction to mount ESP at `/efi`for GRUB: https://wiki.archlinux.org/index.php/EFI_system_partition#Mount_the_partition I did not add an entry to `/etc/fstab` as it did not say if that was needed. Now when I run `fwupdmgr get-devices` I get the error in the subject. When I manually mount the efi partition (`sudo mount /dev/nvme0n1p1 /efi/`), override mount point to `/efi` in the `uefi.conf` file (though that should not be needed, but nothing changes if I don't override) and restart fwupd service, the error changes to:
```
  UpdateError:          invalid OverrideESPMountPoint=/efi specified in config: /efi is read only
```
But the `/etc/mtab` entry suggests the partition is mounted `rw`:
```
$ tail -1 /etc/mtab 
/dev/nvme0n1p1 /efi vfat rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,utf8,errors=remount-ro 0 0
```
I am not sure how should I be mounting the ESP. Or if there is something else that is wrong here. Can someone help?

---

For below info, I am not sure whether I am using the UEFI plugin, I installed fwupd using Arch's package manager.

```shell
$ fwupdmgr --version
client version:	1.1.1
daemon version:	1.1.1
compile-time dependency versions
	appstream-glib:	0.7.10
	gusb:	0.3.0
	efivar:	35
```

```shell
$ fwupdmgr get-devices
XPS 9370 Thunderbolt Controller
  DeviceId:             a7c3a163f927ba97601e38ef011a66525fbe1842
  Guid:                 4eeb9d07-a96c-56d6-92d3-4a23ee7a6e4a
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              28.00
  Icon:                 computer
  Created:              2018-08-22

TS3 Plus
  DeviceId:             129e67cb57725f7fcbfda13cf1ae35d27f46c788
  Guid:                 4b455155-82b1-5d27-8093-70163de758c2
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               CalDigit, Inc.
  VendorId:             TBT:0x003D
  Version:              29.00
  Icon:                 audio-card
  Created:              2018-08-22

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered|needs-reboot
  Version:              0.1.4.0
  VersionLowest:        0.1.4.0
  Icon:                 computer
  Created:              2018-08-22
  UpdateError:          Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf
```

```shell
$ efibootmgr -v
BootCurrent: 0003
Timeout: 0 seconds
BootOrder: 0003,0001,0000
Boot0000* Windows Boot Manager	HD(1,GPT,800cd14c-9469-4baf-ad6b-e9c862263ea0,0x800,0x145000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...K................
Boot0001* ubuntu	HD(1,GPT,800cd14c-9469-4baf-ad6b-e9c862263ea0,0x800,0x145000)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* GRUB	HD(1,GPT,800cd14c-9469-4baf-ad6b-e9c862263ea0,0x800,0x145000)/File(\EFI\GRUB\grubx64.efi)
```

```shell
$ efivar -l | grep fw
<NO OUTPUT>
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**

$ tree -f /boot | grep fw
│       ├── /boot/grub/x86_64-efi/efifwsetup.mod
$ tree /boot | grep cap
$ 
```

Please answer the following questions:

- Operating system and version:
Arch Linux, latest installed yesterday and fully updated.

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
pacman

- Have you tried rebooting?
Yes, several times.

- Are you using an NVMe disk?
Yes

- Is secure boot enabled (only for the UEFI plugin)?
No (AFAIK), how do I confirm?

ArchangeGabriel, 2018-08-22:
So yes, the EFI partition should be mounted `rw` before running `fwupdmgr`. Mounting it during normal operation of the system is indeed not required, and `/efi` is amongst the list of default mount points tried by `fwupd`, so the override shouldn’t be necessary (also I think it won’t work if ever necessary).

Small questions: amongst your several reboot, has there been an instance were you mounted the EFI partition before trying fwupdmgr?

Small note regarding SecureBoot: if you don’t know and are booting ArchLinux fine, then the answer is no.

hughsie, 2018-08-22:
> errors=remount-ro

Are there filesystem errors?

superm1, 2018-08-22:
If it's not mounted by `/etc/fstab` my _guess_ would be it wasn't mounted when the daemon started.

ArchangeGabriel, 2018-08-22:
@superm1 Does the daemon start on boot (or equivalent) or does that happen when running `fwupdmgr`?

ArchangeGabriel, 2018-08-22:
@k4rtik Can you answer my question and try adding you EFI partition to `/etc/fstab`?

superm1, 2018-08-22:
@ArchangeGabriel it can start at boot if something requests it with D-Bus activation (such as gnome-software).

k4rtik, 2018-08-22:
@superm1 thanks, that worked!

Is there a permanent fix to this problem or should we just close this issue?

superm1, 2018-08-22:
At least I don't understand why that's happening.  If `ProtectSystem` was set to strict it would make more sense to me.

> ProtectSystem=
Takes a boolean argument or the special values "full" or "strict". If true, mounts the /usr and /boot directories read-only for processes invoked by this unit. If set to "full", the /etc directory is mounted read-only, too. If set to "strict" the entire file system hierarchy is mounted read-only, except for the API file system subtrees /dev, /proc and /sys (protect these directories using PrivateDevices=, ProtectKernelTunables=, ProtectControlGroups=). This setting ensures that any modification of the vendor-supplied operating system (and optionally its configuration, and local mounts) is prohibited for the service. It is recommended to enable this setting for all long-running services, unless they are involved with system updates or need to modify the operating system in other ways. If this option is used, ReadWritePaths= may be used to exclude specific directories from being made read-only. This setting is implied if DynamicUser= is set. For this setting the same restrictions regarding mount propagation and privileges apply as for ReadOnlyPaths= and related calls, see below. Defaults to off.

ArchangeGabriel, 2018-08-22:
Even with `strict`, if you whitelist the path the man does not says it won’t work. I use `strict` for most things I write and whitelist the needed paths.

superm1, 2018-08-22:
Ah.  Helpfully `/efi` is processed just like the others but not in the documentation.
https://github.com/systemd/systemd/blob/52e4d62550b317ca3693d74e085500575a83dc89/src/core/namespace.c#L148

@k4rtik can you please try to re-enable `ProtectSystem=full` but change `ReadWritePaths` to have `/efi/EFI` instead of `/efi`?

superm1, 2018-08-22:
(and reload/restart)

k4rtik, 2018-08-22:
Tried that to no avail:
```
UpdateError:          Unable to determine EFI system partition location, override using OverrideESPMountPoint in uefi.conf
```

When I changed to `ProtectSystem=strict` with original ESP path, it works (but that's probably the same as commenting out the flag).

ArchangeGabriel, 2018-08-22:
@k4rtik Even if you set the override to `/efi`?

Wait what? `strict` is supposed to be more stringent that `full`…

ArchangeGabriel, 2018-08-22:
I’m wondering if this line is not wrong (from above @superm1 link, in strict definition):
```
 { "/", READONLY, false },
```

superm1, 2018-08-22:
We probably need a similar case to this then for `/efi`.
https://github.com/hughsie/fwupd/commit/9bdbbc5ca8349661f2d664c8f9529070159c6e37#diff-2f5c39c775e4cac7ebbeabed4cefca6a

superm1, 2018-08-22:
@k4rtik can you please try https://github.com/hughsie/fwupd/pull/682

ArchangeGabriel, 2018-08-22:
@k4rtik I can provide you an arch package with the patch added if you want so.

@superm1 Do you know what the boolean value is in those MountEntry tables? I don’t know C/C++ enough to understand that, they are way too many booleans in the definition for me to make any sense of that.

k4rtik, 2018-08-22:
@ArchangeGabriel thanks, that will help

k4rtik, 2018-08-22:
Thanks @ArchangeGabriel It seems to work, but I want to double-check if the `fwupd.service` should have the following in it:
```
ProtectSystem=full
[...]
ReadWritePaths=/var/lib/fwupd /etc/fwupd/remotes.d -/boot/efi -/efi/EFI -/boot/EFI
```

I don't recall where I left this file before installing the package. Perhaps the package rewrote this file.

superm1, 2018-08-22:
Yes it should have been updated by the package. That looks right.

commit df60f698 (2018-08-22): uefi: Add special handling for /efi (Closes #680)

This is a similar problem to #627 (which was fixed by 9bdbbc5ca).
Add special handling to allow writing to /efi/EFI
closed 2018-08-22 by superm1