fwupd/fwupd #683: Thinkpad P50 firmware won't update

Keruspe, 2018-08-23:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.1.1
daemon version: 1.1.1
compile-time dependency versions
        appstream-glib: 0.7.12
        gusb:   0.3.0
        efivar: 36
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
20ENCTO1WW System Firmware
  DeviceId:             2c1302f31806a0e0d57c377d99e18dae56351413
  Guid:                 ddc0ee61-e7f0-4e7d-acc5-c070a398838e
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.51
  VersionLowest:        0.1.46
  Icon:                 computer
  Created:              2018-08-23
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             1f9ed21ce218ebcd0e1c803b7da08dd4e33d01a3
  Guid:                 671d19d0-d43c-4852-98d9-1ce16f9967e4
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              176.12.1001
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-23

UEFI Device Firmware
  DeviceId:             6c8326bcda55dc53571921c60da36f320d3efa3f
  Guid:                 a9971959-9246-4a5b-b2f2-ba6fdcb19349
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.17
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-08-23
```

```shell
$ efibootmgr -v
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0001,0000,0018,001F,0019,001B,0017,001A,001C,001D,001E,0020,0025,0002
Boot0000* Windows Boot Manager  HD(1,GPT,9de6165a-84bd-43a2-9c34-951641fb9eb6,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Linux Boot Manager    HD(1,GPT,95b9d7ae-5c3b-4edc-a8e2-b6545dae2d80,0x800,0x82000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0002* Linux Firmware Updater        PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-00-00-00-00-00-00-00)/File(\EFI\exherbo\fwupdx64.efi)
Boot0010  Setup FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu     FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen      FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics    FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu        FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery   FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key  FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD        VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* NVMe1 VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a401)
Boot001B* ATA HDD0      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f603)
Boot001C* ATA HDD1      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f602)
Boot001D* ATA HDD2      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001E* ATA HDD3      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f604)
Boot001F* USB HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot0020* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0021* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0022* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0023* ATA HDD       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0024* ATAPI CD      VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0025* PCI LAN       VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0026* IDER BOOT CDROM       PciRoot(0x0)/Pci(0x14,0x0)/USB(15,1)
Boot0027* IDER BOOT Floppy      PciRoot(0x0)/Pci(0x14,0x0)/USB(15,0)
```

```shell
$ efivar -l | grep fw
no output
```

```shell
$ tree /boot
/boot
├── ae521a81850e41e7859953e11d31b385
│   └── 4.18.4+
│       ├── initrd
│       └── linux
├── EFI
│   ├── Boot
│   │   └── BOOTX64.EFI
│   ├── exherbo
│   │   ├── fw
│   │   └── fwupdx64.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   └── ae521a81850e41e7859953e11d31b385-4.18.4+.conf
    └── loader.conf

9 directories, 7 files
```

Please answer the following questions:

- Operating system and version: Exherbo
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Package manager, from source
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

Note that efibootmgr lists the firmware updater as current boot but system is fully booted and firware not udpated.
I'm not using shim.efi.
Here is what I did prior to the reboot:

```
Lou ~ # fwupdmgr refresh                                                                                                                         
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

Lou ~ # fwupdmgr get-updates                                                                                                                     
20ENCTO1WW System Firmware has firmware updates:
GUID:                    ddc0ee61-e7f0-4e7d-acc5-c070a398838e
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
ID:                      com.lenovo.ThinkPadN1EET.firmware
Update Version:          0.1.52
Update Name:             ThinkPad P50
Update Summary:          Lenovo ThinkPad P50 System Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(eaa39283a0f75dddf9f98409258e3a09027b3f9d)
Update Location:         https://fwupd.org/downloads/1c03e1b5491094429129b3eef952816c21777281-Lenovo-ThinkPad-P50-SystemFirmware-1.52.cab
Update Description:      CHANGES IN THIS RELEASE Version 1.52
                         
                         [Important updates] Nothing.
                         
                         [New functions or enhancements] - Updated the Diagnostics module to version 04.00.001.
                         
                         [Problem fixes] - Fixed an issue where BIOS POST might display the error message "Boot Manager recover from an error."
Lou ~ # fwupdmgr update                                                                                                                          
Downloading 0.1.52 for 20ENCTO1WW System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating 20ENCTO1WW System Firmware from 0.1.51 to 0.1.52…     ]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: 
```

Keruspe, 2018-08-23:
Also

```
Lou ~ # fwupdmgr update                                           
Downloading 0.1.52 for 20ENCTO1WW System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating 20ENCTO1WW System Firmware from 0.1.51 to 0.1.52…     ]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: n
Lou ~ # tree /boot                                                
/boot
├── ae521a81850e41e7859953e11d31b385
│   └── 4.18.4+
│       ├── initrd
│       └── linux
├── EFI
│   ├── Boot
│   │   └── BOOTX64.EFI
│   ├── exherbo
│   │   ├── fw
│   │   │   └── fwupd-ddc0ee61-e7f0-4e7d-acc5-c070a398838e.cap
│   │   └── fwupdx64.efi
│   └── systemd
│       └── systemd-bootx64.efi
└── loader
    ├── entries
    │   └── ae521a81850e41e7859953e11d31b385-4.18.4+.conf
    └── loader.conf

9 directories, 8 files
Lou ~ # efivar -l | grep fw                                                      
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ddc0ee61-e7f0-4e7d-acc5-c070a398838e-0
Lou ~ # efibootmgr -v 
BootNext: 0002
[snip]
```

ArchangeGabriel, 2018-08-23:
So did you reboot after that? All I can see is a system waiting to be rebooted for the firmware to be updated.

Keruspe, 2018-08-23:
My first message was after the reboot.
My second message was just me re-running the same commands.
Rebooting once more does nothing more (except the BootNext disappears again from efibootmgr)

superm1, 2018-08-24:
Before update turn on debugging.
`/usr/lib/fwupd/fwupdate --set-debug` (or `/usr/libexec/fwupd/fwupdate --set-debug`

Do the update.

After update:
`/usr/lib/fwupd/fwupdate --log` (or `/usr/libexec/fwupd/fwupdate --log`

Share this here.

Turn off debugging:
`/usr/lib/fwupd/fwupdate --unset-debug` (or `/usr/libexec/fwupd/fwupdate --unset-debug`

Keruspe, 2018-08-28:
```
Lou ~ # /usr/libexec/fwupd/fwupdate --set-debug                 
Enabled fwupdate debugging
Lou ~ # fwupdmgr update                                         
Downloading 0.1.52 for 20ENCTO1WW System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Updating 20ENCTO1WW System Firmware from 0.1.51 to 0.1.52…     ]
Scheduling…            [***************************************]

An update requires a reboot to complete. Restart now? [Y|n]: 
```

Then

```
Lou ~ # /usr/libexec/fwupd/fwupdate --log                                                                                              
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
Lou ~ # efivar -l | grep FW                                                                                                            
0abba7dc-e516-4167-bbf5-4d9d1c739416-FWUPDATE_VERBOSE
Lou ~ # /usr/libexec/fwupd/fwupdate --unset-debug                                                                                      
Disabled fwupdate debugging
```

Not sure what more I can do :/

Is it expected that the variable created by --set-debug is called VERBOSE and not DEBUG?

superm1, 2018-08-28:
It is supposed to be `FWUPDATE_DEBUG_LOG` (corresponding pieces: https://github.com/hughsie/fwupd/blob/master/plugins/uefi/efi/fwupdate.c#L82, https://github.com/hughsie/fwupd/blob/master/plugins/uefi/fu-uefi-tool.c#L180)

So it didn't even launch the updater that tells me.

Looking at your efibootmgr output again I think the entry wasn't built properly for your disk.
```
Boot0002* Linux Firmware Updater        PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-00-00-00-00-00-00-00)/File(\EFI\exherbo\fwupdx64.efi)
```

That's an invalid NVMe disk entry I think.  Notice how the other entries you have are built using abbreviated entries?

Keruspe, 2018-08-28:
Ok,

Deleting the entry and anually recreating it, then launching another update...

First, fwupd recreated another entry, which didn't boot as usual, then I set bootnext to the entry I created and now I get logs:

```
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:464:find_updates(): Found update fwupd-ddc0ee61-e7f0-4e7d-acc5-c070a398838e-0
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:1286:efi_main(): Adding new capsule
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:575:search_file(): Searching Device Path: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084048400)/\EFI\exherbo\fw\fwupd-ddc0ee61-e7f0-4e7d-acc5-c070a398838e.cap ...
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:603:search_file(): Device Path prepared: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:615:search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x3,0x0,0x0)/HD(Part1,Sig9DE6165A-84BD-43A2-9C34-951641FB9EB6)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x3,0x0,0x0)/HD(Part1,Sig9DE6165A-84BD-43A2-9C34-951641FB9EB6)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and Pci(0x17,0x0)/Sata(0x3,0x0,0x0)/HD(Part1,Sig9DE6165A-84BD-43A2-9C34-951641FB9EB6)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and Sata(0x3,0x0,0x0)/HD(Part1,Sig9DE6165A-84BD-43A2-9C34-951641FB9EB6)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and HD(Part1,Sig9DE6165A-84BD-43A2-9C34-951641FB9EB6)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:615:search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig95B9D7AE-5C3B-4EDC-A8E2-B6545DAE2D80)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig95B9D7AE-5C3B-4EDC-A8E2-B6545DAE2D80)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig95B9D7AE-5C3B-4EDC-A8E2-B6545DAE2D80)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and Pci(0x0,0x0)/Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig95B9D7AE-5C3B-4EDC-A8E2-B6545DAE2D80)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and Path(131,23,0x01000000000000000000000084012C00)/HD(Part1,Sig95B9D7AE-5C3B-4EDC-A8E2-B6545DAE2D80)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:622:search_file(): Comparing: PciRoot(0)/Pci(0x1D,0x0)/Pci(0x0,0x0)/Path(131,23,0x010000000000000000000000FFFF0400) and HD(Part1,Sig95B9D7AE-5C3B-4EDC-A8E2-B6545DAE2D80)
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:664:open_file(): Could not locate device handle: Unsupported
../fwupd-1.1.1/plugins/uefi/efi/fwupdate.c:1294:efi_main(): fwupdate: Could not build update list: Unsupported
```

It's probably linked to the same problem I guess, but why does it use a wrong path like that?

superm1, 2018-08-28:
Could be bugs in fwupd, efivar, or the kernel.
If it's efivar, I'd guess this: https://github.com/rhboot/efivar/commit/d8d7e54fe01a7a255e649a7734820800edf82633
I can't view that bug, so I don't know for sure.

I think this needs someone like @vathpela to take a look what happened.

hughsie, 2018-09-12:
@Keruspe can you try that efivar fix and see if it fixes your problem please? Thanks.

Keruspe, 2018-09-12:
Heh, sorry, Ithought I replied on this.

I already have this patch, and also https://github.com/rhboot/efivar/commit/6d87113fc3d9b2b0f520fc97eab9f5a60fda7d30 backported. Without these two `fwupdmgr update` wouldn't even run

Keruspe, 2018-09-12:
Just tested with efivar git master and fwupd 1.1.2, same issue

Keruspe, 2018-09-13:
To sum things up, what seems not to be working correctly is that fwupd tries to use `Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-00-00-00-00-00-00-00)` which doesn't work, when efibootmgr generates `HD(1,GPT,95b9d7ae-5c3b-4edc-a8e2-b6545dae2d80,0x800,0x82000)` which works.

@vathpela Any idea why one would work and not the other, and how to find out what's wrong in the non-working one?

Keruspe, 2018-09-19:
Will try tomorrow thanks!

Keruspe, 2018-09-19:
Works like a charm, thanks!

Looking forward to the next efivar release :)

closed 2018-09-19 by Keruspe

superm1, 2018-09-19:
@vathpela Can you explain for those of us not following super closely what's going on and when this should/shouldn't work?  Why does this fail for @Keruspe with current efivar releases?  Some new kernel interface or something else?  Eg. Do we have a need to push new efivar into all the distros ASAP?

vathpela, 2018-09-20:
More or less: yes, my refactor to make nvdimm halfway sane broke partition detection. So "efibootmgr -d /dev/sda1" (as opposed to "efibootmgr -d /dev/sda -p 1") and libfwup broke. So distros that have already shipped that patchset will need updates. 

The good news is that my coworker Erico has built a test robot, and that's how we found this to write that patch, so next time we should see it *before* shipping things.

(We'll also be looking into how to get it hooked up to GitHub.)

superm1, 2018-09-20:
I'd guess 36 at least