fwupd/fwupd #689: TPM 2.0 firmware update not working on Dell XPS 9360

paulmenzel, 2018-08-27:
Ubuntu 18.04 tells me that a firmware update is available.

```shell
client version:	1.0.6
daemon version:	1.0.6
compile-time dependency versions
	appstream-glib:	0.7.4
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

```
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.50
  VersionBootloader:    11.8.50
  Icon:                 computer
  Created:              2018-08-27

XPS 13 9360 TPM 2.0
  DeviceId:             a60b665c769b019e30f31001e91ab31e24d7a035
  Guid:                 73c2051d-8688-56fb-a93f-d56a9b455e52
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              1.3.0.1
  Icon:                 computer
  Created:              2018-08-27
  Modified:             2018-08-27
  UpdateState:          needs-reboot

XPS 13 9360 TPM 1.2
  DeviceId:             6f9095025a636a5185bfce08ed8e9fc845bc4b8a
  Guid:                 f9bdd338-b410-5e73-902d-7b6e4694bb56
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2018-08-27

XPS 13 9360 System Firmware
  DeviceId:             3f0ecc7823ee99c718693e8ecca38f54e8738dc5
  Guid:                 5ffdbc0d-f340-441c-a803-8439c8c0ae10
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.2.8.1
  VersionLowest:        0.2.8.1
  Icon:                 computer
  Created:              2018-08-27

HD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 3ec3df3a-2290-56e5-9d2f-eda62e9ab50b
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-08-27

XPS13 9360 Thunderbolt Controller
  DeviceId:             9516070bdf03a7484d1145383fc0faeb9c51a521
  Guid:                 dfd51125-338f-56ff-b721-fa3bea8e534e
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              21.00
  Icon:                 computer
  Created:              2018-08-27

```

```shell
$ efibootmgr -v
Skipping unreadable variable "Boot0001": Permission denied
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001
Boot0000* ubuntu	HD(1,GPT,142dc41d-2c62-47f2-94d2-75165b1bc9cf,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0006* Windows Boot Manager	HD(3,GPT,1cb9c60d-fbc9-4f52-94d2-fe1b0a342fa4,0xe1800,0x32000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...d................
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupdate-73c2051d-8688-56fb-a93f-d56a9b455e52-0
```

```shell
$ tree /boot
[…]
├── efi
│   └── EFI
│       ├── BOOT
│       │   ├── BOOTX64.EFI
│       │   └── fbx64.efi
│       ├── Dell
│       │   ├── Bios
│       │   │   └── Recovery
│       │   │       ├── BIOS_CUR.RCV
│       │   │       └── BIOS_PRE.rcv
│       │   └── logs
│       │       ├── diags_current.xml
│       │       └── diags_previous.xml
│       └── ubuntu
│           ├── BOOTX64.CSV
│           ├── fw
│           │   └── fwupdate-VoqeNJ.cap
│           ├── fwupx64.efi
│           ├── grub.cfg
│           ├── grubx64.efi
│           ├── mmx64.efi
│           └── shimx64.efi
[…]
```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

Starting the update seems to work.

```
$ fwupdmgr update
Downloading 1.3.1.0 for XPS 13 9360 TPM 2.0...
Updating 1.3.1.0 on XPS 13 9360 TPM 2.0...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
No upgrades for XPS 13 9360 System Firmware, current is 0.2.8.1: 0.2.8.1=same, 0.2.7.1=older, 0.2.6.2=older, 0.2.5.1=older, 0.2.4.2=older, 0.2.3.1=older, 0.2.2.1=older, 0.2.1.0=older, 0.1.3.7=older, 0.1.3.5=older, 0.1.3.2=older, 0.1.2.3=older
No upgrades for XPS13 9360 Thunderbolt Controller, current is 21.00: 21.00=same

An update requires a reboot to complete. Restart now? [Y|n]:
```

On the next startup, the firmware shows a message, that it tries to apply the update, but then after the (automatic) reboot the old firmware is still there.

superm1, 2018-08-27:
What's the message it shows?  And if you run `fwupdmgr get-results 3f0ecc7823ee99c718693e8ecca38f54e8738dc5` what does it show?

paulmenzel, 2018-08-27:
On 08/27/18 17:49, Mario Limonciello wrote:
> What's the message it shows?

Something like, that it found an update file. The screen turns
black then. Can I access these messages from GNU/Linux?

> And if you run `fwupdmgr get-results
> 3f0ecc7823ee99c718693e8ecca38f54e8738dc5` what does it show?

Here are all the results, but that’s for the system firmware
and not TPM.

```
$ fwupdmgr get-results a60b665c769b019e30f31001e91ab31e24d7a035Device a60b665c769b019e30f31001e91ab31e24d7a035 has no results to report
$ fwupdmgr get-results 3f0ecc7823ee99c718693e8ecca38f54e8738dc5Failed to find 3f0ecc7823ee99c718693e8ecca38f54e8738dc5 in history database: No devices found
$ fwupdmgr get-results 9516070bdf03a7484d1145383fc0faeb9c51a521XPS13 9360 Thunderbolt Controller
  DeviceId:             9516070bdf03a7484d1145383fc0faeb9c51a521
  Guid:                 4d86f168-e1cc-5995-afd3-ae9df6a14f5e
  Plugin:               thunderbolt
  Flags:                internal|updatable|reported
  Version:              00.00
  Created:              2018-08-27
  Modified:             2018-08-27
  UpdateState:          success
  
  [Release]
  Version:              21.00
  Checksum:             SHA1(d621932738d56a024cbcc516d9810e4c6ffe477a)
  TrustFlags:           none
  AppstreamGlibVersion: 0.7.4
  KernelVersion:        4.18.3-041803-generic
  FwupdVersion:         1.0.6
  ThunderboltSafeMode:  True
  GUsbVersion:          0.2.11
  BootTime:             1535363748
  CpuArchitecture:      x86_64
```

superm1, 2018-08-27:
> Something like, that it found an update file. The screen turns
black then. Can I access these messages from GNU/Linux?

So it found an update, and then went black and rebooted?  You never saw the Dell screen and a progress bar (like you do with UEFI capsule updates)?

Sorry, my mistake, can you see if there are any results for `6f9095025a636a5185bfce08ed8e9fc845bc4b8a`?

paulmenzel, 2018-08-27:
On 08/27/18 17:59, Mario Limonciello wrote:
>> Something like, that it found an update file. The screen turns
> black then. Can I access these messages from GNU/Linux?
> 
> So it found an update, and then went black and rebooted?  You never
> saw the Dell screen and a progress bar (like you do with UEFI capsule
> updates)?

Yes, I never saw that.

> Sorry, my mistake, can you see if there are any results for
> `6f9095025a636a5185bfce08ed8e9fc845bc4b8a`?

`fwupdmgr get-results` doesn’t list this ID. There is only the TPM 2.0
ID (a60b665c769b019e30f31001e91ab31e24d7a035), which has no results to
report.

superm1, 2018-08-27:
> Yes, I never saw that.

OK.  Do you know if in this configuration you have UEFI capsule updates have been working in general (non-TPM)?

I notice that you have a newer kernel version in place.  Did you also change anything else?

Would you care to please check your `fwupd.service` journal?  I'd like to see the cached `efibootmgr` output it did before the update attempt.

paulmenzel, 2018-08-28:
On 08/27/18 18:14, Mario Limonciello wrote:
>> Yes, I never saw that.
> 
> OK.  Do you know if in this configuration you have UEFI capsule
> updates have been working in general (non-TPM)?

Yes, the Thunderbolt firmware update worked, and the previous UEFI
firmware updates too.

> I notice that you have a newer kernel version in place.  Did you also
> change anything else?

No, I didn’t change anything else.

> Would you care to please check your `fwupd.service` journal?  I'd
> like to see the cached `efibootmgr` output it did before the update
> attempt.
```
$ sudo fwupdmgr update
Downloading 1.3.1.0 for XPS 13 9360 TPM 2.0...
Updating 1.3.1.0 on XPS 13 9360 TPM 2.0...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
No upgrades for XPS 13 9360 System Firmware, current is 0.2.8.1: 0.2.8.1=same, 0.2.7.1=older, 0.2.6.2=older, 0.2.5.1=older, 0.2.4.2=older, 0.2.3.1=older, 0.2.2.1=older, 0.2.1.0=older, 0.1.3.7=older, 0.1.3.5=older, 0.1.3.2=older, 0.1.2.3=older
Downloading 21.00 for XPS13 9360 Thunderbolt Controller...
Updating 21.00 on XPS13 9360 Thunderbolt Controller...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Restarting device…     [***************************************]
An update requires a reboot to complete. Restart now? [Y|n]: n
$ journalctl -u fwupd.service -b
-- Logs begin at Mon 2017-10-23 13:12:43 CEST, end at Tue 2018-08-28 11:48:24 CEST. --
Aug 28 11:43:24 Ixpees systemd[1]: Starting Firmware update daemon...
Aug 28 11:43:46 Ixpees fwupd[2013]: failed to get native mode status: Error reading from file: Eingabe-/Ausgabefehler
Aug 28 11:43:46 Ixpees fwupd[2013]: using plugins: upower, amt, colorhug, thunderbolt, altos, ebitdo, udev, dell, csr, nitrokey, dfu, steelseries, unifying, uefi, synapticsmst, thunderbolt_p
Aug 28 11:43:46 Ixpees fwupd[2013]: Daemon ready for requests
Aug 28 11:43:46 Ixpees systemd[1]: Started Firmware update daemon.
Aug 28 11:43:47 Ixpees fwupd[2013]: /sys/devices/pci0000:00/0000:00:1c.0/0000:01:00.0/0000:02:00.0/0000:03:00.0/domain0/0-0 is in safe mode --  VID/DID will need to be set by another plugin
Aug 28 11:47:15 Ixpees fwupd[2013]: failed to add USB device: 1d6b:0002 is not supported: USB error on device 1d6b:0002 : Input/output error [-1]
Aug 28 11:47:15 Ixpees fwupd[2013]: failed to add USB device: 1d6b:0003 is not supported: USB error on device 1d6b:0003 : Input/output error [-1]
Aug 28 11:47:44 Ixpees fwupd[2013]: new device version '21.0' was is not '21.00', fixing up
```

bdellegrazie, 2018-09-02:
Just FYI , I have this exact same issue:
TPM 2.0 firmware failing to update on Dell XPS 9360, running Ubuntu 18.04, 1.3.0.1 -> 1.3.1.0

topochan, 2018-09-16:
Same issue here, failing to update on Dell XPS 9360, running Ubuntu 18.04, 1.3.0.1 -> 1.3.1.0

CaineThanatos, 2018-09-27:
Same here! I think
sudo fwupdmgr update:

`ignoring a60b665c769b019e30f31001e91ab31e24d7a035 [XPS 13 9360 TPM 2.0] as not updatable
No upgrades for XPS 13 9360 System Firmware, current is 0.2.9.0: 0.2.9.0=same, 0.2.8.1=older, 0.2.7.1=older, 0.2.6.2=older, 0.2.5.1=older, 0.2.4.2=older, 0.2.3.1=older, 0.2.2.1=older, 0.2.1.0=older, 0.1.3.7=older, 0.1.3.5=older, 0.1.3.2=older, 0.1.2.3=older
No upgrades for XPS13 9360 Thunderbolt Controller, current is 21.00: 21.00=same`

superm1, 2018-09-27:
Those affected by this, is your tpm owned?

https://github.com/hughsie/fwupd/commit/736be7440fdee3f5339f3b1dc193f674a6191696

CaineThanatos, 2018-09-27:
I will check

adeptg, 2018-10-24:
Hi! Same issue here, failing to update on Dell XPS 9360, running Ubuntu 18.04, 1.3.0.1 -> 1.3.1.0
Please let me know if I can help somehow

superm1, 2018-10-24:
Please try the snap and see if it still happens there.

adeptg, 2018-10-24:
@superm1  thanks for advise! For me it works with latest snap

sudo fwupdmgr --version
client version: 1.1.3
checkout info:  1.1.3
compile-time dependency versions
        appstream-glib: 0.7.9
        gusb:   0.3.0
        efivar: 35
daemon version: 1.1.3

sudo fwupdmgr get-updates -v
No upgrades for XPS 13 9360 System Firmware, current is 0.2.9.0: 0.2.9.0=same, 0.2.8.1=older, 0.2.7.1=older, 0.2.6.2=older, 0.2.5.1=older, 0.2.4.2=older, 0.2.3.1=older, 0.2.2.1=older, 0.2.1.0=older, 0.1.3.7=older, 0.1.3.5=older, 0.1.3.2=older, 0.1.2.3=older
XPS 13 9360 TPM 2.0 has firmware updates:
GUID:                    73c2051d-8688-56fb-a93f-d56a9b455e52
GUID:                    dcaeb1a0-105f-5b2a-aaad-61a13bf566d5
ID:                      com.dell.uefi22d63f4.firmware
Update Version:          1.3.1.0
Update Name:             TPM 2.0 Update
Update Summary:          Firmware for the Dell TPM 2.0
Update Remote ID:        lvfs
Update Checksum:         SHA1(1b6af0a6946181fae71388858674243bf59d2bea)
Update Location:         https://fwupd.org/downloads/a1a6e10beb96281fa78c62a7d967c8c3a1cf7430-DellTpm2.0_Fw1.3.1.0.cab
No upgrades for XPS13 9360 Thunderbolt Controller, current is 26.01: 21.00=older

sudo fwupdmgr update -v
No upgrades for XPS 13 9360 System Firmware, current is 0.2.9.0: 0.2.9.0=same, 0.2.8.1=older, 0.2.7.1=older, 0.2.6.2=older, 0.2.5.1=older, 0.2.4.2=older, 0.2.3.1=older, 0.2.2.1=older, 0.2.1.0=older, 0.1.3.7=older, 0.1.3.5=older, 0.1.3.2=older, 0.1.2.3=older
Downloading 1.3.1.0 for XPS 13 9360 TPM 2.0...
(fwupdmgr:27374): FuCommon-DEBUG: creating path /root/snap/fwupd/common/.cache/fwupd
(fwupdmgr:27374): FuMain-DEBUG: skpping download as file already exists
(fwupdmgr:27374): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing?         [-                                      ](fwupdmgr:27374): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Decompressing?         [***************************************]
(fwupdmgr:27374): Fwupd-DEBUG: Emitting ::status-changed() [waiting-for-auth]
Authenticating?        [ -                                     ](fwupdmgr:27374): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Authenticating?        [***************************************]
(fwupdmgr:27374): Fwupd-DEBUG: Emitting ::status-changed() [scheduling]
Updating XPS 13 9360 TPM 2.0?                                  ]
Scheduling?            [ -                                     ](fwupdmgr:27374): Fwupd-DEBUG: Emitting ::device-changed(a60b665c769b019e30f31001e91ab31e24d7a035)
Scheduling?            [    \                                  ](fwupdmgr:27374): Fwupd-DEBUG: Emitting ::status-changed() [idle]
Scheduling?            [***************************************]
(fwupdmgr:27374): Fwupd-DEBUG: Emitting ::changed()
No upgrades for XPS13 9360 Thunderbolt Controller, current is 26.01: 21.00=older

An update requires a reboot to complete. Restart now? [Y|n]: Y

superm1, 2018-10-24:
Great thanks.

closed 2018-10-24 by superm1

bestouff, 2019-07-04:
Hi,

I have the very same problem under Debian.
```
root@xav13:~# fwupdmgr refresh && fwupdmgr update 
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…             [                                       ] Less than one minute remDownloading…             [*                                      ] Less than one minute remDownloading…             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

No upgrades for XPS 13 9360 System Firmware, current is 0.2.12.0: 0.2.12.0=same, 0.2.11.0=older, 0.2.10.0=older, 0.2.9.0=older, 0.2.8.1=older
ignoring XPS 13 9360 TPM 2.0 [a60b665c769b019e30f31001e91ab31e24d7a035] as not updatable
root@xav13:~# fwupdmgr --version
client version:	1.2.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.6
```
What does "owned TPM" mean ?

superm1, 2019-07-08:
You might consider looking at https://docs.microsoft.com/en-us/windows/security/information-protection/tpm/initialize-and-configure-ownership-of-the-tpm for some high level information about TPM ownership.

To fwupd being owned means that it contains data that can not be cleared by the OS.  You'll need to enter BIOS firmware setup and clear the TPM there.  Then fwupd will be able to perform a TPM firmware upgrade.

bestouff, 2019-07-08:
They explicitly advise against clearing the TPM via the BIOS.

superm1, 2019-07-08:
If you're using Windows on the machine this is true, Bitlocker keys are stored in the TPM and you might not be able to boot the machine again if you use it.

fwupd is disabling updates (ignoring the device) for the same reason.  So it's up to you if you can clear the TPM or not to perform this update.

bestouff, 2019-07-09:
Works like a charm, thanks !