fwupd/fwupd #693: UEFI firmware of Dell OptiPlex 5055 with Ryzen CPU not listed

paulmenzel, 2018-08-28:
Booting the Ubuntu 18.04.1 live system on the Dell OptiPlex 5055 with AMD Ryzen CPU, the UEFI firmware is not listed by `fwupdmgr`.

```
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 4.15.0-29-generic #31-Ubuntu SMP Tue Jul 17 15:39:52 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
ubuntu@ubuntu:~$ fwupdmgr get-devices
OptiPlex 5055 Ryzen CPU TPM 2.0
  DeviceId:             4472633f8a1aabf2b5cfe0865c54bfd935a958c0
  Guid:                 cee8a722-32b1-5b34-8d7c-6c5a83f3a887
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  Version:              1.3.2.8
  Icon:                 computer
  Created:              2018-08-28

300 Series Chipset SATA Controller
  DeviceId:             8fd51adb8bbd4bb887cdad1898cb267494c54e4a
  Guid:                 4ecee959-87a1-5119-b77d-5f591a618614
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Advanced Micro Devices, Inc. [AMD]
  VendorId:             PCI:0x1022
  Icon:                 audio-card
  Created:              2018-08-28

Venus LE / Tropo PRO-L [Radeon HD 8830M / R7 M465X]
  DeviceId:             72d6f65bc50f387cffeea8c916ce7441715082cc
  Guid:                 4c8907d2-fcf6-5422-8415-d7d9407b3660
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Advanced Micro Devices, Inc. [AMD/ATI]
  VendorId:             PCI:0x1002
  Icon:                 audio-card
  Created:              2018-08-28

ubuntu@ubuntu:~$ fwupdmgr update
ubuntu@ubuntu:~$
```

superm1, 2018-08-28:
Did you check that you have UEFI capsule updates turned on in BIOS setup?

paulmenzel, 2018-08-29:
Ubuntu 18.04 LTS seems to ship version 1.0.6-2. With version 1.1.1 from Debian Buster/testing it seems to work.

```
$ fwupdmgr update
No upgrades for OptiPlex 5055 Ryzen CPU System Firmware, current is 0.1.1.15: 0.1.0.7=older, 0.1.0.1=older
$ fwupdmgr --version
client version: 1.1.1
daemon version: 1.1.1
compile-time dependency versions
        appstream-glib: 0.7.10
        gusb:   0.2.11
        efivar: 34
```

closed 2018-08-29 by paulmenzel

superm1, 2018-08-29:
That's a bit odd to me.  There isn't any reason that 1.0.6 _shouldn't_ work though I know of.