fwupd/fwupd #701: Can't install older version even with --allow-older

guss77, 2018-08-30:
Running fwupdmgr on a downloaded cab file:

```
$ sudo fwupdmgr install ~/Downloads/9c9702786bd6883845c0a1c49216a9af735efe70-firmware_XPS9560_Precision5520_1.10.1.cab f49999b6e3af5b4bf8ad1f2ed2922b76df457271 --allow-older --force
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Specified firmware is older than the minimum required version '0.1.11.0 < 0.1.10.1'
```

Seems neither `--allow-older` nor `--force` can get fwupdmgr to install an older firmware file.

BTW: The status line at the end seems incorrect.

```shell
$ fwupdmgr --version
client version: 1.0.6
daemon version: 1.0.6
compile-time dependency versions
        appstream-glib: 0.7.4
        gusb:   0.2.11
        fwupdate:       10
        efivar: 34
```

```shell
$ fwupdmgr get-devices
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.55
  VersionBootloader:    11.8.55
  Icon:                 computer
  Created:              2018-08-30

Precision 5520 TPM 2.0
  DeviceId:             de98f10c9b146ba4ced2fc37cad31ee5b8808f6d
  Guid:                 8bc551f7-f7ca-5ea8-a4cb-38ae16ecfa17
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              1.3.1.0
  Icon:                 computer
  Created:              2018-08-30

Precision 5520 TPM 1.2
  DeviceId:             e0527307c0aa37127c77c5c1f777cbe348591d1a
  Guid:                 61c84b2d-1924-5867-91e2-63ae6a4373af
  Summary:              Alternate mode for platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Icon:                 computer
  Created:              2018-08-30

Unifying Receiver
  DeviceId:             16324df8d782cbc6d3396541199c8fc397182242
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0014
  Icon:                 preferences-desktop-keyboard
[... 344 lines cut ...]
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.15.0-32-generic
├── initrd.img-4.15.0-33-generic
├── lost+found [error opening dir]
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── retpoline-4.15.0-32-generic
├── retpoline-4.15.0-33-generic
├── System.map-4.15.0-32-generic
├── System.map-4.15.0-33-generic
├── vmlinuz-4.15.0-32-generic
└── vmlinuz-4.15.0-33-generic

6 directories, 291 files
```

Please answer the following questions:

- Operating system and version: Ubuntu 18.04
- How did you install fwupd: was installed by the operating system
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

superm1, 2018-08-30:
This version is advertised by the existing firmware in the ESRT.  Typically it's used to indicate that programmatic downgrading is disabled.  So fwupd is doing the right thing.

Dell will probably allow you to put the EXE on an flash drive and downgrade manually with the F12 POST menu however.

guss77, 2018-08-30:
I see. But isn't `--force` (or `--allow-older`) supposed to allow the downgrade regardless?

Also, what do you say about the last line in the output? Shouldn't the comparison operator be reversed?

superm1, 2018-08-30:
`--force` doesn't ignore UEFI requirements checks.

I do think that last line looks wrong though, you're right.

hughsie, 2018-08-30:
> --force doesn't ignore UEFI requirements checks.

Could it? I mean, if you can do it in the system firmware with an .exe there's no real technical reason why we prevent it.

superm1, 2018-08-30:
I suppose I don't know how hard this is enforced by firmware coming through capsule.  Can certainly try.

commit 8cbdc1e5 (2018-09-05): fu-install-task: Allow using --force to ignore version_lowest (Closes #701)
commit 3055e794 (2018-09-05): fu-install-task: Allow using --force to ignore version_lowest (Closes #701)
commit 6fd1cad8 (2018-09-05): fu-install-task: Allow using --force to ignore version_lowest (Closes #701)
commit da0e90f2 (2018-09-05): fu-install-task: Allow using --force to ignore version_lowest (Closes #701)
closed 2018-09-05 by superm1