fwupd/fwupd #702: Use UpdateCapsule at runtime

jethrogb, 2018-08-30:
I couldn't find any design docs, so please let me know if this has already been considered and rejected, and if so for what reason.

As I understand it, fwupd currently puts all the UEFI update capsules on the boot device along with a special UEFI application that will call `UpdateCapsule` with those files, and fwupd will modify the boot order to run that application on reboot. Since `UpdateCapsule` is a function in `EFI_RUNTIME_SERVICES`, it should be possible to call this while the OS is running and avoid messing with the boot manager. In fact, there's already a kernel driver for this: `capsule_loader`.

hughsie, 2018-08-31:
I believe @vathpela tried this a while ago. IIRC, the number of platforms that actually did the right thing when using runtime services is close to zero, and this is why we did the same thing as Windows. Happy to be corrected.

superm1, 2018-08-31:
In the ESRT one of the bitfields advertises whether or not the capsule needs a reboot to apply.  Nearly every one of them does today.