fwupd/fwupd #703: Unable to update 8bitdo zero FC30

KingDuckZ, 2018-08-31:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.1.1
daemon version: 1.1.1
compile-time dependency versions
        appstream-glib: 0.7.9
        gusb:   0.2.11
        efivar: 35
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 1 seconds
BootOrder: 0000,0001,001E,0020,001C,001D
Boot0000* ubuntu        HD(1,GPT,7c170bbd-add3-4d0e-becc-e6d7b44c5503,0x800,0x40000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* gentoo        HD(1,GPT,7c170bbd-add3-4d0e-becc-e6d7b44c5503,0x800,0x40000)/File(\EFI\gentoo\grubx64.efi)
Boot001C* TOSHIBA HDWE160       BBS(HD,,0x0)AMBO
Boot001D* WDC WD1001FALS-00E8B0 BBS(HD,,0x0)AMBO
Boot001E* ubuntu        HD(1,GPT,7c170bbd-add3-4d0e-becc-e6d7b44c5503,0x800,0x40000)/File(\EFI\Ubuntu\grubx64.efi)
Boot0020* HL-DT-STDVDRAM GP50NB40 1.00  BBS(CDROM,,0x0)AMBO
```

```shell
$ efivar -l | grep fw
```

Please answer the following questions:

- Operating system and version: 4.14.65-gentoo
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): from portage (emerge -av fwupd)
- Have you tried rebooting? yes, though I don't see why that would make a difference
- Are you using an NVMe disk? not that I know
- Is secure boot enabled (only for the UEFI plugin)? no

As per the title, my 8BitDo Zero FC30 (the white and light-blue one) does not show up anywhere. When I plug it in, dmesg will show this:

```
[  761.545304] usb 3-10.2: new full-speed USB device number 15 using xhci_hcd
[  761.645346] usb 3-10.2: device descriptor read/64, error -32
[  761.865329] usb 3-10.2: device descriptor read/64, error -32
[  762.085293] usb 3-10.2: new full-speed USB device number 16 using xhci_hcd
[  762.185327] usb 3-10.2: device descriptor read/64, error -32
[  762.405330] usb 3-10.2: device descriptor read/64, error -32
[  762.525353] usb 3-10-port2: attempt power cycle
[  763.185286] usb 3-10.2: new full-speed USB device number 17 using xhci_hcd
[  763.185452] usb 3-10.2: Device not responding to setup address.
[  763.405426] usb 3-10.2: Device not responding to setup address.
[  763.625283] usb 3-10.2: device not accepting address 17, error -71
[  763.725275] usb 3-10.2: new full-speed USB device number 18 using xhci_hcd
[  763.725400] usb 3-10.2: Device not responding to setup address.
[  763.945425] usb 3-10.2: Device not responding to setup address.
[  764.165281] usb 3-10.2: device not accepting address 18, error -71
[  764.165336] usb 3-10-port2: unable to enumerate USB device
```

and lsusb is unaffected. I tried starting my 8BitDo normally and in L+R+Start mode as explained [here](https://github.com/hughsie/fwupd/issues/165) but it made no difference. I tried rebooting with the 8BitDo plugged in (in both modes) and it just wouldn't appear in dmesg, but still not visible in either lsusb and fwupd.

fwupdmgr get-devices shows nothing. fwupdmgr --verbose --show-all-devices get-devices only shows this:

```
GK104 [GeForce GTX 760]
  DeviceId:             3f0f51a4f03bccdad9f8ae1334bad4584b5ba875
  Guid:                 dc9bcb95-4fe4-5ea1-986c-c6ccf232bf8f
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Icon:                 audio-card
  Created:              2018-08-31
```

The daemon doesn't even blink as I plug/unplug the joypad, to the point that I'm wondering if that hardware is really supported.
I gave it a try on my Linux Mint install too, but it looks like they're stuck at fwupd 0.8.6 there. Neither get-devices nor lsusb will show my joypad on Mint.

My joypad charges just fine and works everywhere (phone, laptop, desktop) in bluetooth mode.

hughsie, 2018-08-31:
> [  762.405330] usb 3-10.2: device descriptor read/64, error -32
> [  762.525353] usb 3-10-port2: attempt power cycle
> [  763.185286] usb 3-10.2: new full-speed USB device number 17 using xhci_hcd
> [  763.185452] usb 3-10.2: Device not responding to setup address.

Can you try with another USB cable or with a different USB port please? This looks to me like a transport-layer problem rather than anything to do with userspace.

KingDuckZ, 2018-08-31:
I had already tried different cables. Just now I'm trying on my laptop (different computer than yesterday) with yet another cable and I'm getting this in dmesg:

```
[ 2103.230089] usb 4-1: new full-speed USB device number 2 using ohci-pci
[ 2103.670207] usb 4-1: device not accepting address 2, error -62
[ 2103.900063] usb 4-1: new full-speed USB device number 3 using ohci-pci
[ 2104.350072] usb 4-1: device not accepting address 3, error -62
[ 2104.350114] usb usb4-port1: attempt power cycle
[ 2104.900231] usb 4-1: new full-speed USB device number 4 using ohci-pci
[ 2105.120064] usb 4-1: device descriptor read/64, error -62
[ 2105.450168] usb 4-1: device descriptor read/64, error -62
[ 2105.780295] usb 4-1: new full-speed USB device number 5 using ohci-pci
[ 2105.990211] usb 4-1: device descriptor read/64, error -62
[ 2106.320091] usb 4-1: device descriptor read/64, error -62
[ 2106.440135] usb usb4-port1: unable to enumerate USB device
```

Unless there's anything that should be changed in my kernel configuration, I have no idea what to do next. If I ran into a strange piece of hardware and it would help you to try it yourself, I could find a way to send it somewhere by post.

hughsie, 2018-08-31:
> device not accepting address 3, error -62

One last thing to try is to put the device into any other "mode", if that controller supports it. e.g. Xinput or MaCOS mode. I honestly think the hardware is broken, e.g. a cracked trace on the socket or something.

KingDuckZ, 2018-08-31:
Ok so [this](https://www.gadgets4geeks.com.au/WebRoot/Master/Shops/gadgets4geeks/Products/FC30/8bitdo-zero-fc30-mini-bluetooth-game-controller-pad.jpg) is the pad I'm trying to update. [This](http://download.8bitdo.com/Manual/Controller/ZERO/ZERO_Manual_V4.pdf) seems to be the manual for it, but it doesn't explain how to activate the various modes. I know the L+R+Start mode from the github issue I linked earlier, but I have no idea on how to switch to other modes :(

superm1, 2018-09-27:
I agree it sounds like broken hardware. I'd say you should contact 8bitdo.

Roboe, 2018-09-28:
I have a fully functional 8bitdo Zero that has the same exact behaviour as exposed above by @KingDuckZ.

superm1, 2018-09-28:
@roboe You're seeing the same USB errors in your kernel log?

Roboe, 2018-11-02:
(Sorry for the delay)

Yes, same USB errors:

```
[14463.393144] usb usb1-port3: unable to enumerate USB device
[14505.423733] usb 1-1: new full-speed USB device number 17 using xhci_hcd
[14505.551760] usb 1-1: device descriptor read/64, error -71
[14505.787747] usb 1-1: device descriptor read/64, error -71
[14506.027701] usb 1-1: new full-speed USB device number 18 using xhci_hcd
[14506.155654] usb 1-1: device descriptor read/64, error -71
[14506.391734] usb 1-1: device descriptor read/64, error -71
[14506.499809] usb usb1-port1: attempt power cycle
[14507.151692] usb 1-1: new full-speed USB device number 19 using xhci_hcd
[14507.151920] usb 1-1: Device not responding to setup address.
```

System is a regular Debian Testing (`buster`), up-to-date. `fwupd` installed from the Debian repos:
```
$ fwupdmgr --version 
client version:	1.1.3
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34
daemon version:	1.1.3
```

`fwupdmgr get-devices` shows nothing, just like @KingDuckZ. I don't have any `efibootmgr` in my path, probably because my OS is booting in BIOS legacy mode?

i have the impression that this device is unable to interact via USB. The included microUSB-USB cable doesn't even transport data, it just charges whatever you plug with it (I used another to get the above logs, naturally).

closed 2019-09-30 by stale[bot]