fwupd/fwupd #721: Expose more metadata with --verbose

superm1, 2018-09-05:
I've gotten an ask to try to make metadata that fwupd reads about a device more readily accessible by other applications that want to access it.

My initial inclination was to use D-Bus or `libfwupd`, but that custom metdata isn't exposed through either.

Any thoughts on how to do this?

hughsie, 2018-09-05:
> to try to make metadata that fwupd reads about a device more readily accessible

Why?

superm1, 2018-09-05:
The application that would be interested in it is like an inventory collection system.  There's probably about 5 strings/integers of interesting stuff that fwupd reads about the device and can set for metadata internally.

Of course there is the option to just have that application go read the data itself from the device, but this seems suboptimal if all the code, device locking, etc is available already by fwupd.  Fwupd already emits signals when device gets plugged in, out, etc and abstracts a lot of that then.

hughsie, 2018-09-05:
> inventory collection system

I'm a little concerned about the privacy implications; any user can read properties from any devices. What kind of data would you want to include?

superm1, 2018-09-05:
Right, I suppose if this was something that could be supported it would need to be a privileged action protected by policy kit or similar.

The interesting fields include:
* Serial number of subcomponents (2 fields)
* Version of a sub device (already readable)
* "Marketing string", this is set by FW on the device
* Status (integer)
* Type of device (integer)

hughsie, 2018-09-05:
> Serial number of subcomponents

This is privileged info.

> (2 fields)

Why 2?

> "Marketing string", this is set by FW on the device

Is that different to the device model? Is it a display version?

> Status (integer)

What kind of mapping? It would need to be API rather than device-defined.

> Type of device (integer)

Again, what kind of enums?

Richard.

superm1, 2018-09-05:
> This is privileged info.

Right.  So I think if this was going to be something that could be offered by fwupd, it would have to be a privileged dbus interface.

>Why 2?

The device has two different serial numbers depending on how it's plugged in.

> Is that different to the device model? Is it a display version?

This can be set to the device name I think no problem.

> What kind of mapping? It would need to be API rather than device-defined.

It's basically setting bits in a byte or two based on if something failed, 0 if it's all OK.

> Again, what kind of enums?

Different configurations of the device.  Probably quite specific to that plugin/device.

hughsie, 2018-09-05:
> The device has two different serial numbers depending on how it's plugged in.

So we'd show the one that corresponds with the connection method used, surely?

> It's basically setting bits in a byte or two based on if something failed, 0 if it's all OK.

That's not going to work for a common API, I think that's feature creep I'm not super keen supporting. I think failure can be put in the update_error string.

> Different configurations of the device

That's hugely vague :) I think that's got a lot of potential for accidental API leakage. Different configurations of the device can be added to the device name, e.g. with a `[DVI]` suffix.

hughsie, 2018-09-06:
> So the "base" has a serial number and the "component" has a serial number. 

That works.

> spark discussion on getting that data out of fwupd

I think we only want to include a new `FWUPD_RESULT_KEY_SERIAL` property for devices if the `GDBusMethodInvocation` is called with sender `UID=0` -- maybe using a boolean to `fu_main_device_array_to_variant()` to indicate if the request is trusted or if specific fields should be filtered out.

superm1, 2018-09-06:
closed with https://github.com/hughsie/fwupd/commit/e3016602f8fa50a96709bd1d25f90b04df4c4440

closed 2018-09-06 by superm1